raspberrypi/linux #6799: Gasket driver required for Google Coral on Pineboards HatDrive! AI doesn't install on 6.12.20

MappyMatty, 2025-04-17:
### Describe the bug

I'm using a Raspberry Pi 5 with a Pineboards HatDrive! AI and Google Coral M.2 Accelerator A+E Key.
Unfortunately, this is now a dead link: https://pineboards.io/blogs/tutorials/how-to-configure-the-google-coral-edge-tpu-on-the-raspberry-pi-5
I'm trying to install the gasket driver but it's failing using the following 2 methods.

### Steps to reproduce the behaviour

Method 1:
`sudo apt-get install gasket-dkms -y`

Error:
```
Setting up gasket-dkms (1.0-18) ...
Loading new gasket-1.0 DKMS files...
Deprecated feature: REMAKE_INITRD (/usr/src/gasket-1.0/dkms.conf)
Building for 6.12.20+rpt-rpi-v8
Building initial module for 6.12.20+rpt-rpi-v8
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/gasket/1.0/source/dkms.conf)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/gasket/1.0/build/make.log for more information.
dpkg: error processing package gasket-dkms (--configure):
 installed gasket-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 gasket-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Contents of the referenced log file:
```
DKMS make.log for gasket-1.0 for kernel 6.12.20+rpt-rpi-v8 (aarch64)
Thu 17 Apr 21:40:34 BST 2025
make: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_core.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_ioctl.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_page_table.o
/var/lib/dkms/gasket/1.0/build/gasket_interrupt.c: In function 'gasket_handle_interrupt':
/var/lib/dkms/gasket/1.0/build/gasket_interrupt.c:161:17: error: too many arguments to function 'eventfd_signal'
  161 |                 eventfd_signal(ctx, 1);
      |                 ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/gasket/1.0/build/gasket_interrupt.h:11,
                 from /var/lib/dkms/gasket/1.0/build/gasket_interrupt.c:4:
/usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/eventfd.h:87:20: note: declared here
   87 | static inline void eventfd_signal(struct eventfd_ctx *ctx)
      |                    ^~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/gasket/1.0/build/gasket_core.c:1375:19: error: 'no_llseek' undeclared here (not in a function); did you mean 'noop_llseek'?
 1375 |         .llseek = no_llseek,
      |                   ^~~~~~~~~
      |                   noop_llseek
/var/lib/dkms/gasket/1.0/build/gasket_core.c: In function 'gasket_register_device':
/var/lib/dkms/gasket/1.0/build/gasket_core.c:1841:41: error: passing argument 1 of 'class_create' from incompatible pointer type [-Werror=incompatible-pointer-types]
 1841 |                 class_create(driver_desc->module, driver_desc->name);
      |                              ~~~~~~~~~~~^~~~~~~~
      |                                         |
      |                                         struct module *
In file included from /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/device.h:31,
                 from /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/cdev.h:8,
                 from /var/lib/dkms/gasket/1.0/build/gasket_core.h:11,
                 from /var/lib/dkms/gasket/1.0/build/gasket_core.c:12:
/usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/device/class.h:228:54: note: expected 'const char *' but argument is of type 'struct module *'
  228 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/var/lib/dkms/gasket/1.0/build/gasket_core.c:1841:17: error: too many arguments to function 'class_create'
 1841 |                 class_create(driver_desc->module, driver_desc->name);
      |                 ^~~~~~~~~~~~
/usr/src/linux-headers-6.12.20+rpt-common-rpi/include/linux/device/class.h:228:29: note: declared here
  228 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/gasket/1.0/build/gasket_core.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/gasket/1.0/build] Error 2
make: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
```

Method 2:
```
git clone https://github.com/google/gasket-driver.git
cd gasket-driver
sudo debuild -us -uc -tc -b
cd ..
sudo dpkg -i gasket-dkms_1.0-18_all.deb
```

Error:
```
Setting up gasket-dkms (1.0-18) ...
Loading new gasket-1.0 DKMS files...
Deprecated feature: REMAKE_INITRD (/usr/src/gasket-1.0/dkms.conf)
Building for 6.12.20+rpt-rpi-v8
Building initial module for 6.12.20+rpt-rpi-v8
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/gasket/1.0/source/dkms.conf)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/gasket/1.0/build/make.log for more information.
dpkg: error processing package gasket-dkms (--install):
 installed gasket-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 gasket-dkms
```

Contents of the referenced log file:
```
DKMS make.log for gasket-1.0 for kernel 6.12.20+rpt-rpi-v8 (aarch64)
Thu 17 Apr 21:48:25 BST 2025
make: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_core.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_ioctl.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_interrupt.o
  CC [M]  /var/lib/dkms/gasket/1.0/build/gasket_page_table.o
/var/lib/dkms/gasket/1.0/build/gasket_core.c:1376:19: error: 'no_llseek' undeclared here (not in a function); did you mean 'noop_llseek'?
 1376 |         .llseek = no_llseek,
      |                   ^~~~~~~~~
      |                   noop_llseek
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/gasket/1.0/build/gasket_core.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/gasket/1.0/build] Error 2
make: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
```

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

2025/03/19 13:41:26 
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)

Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

MappyMatty, 2025-04-19:
Maybe this bug is not for here and is for the Google gasket-driver repo? I have built and installed the package using info found here: https://github.com/google/gasket-driver/issues/36

MappyMatty, 2025-04-23:
dmesg returns `apex 0001.03.00.0: Couldn’t initialize interrupts: -524`

pelwell, 2025-04-23:
You could try with `dtoverlay=pciex1-compat-pi5,nomip`.

hungnguyen167, 2025-04-25:
Hi guys, is there any update on this issue?

MappyMatty, 2025-04-26:
Thanks for the suggestion @pelwell 
I've tried `dtoverlay=pciex1-compat-pi5,no-mip` by itself and in combination with `dtoverlay=pineboards-hat-ai` and after a reboot I get the same `apex 0001.03.00.0: Couldn’t initialize interrupts: -524` in both scenarios.

MappyMatty, 2025-04-26:
With both overlays removed and a reboot I get a different message from dmesg of `apex 0001:03:00.0: Couldn't initialize interrupts: -28`

mattblonski, 2025-04-29:
I'm having exactly same issue with kernel 6.12.20 and gasket driver pulled from git repo and llseek modified.
The accellerator does not work with frigate and I get dmesg:
`kernel: apex 0001:03:00.0: Couldn't initialize interrupts: -524`

Finally I ended up with restoring kernel 6.6.74 via apt, I've had to remove wireguard-tools and wireguard-modules to be able to remove latest kernel.
`apt install --reinstall raspi-firmware`
Then compiled wireguard-tools by hand, to restore my VPN.

mattblonski, 2025-05-19:
Latest update: my PCI Edge TPU now works with linux-image-rpi-v8 kernel package, ver. 6.12.25+rpt-rpi-v8 and https://github.com/google/gasket-driver/issues/36 modification. 

Hopefully future kernel releases will not cause the same issues.