raspberrypi/linux #6800: mcp251xfd: Incorrect TDC behavior

optlink, 2025-04-18:
### Describe the bug

The TDC configuration used in the driver is incorrect. It is forced on for all CAN FD bit rates which causes bus-off at lower bit rates that require a DBRP larger than 1. Hacking the driver code to disable TDC when DBRP > 1 fixes the issue but I suspect that the real problem is that this needs to be controllable from user space as TDC usage is dependent on circuit design, data bit rate, and system clock.

### Steps to reproduce the behaviour

In `config.txt`:
```
[all]
dtoverlay=mcp251xfd,spi0-0,interrupt=25,rx_interrupt=24
```

```sudo ip link set can0 down
sudo ip link set can0 type can sample-point 0.875 bitrate 125000 dsample-point 0.875 dbitrate 500000 fd on berr-reporting on restart-ms 100
sudo ethtool -C can0 rx-usecs-irq 10000 rx-frames-irq 8 tx-usecs-irq 10000 tx-frames-irq 2
sudo ip link set can0 up
cansend can0 00000000##1.00.00
```

Then observe the bus-off condition in `dmesg`. This also happens for 250 / 1000 but works for 500 / 2000 and 1000 / 4000. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

`cat /etc/rpi-issue`:
```
Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ecda2579ebbc125b321984c571e3128f6fd959d4, stage2
```

`vcgencmd version`:
```
Mar 19 2025 18:24:21 
Copyright (c) 2012 Broadcom
version ca6e8171a80ea46924ffaa629250bfb482f3a02c (clean) (release) (start)
```

`uname -a`:
```
Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

### Additional context

I am using a custom PCB with an estimated worst case loop delay of ~216 ns and average case ~100 ns. The system clock is 40 MHz.

pelwell, 2025-04-18:
This is not our driver, and not our area of expertise (an understatement). But there are some CAN bus users who visit these parts who may have some ideas.

optlink, 2025-04-18:
Fair enough, I just thought with the other issues involving this driver and how they got resolved that somebody here would be able to help. I'll consider braving the linux mailing list next if nothing happens.

Sorry for the trouble.

closed 2025-05-19 by optlink

pelwell, 2025-05-19:
Thanks for the update. I'm glad it's sorted for you.