raspberrypi/linux #6808: kernel unicode support missing - ext4 casefold feature not avaailabe

bjacke, 2025-04-23:
when trying to mount a ext4 fs with casefold enabled this error comes up:

EXT4-fs (dm-4): Filesystem with casefold feature cannot be mounted without CONFIG_UNICODE

I want the casefold feature enabled because this allows a huge speed-up for samba fileservers.

pelwell, 2025-04-23:
This feature looks like it can be enabled as a module, with loading only when actually needed, so the only cost should be a bit of storage space.

pelwell, 2025-04-23:
See https://github.com/raspberrypi/linux/pull/6809. In about 40 minutes you should be able to install a trial kernel using `sudo rpi-update pulls/6809`. Back up any important data first.

pelwell, 2025-04-25:
#6809 has now been merged.

bjacke, 2025-05-12:
thank you very much for the super quick change! :-) It works well for me now!

closed 2025-05-12 by bjacke