raspberrypi/linux #6821: no disk created for nvme on pcie ASM1806 switch port 0, on raspberry pi 5

atmgnd, 2025-04-28:
### Describe the bug

## Describe the bug
I am using a ASM1806 4-Port PCIe switch, I tested it on x86 pc, all disk port are working fine. But raspberry pi 5, nvme on port 0 shows in lspci but not lsblk

lspci

```
qizhou@raspberrypi:/dev/shm $ lspci -nn
0001:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
0001:01:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:02.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:06.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:0e.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:03:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD [2646:5016] (rev 01)
0001:04:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD [2646:5016] (rev 01)
0001:05:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PDP3 NVMe SSD [2646:500d] (rev 01)
0002:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
0002:01:00.0 Ethernet controller [0200]: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge [1de4:0001]
qizhou@raspberrypi:/dev/shm $ lspci -nn > lspci_nn.txt
qizhou@raspberrypi:/dev/shm $ ls
cmdline.txt  ddns.log  dmesg.txt  lspci_nn.txt  lspci_tv.txt  lspci_vvvv.txt  tailscale_keepalive.log
qizhou@raspberrypi:/dev/shm $ lspci -nn
0001:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
0001:01:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:02.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:06.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:02:0e.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
0001:03:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD [2646:5016] (rev 01)
0001:04:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD [2646:5016] (rev 01)
0001:05:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PDP3 NVMe SSD [2646:500d] (rev 01)
0002:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
0002:01:00.0 Ethernet controller [0200]: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge [1de4:0001]
qizhou@raspberrypi:/dev/shm $ lspci -vt
-+-[0000:00]-
 +-[0001:00]---00.0-[01-06]----00.0-[02-06]--+-00.0-[03]----00.0  Kingston Technology Company, Inc. OM3PGP4 NVMe SSD
 |                                           +-02.0-[04]----00.0  Kingston Technology Company, Inc. OM3PGP4 NVMe SSD
 |                                           +-06.0-[05]----00.0  Kingston Technology Company, Inc. OM3PDP3 NVMe SSD
 |                                           \-0e.0-[06]--
 \-[0002:00]---00.0-[01]----00.0  Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
qizhou@raspberrypi:/dev/shm $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0    58G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
├─mmcblk0p2 179:2    0    30G  0 part /
└─mmcblk0p3 179:3    0  27.4G  0 part /home
nvme0n1     259:0    0 119.2G  0 disk 
nvme1n1     259:1    0 119.2G  0 disk 
└─nvme1n1p1 259:2    0 119.2G  0 part
```

detailed lspci -vv output

[lspci_vvvv.txt](https://github.com/user-attachments/files/19941771/lspci_vvvv.txt)

dmesg

[dmesg.txt](https://github.com/user-attachments/files/19941775/dmesg.txt)

### Steps to reproduce the behaviour

Insert a ASM1806 4-Port pcie switch on raspberry pi 5, then insert a nvme on port 0

### Device (s)

Raspberry Pi 5

### System
[... 725 lines cut ...]
[    3.373958] Bluetooth: hci0: BCM: chip id 107
[    3.374452] Bluetooth: hci0: BCM: features 0x2f
[    3.375563] Bluetooth: hci0: BCM4345C0
[    3.375574] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    3.377054] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    4.046096] Bluetooth: hci0: BCM: features 0x2f
[    4.047442] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    4.047444] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    4.047751] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    4.346085] nvme nvme0: using unchecked data buffer
[    4.416496] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.416502] Bluetooth: BNEP filters: protocol multicast
[    4.416507] Bluetooth: BNEP socket layer initialized
[    4.418782] Bluetooth: MGMT ver 1.23
[    4.424614] NET: Registered PF_ALG protocol family
[    5.213994] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    5.214002] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    5.216661] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    5.225136] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    6.231207] tun: Universal TUN/TAP device driver, 1.6
[    9.315344] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   10.960927] Bluetooth: RFCOMM TTY layer initialized
[   10.960942] Bluetooth: RFCOMM socket layer initialized
[   10.960960] Bluetooth: RFCOMM ver 1.11
```

P33M, 2025-04-28:
This is odd:
```
[    0.466530] nvme nvme0: pci function 0001:03:00.0
[    0.466535] nvme 0001:03:00.0: enabling device (0000 -> 0002)
[    0.518091] nvme nvme0: pci function 0001:04:00.0
[    0.518096] nvme 0001:04:00.0: enabling device (0000 -> 0002)
```
According to that, two different PCI devices got assigned the same device node.
What are the contents of the directory `/dev/disk/by-path`?

P33M, 2025-04-28:
What happens if you replace the Kingston NVMe drive on port 0 with one not manufactured by Kingston?

atmgnd, 2025-04-28:
> What happens if you replace the Kingston NVMe drive on port 0 with one not manufactured by Kingston?

I do not have any none Kingston nvme, I have two type Kingston nvme,  OM3PGP4 and OM3PDP3. Even I only plug one nvme on the pcie switch in port 0, The problem is the same(only shows in lspci, not lsblk)

```
qizhou@raspberrypi:~ $ lspci -knn
0001:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
	Kernel driver in use: pcieport
0001:01:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
	Kernel driver in use: pcieport
0001:02:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
	Kernel driver in use: pcieport
0001:02:02.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
	Kernel driver in use: pcieport
0001:02:06.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
	Kernel driver in use: pcieport
0001:02:0e.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1806 4-Port PCIe x2 Gen2 Packet Switch [1b21:1806] (rev 01)
	Kernel driver in use: pcieport
0001:03:00.0 Non-Volatile memory controller [0108]: Kingston Technology Company, Inc. OM3PDP3 NVMe SSD [2646:500d] (rev 01)
	Subsystem: Kingston Technology Company, Inc. OM3PDP3 NVMe SSD [2646:500d]
0002:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge [14e4:2712] (rev 21)
	Kernel driver in use: pcieport
0002:01:00.0 Ethernet controller [0200]: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge [1de4:0001]
	Kernel driver in use: rp1
qizhou@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0   58G  0 disk 
├─mmcblk0p1 179:1    0  512M  0 part /boot/firmware
├─mmcblk0p2 179:2    0   30G  0 part /
└─mmcblk0p3 179:3    0 27.4G  0 part /home
qizhou@raspberrypi:~ $ ls -l /dev/disk/by-id
total 0
lrwxrwxrwx 1 root root 13 Apr 28 23:36 mmc-SD64G_0x3940041e -> ../../mmcblk0
lrwxrwxrwx 1 root root 15 Apr 28 23:36 mmc-SD64G_0x3940041e-part1 -> ../../mmcblk0p1
lrwxrwxrwx 1 root root 15 Apr 28 23:36 mmc-SD64G_0x3940041e-part2 -> ../../mmcblk0p2
lrwxrwxrwx 1 root root 15 Apr 28 23:36 mmc-SD64G_0x3940041e-part3 -> ../../mmcblk0p3
qizhou@raspberrypi:~ $ ls -l /dev/disk/by-path
total 0
lrwxrwxrwx 1 root root 13 Apr 28 23:36 platform-1000fff000.mmc -> ../../mmcblk0
lrwxrwxrwx 1 root root 15 Apr 28 23:36 platform-1000fff000.mmc-part1 -> ../../mmcblk0p1
lrwxrwxrwx 1 root root 15 Apr 28 23:36 platform-1000fff000.mmc-part2 -> ../../mmcblk0p2
lrwxrwxrwx 1 root root 15 Apr 28 23:36 platform-1000fff000.mmc-part3 -> ../../mmcblk0p3
qizhou@raspberrypi:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.25-v8-16k+ (qizhou@raspberrypi) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #5 SMP PREEMPT Mon Apr 28 21:34:16 CST 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000032000000, size 192 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000032000000..0x000000003dffffff (196608 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16640..0x000000003fd1667d (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000001fffffff] (512MB)
[    0.000000] Faking node 1 at [mem 0x0000000020000000-0x000000003fffffff] (512MB)
[    0.000000] Faking node 2 at [mem 0x0000000040000000-0x000000005fffffff] (512MB)
[    0.000000] Faking node 3 at [mem 0x0000000060000000-0x000000007fffffff] (512MB)
[    0.000000] Faking node 4 at [mem 0x0000000080000000-0x000000009fffffff] (512MB)
[    0.000000] Faking node 5 at [mem 0x00000000a0000000-0x00000000bfffffff] (512MB)
[    0.000000] Faking node 6 at [mem 0x00000000c0000000-0x00000000dfffffff] (512MB)
[    0.000000] Faking node 7 at [mem 0x00000000e0000000-0x00000000ffffffff] (512MB)
[    0.000000] NUMA: Initialized distance table, cnt=8
[    0.000000] NODE_DATA(0) allocated [mem 0x1fffd200-0x1fffffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x3fbfd200-0x3fbfffff]
[    0.000000] NODE_DATA(2) allocated [mem 0x5fffd200-0x5fffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0x7fffd200-0x7fffffff]
[    0.000000] NODE_DATA(4) allocated [mem 0x9fffd200-0x9fffffff]
[    0.000000] NODE_DATA(5) allocated [mem 0xbfffd200-0xbfffffff]
[... 724 lines cut ...]
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 44
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
[... cut ...]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 44
	Region 0: Memory at 1f00410000 (32-bit, non-prefetchable) [size=16K]
	Region 1: Memory at 1f00000000 (32-bit, non-prefetchable) [virtual] [size=4M]
	Region 2: Memory at 1f00400000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: rp1

qizhou@raspberrypi:~ $ 
```

atmgnd, 2025-04-28:
I noticed there no driver in use and Latency element in `lspci -vvv` node 0001:03:00.0, dont know why.
what's the `Latency: 0` mean in node 0001:04:00.0

```
0001:03:00.0 Non-Volatile memory controller: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD (rev 01) (prog-if 02 [NVM Express])
        Subsystem: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD
        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 39
        Region 0: Memory at 1b00000000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>

0001:04:00.0 Non-Volatile memory controller: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD (rev 01) (prog-if 02 [NVM Express])
        Subsystem: Kingston Technology Company, Inc. OM3PGP4 NVMe SSD
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 41
        Region 0: Memory at 1b00200000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>
        Kernel driver in use: nvme
```

[lspci_vvv_02.txt](https://github.com/user-attachments/files/19943638/lspci_vvv_02.txt)

P33M, 2025-04-29:
This is one of the set of PCIe switches that fail to correctly translate addresses below ~64KiB.
We've lost the mmio-hi feature of the downstream driver/DT which puts 32-bit BARs at +2GiB.

What's annoying is that the nvme driver doesn't correctly report failure to probe.

atmgnd, 2025-04-29:
There is also some other warnings, no idea what it means

```
[    0.445173] pci 0001:01:00.0: bridge window [io  size 0x1000]: can't assign; no space
[    0.445175] pci 0001:01:00.0: bridge window [io  size 0x1000]: failed to assign
[    0.638480] pci 0002:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
```

Some additional information about kernel comandline and dtoverlay

```
qizhou@raspberrypi:~ $ cat /boot/firmware/config.txt | tail -3
dtoverlay=cma,cma-192
dtparam=pcie-32bit-dma-pi5
dtoverlay=pciex1-compat-pi5,no-mip
qizhou@raspberrypi:~ $ cat /boot/firmware/cmdline.txt | fold -w 80 
console=serial0,115200 console=tty1 root=PARTUUID=a7df9f78-02 rootfstype=ext4 fs
ck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.iee
e80211_regdom=CN    nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_por
t_pm=off
```

@P33M do you have any sugguestions ?

atmgnd, 2025-04-29:
> Some additional information about kernel comandline and dtoverlay
>
> ```
> qizhou@raspberrypi:~ $ cat /boot/firmware/config.txt | tail -3
> dtoverlay=cma,cma-192
> dtparam=pcie-32bit-dma-pi5 [...]

If I don't add the above configuration, I won't be able to successfully mount any NVMe disks
After adding the above configurations, you can mount the recognized disks
However, regardless of whether the above configurations are present, the disk on port 0 cannot be correctly identified

P33M, 2025-04-29:
Can you test with `dtoverlay=pciex1-compat-pi5,no-mip,mmio-hi`?

pelwell, 2025-04-29:
> The read speed is too slow

I'm more interested in what happened to the remaining ~7900MB, since only 280MB were read.

pelwell, 2025-04-29:
How is your system powered?

atmgnd, 2025-04-29:
> I'm more interested in what happened to the remaining ~7900MB, since only 280MB were read.

I stopped it manually because it would take a long time to finish

> If adding pcie_aspm=off to /boot/firmware/cmdline.txt doesn't fix it, then it's up to your hardware vendor to provide support.

I have already added this kernel parameter before `nvme_core.default_ps_max_latency_us=0 pcie_aspm=off pcie_por
t_pm=off`

atmgnd, 2025-04-29:
> How is your system powered?

Hi [pelwell](https://github.com/pelwell), I am using the official pi5 27W power supply

atmgnd, 2025-04-29:
> `[ 93.161628] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0xffff` = symptomatic of poor link quality, often caused by non-compliant connections between the Pi 5 expansion connector and switch, or between switch and endpoints. I'm not surprised that read speeds are poor (though that doesn't explain write speeds).

I am using Geekworm X1010 expansion board, If I plug the nvme disk directly into the x1010, the read and write speeds are normal. Only when the asm1806 pcie switch is used, the read speed is too low.

closed 2025-05-09 by atmgnd