raspberrypi/linux #6832: 6.12 regression: igc: probe failed

delgh1, 2025-05-02:
### Describe the bug

hi,

I use an Intel I226-V pcie adapter with two ports. But with igc in linux 6.12.20, only one ethernet port can be enabled, the other one fails. This is a regression, it was working with linux 6.6.

### Steps to reproduce the behaviour

linux-image-rpi-2712/stable,now 1:6.12.20-1+rpt1~bpo12+1 arm64 [installed]

### Device (s)

Raspberry Pi 5

### System

# cat /etc/os-release

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
Linux raspberrypi10 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

### Logs

```
[  324.296910] Intel(R) 2.5G Ethernet Linux Driver
[  324.302080] Copyright(c) 2018 Intel Corporation.
[  324.307390] igc 0001:03:00.0: PCIe PTM not supported by PCIe bus/controller
[  324.354492] igc 0001:03:00.0: probe with driver igc failed with error -13
[  324.361960] igc 0001:04:00.0: PCIe PTM not supported by PCIe bus/controller
[  324.395036] igc 0001:04:00.0 (unnamed net_device) (uninitialized): PHC added
[  324.427994] igc 0001:04:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)
[  324.436813] igc 0001:04:00.0 eth0: MAC: 02:76:c6:00:fc:db
[  324.446385] igc 0001:04:00.0 enP1p4s0: renamed from eth0
[  324.504562] igc 0001:04:00.0 enP1p4s0: entered promiscuous mode
[  324.511158] igc 0001:04:00.0 enP1p4s0: entered allmulticast mode
[  328.103107] igc 0001:04:00.0 enP1p4s0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
```

```
0001:03:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
        Subsystem: Intel Corporation Ethernet Controller I226-V
        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 39
        Region 0: Memory at 1b00000000 (32-bit, non-prefetchable) [virtual] [size=1M]
        Region 3: Memory at 1b00100000 (32-bit, non-prefetchable) [size=16K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [70] MSI-X: Enable- Count=5 Masked-
                Vector table: BAR=3 offset=00000000
                PBA: BAR=3 offset=00002000
        Capabilities: [a0] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
[... 113 lines cut ...]
                          PortCommonModeRestoreTime=55us PortTPowerOnTime=70us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=70us
        Kernel driver in use: igc
        Kernel modules: igc
```

```
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0001:01:00.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0001:02:03.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0001:02:07.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0001:03:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
0001:04:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
0002:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0002:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

```
-+-[0000:00]-
 +-[0001:00]---00.0-[01-04]----00.0-[02-04]--+-03.0-[03]----00.0  Intel Corporation Ethernet Controller I226-V [8086:125c]
 |                                           \-07.0-[04]----00.0  Intel Corporation Ethernet Controller I226-V [8086:125c]
 \-[0002:00]---00.0-[01]----00.0  Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge [1de4:0001]
```

P33M, 2025-05-06:
Please update to the latest apt packages and re-test. `sudo apt update && sudo apt full-upgrade`.

delgh1, 2025-05-06:
> Please update to the latest apt packages and re-test. `sudo apt update && sudo apt full-upgrade`.

It was fixed somewhere between 6.12.20 and 6.12.25. Thanks

closed 2025-05-06 by delgh1