raspberrypi/linux #6833: Incompatibility between Raspberry Pi 5 VL805 USB controller and Oregon Scientific WMRS200 (USB 1.1 HID device)

jpuigsegur, 2025-05-02:
### Describe the bug

The Oregon Scientific WMRS200 (USB 1.1 HID weather station, ID 0fde:ca01) fails to initialize when connected directly to a Raspberry Pi 5. The device is not recognized (lsusb shows nothing) and the kernel logs show repeated Cannot enable and unable to enumerate USB device errors.

However, the same device:
- Works flawlessly on a Linux laptop
- Works correctly on the same Raspberry Pi 5 if connected through a USB 2.0 hub

This points to a hardware/firmware-level compatibility issue between the VL805 USB controller on the Raspberry Pi 5 and older USB 1.1 HID-class devices.

### Steps to reproduce the behaviour

Just plug a Oregon Scientific WMRS200 (USB 1.1 HID weather station, ID 0fde:ca01) to a RPI5

### Device (s)

Raspberry Pi 5

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2

$ vcgencmd version
2024/11/12 16:10:44 
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)

$ uname -a
Linux prova 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

### Logs

$ dmesg | grep usb

[  682.406587] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[  683.306580] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[  683.306627] usb usb1-port2: attempt power cycle
[  684.282584] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[  685.182580] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[  685.182632] usb usb1-port2: unable to enumerate USB device

usbmon logs when plugging device. Logs appear for some seconds and then stop:

$ sudo cat /sys/kernel/debug/usb/usbmon/0u
ffff80000b9c6580 212689319 S Ci:1:001:0 s a3 00 0000 0001 0004 4 <
ffff80000b9c6580 212689328 C Ci:1:001:0 0 4 = 00010000
ffff80000b9c6580 212689332 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 212689335 C Ci:1:001:0 0 4 = 01030100
ffff80000b9c6580 212689337 S Co:1:001:0 s 23 01 0010 0002 0000 0
ffff80000b9c6580 212689341 C Co:1:001:0 0 0
ffff8000068b8540 212796016 S Ii:1:001:1 -115:2048 4 <
ffff80000b9c6580 212796025 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 212796035 C Ci:1:001:0 0 4 = 01030000
ffff80000b9c6580 212796091 S Co:1:001:0 s 23 03 0004 0002 0000 0
ffff80000b9c6580 212796102 C Co:1:001:0 0 0
ffff8000068b8540 212846122 C Ii:1:001:1 0:2048 1 = 04
ffff8000068b8540 212846124 S Ii:1:001:1 -115:2048 4 <
ffff80000b9c6580 212864007 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 212864015 C Ci:1:001:0 0 4 = 01031100
ffff80000b9c6580 212864017 S Co:1:001:0 s 23 01 0010 0002 0000 0
ffff80000b9c6580 212864021 C Co:1:001:0 0 0
ffff80000b9c6580 212864023 S Co:1:001:0 s 23 03 0004 0002 0000 0
ffff80000b9c6580 212864027 C Co:1:001:0 0 0
ffff8000068b8540 212936011 C Ii:1:001:1 0:2048 1 = 04
ffff8000068b8540 212936012 S Ii:1:001:1 -115:2048 4 <
ffff80000b9c6580 213072007 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 213072016 C Ci:1:001:0 0 4 = 01031100
ffff80000b9c6580 213072018 S Co:1:001:0 s 23 01 0010 0002 0000 0
[... 129 lines cut ...]
ffff80000b9c6580 216056022 S Co:1:001:0 s 23 01 0010 0002 0000 0
ffff80000b9c6580 216056026 C Co:1:001:0 0 0
ffff80000b9c6580 216056027 S Co:1:001:0 s 23 03 0004 0002 0000 0
ffff80000b9c6580 216056032 C Co:1:001:0 0 0
ffff8000068b8540 216160025 C Ii:1:001:1 0:2048 1 = 04
ffff8000068b8540 216160026 S Ii:1:001:1 -115:2048 4 <
ffff80000b9c6580 216264010 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 216264023 C Ci:1:001:0 0 4 = 01031100
ffff80000b9c6580 216264027 S Co:1:001:0 s 23 01 0010 0002 0000 0
ffff80000b9c6580 216264031 C Co:1:001:0 0 0
ffff80000b9c6580 216264032 S Co:1:001:0 s 23 03 0004 0002 0000 0
ffff80000b9c6580 216264037 C Co:1:001:0 0 0
ffff8000068b8540 216408017 C Ii:1:001:1 0:2048 1 = 04
ffff8000068b8540 216408019 S Ii:1:001:1 -115:2048 4 <
ffff80000b9c6580 216472010 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 216472018 C Ci:1:001:0 0 4 = 01031100
ffff80000b9c6580 216472020 S Co:1:001:0 s 23 01 0010 0002 0000 0
ffff80000b9c6580 216472023 C Co:1:001:0 0 0
ffff80000b9c6580 216472029 S Co:1:001:0 s 23 01 0001 0002 0000 0
ffff80000b9c6580 216472034 C Co:1:001:0 0 0
ffff80000b9c6580 216472125 S Co:1:001:0 s 23 01 0001 0002 0000 0
ffff80000b9c6580 216472132 C Co:1:001:0 0 0
ffff80000b9c6580 216472137 S Ci:1:001:0 s a3 00 0000 0002 0004 4 <
ffff80000b9c6580 216472144 C Ci:1:001:0 0 4 = 01030000
ffff8000068b8540 216472153 C Ii:1:001:1 -2:2048 0

pelwell, 2025-05-02:
The Pi 5 does not have a VL805 USB controller.

jpuigsegur, 2025-05-03:
> The Pi 5 does not have a VL805 USB controller.

Corrected.

jpuigsegur, 2025-05-03:
Just had access to a pi4 and was able to chec that it works:

dmesg logs:
```
[2417440.474449] usb 1-1.3: new low-speed USB device number 4 using xhci_hcd
[2417440.580984] usb 1-1.3: New USB device found, idVendor=0fde, idProduct=ca01, bcdDevice= 3.02
[2417440.581009] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[2417440.581023] usb 1-1.3: Product: Universal Bridge
[2417440.589951] hid-generic 0003:0FDE:CA01.0001: hiddev96,hidraw0: USB HID v1.10 Device [Universal Bridge] on usb-0000:01:00.0-1.3/input0

```

```
$ lsusb | grep 0fde
Bus 001 Device 004: ID 0fde:ca01 Oregon Scientific WMRS200 weather station
```

P33M, 2025-05-06:
http://www.distec1.be/gs/mediaobjects/Oregon%20Scientific/Manuals/S%20-%20Z/WMRS200_EN.pdf

I see the minimum system requirements are "Windows 98" and "300MHz or above CPU speed". So this thing is from the stone age, which means ancient microcontroller doing USB device port. Hence:

> Steps to reproduce the behaviour
> Just plug a Oregon Scientific WMRS200 (USB 1.1 HID weather station, ID 0fde:ca01) to a RPI5

is going to be a non-starter.

Ordinarily, an interop issue with something this obsolete isn't going to attract much attention, but as you have it working on a Pi 4, this is a regression. It's signalling attach, but the bus reset upsets it. 

What happens with `usb_max_current_enable=1` in /boot/firmware/config.txt?
Alternatively, what happens if you power cycle the port with the device attached (and only the device)?

```
$ pinctrl get | grep VBUS
42: a2    pd | hi // USB_VBUS_EN/GPIO42 = VBUS_EN1
43: a2    pu | hi // USB_OC_N/GPIO43 = VBUS_OC1
$ pinctrl set 42 op dl ; sleep 1 ; pinctrl set 42 op dh
```

jpuigsegur, 2025-05-07:
> What happens with usb_max_current_enable=1 in /boot/firmware/config.txt?

Same behaviour

> Alternatively, what happens if you power cycle the port with the device attached (and only the device)?

Nothing happens:
- nothing written on dmesg
- nothing written on usbmon
- nothing new appears in lsusb

P33M, 2025-05-08:
With a suitable multimeter and the device plugged into the top USB2.0 port (after enumeration has failed), can you measure the voltages on the underside of the Pi 5 between the following test points?

TP25-TP68
TP25-TP69

jpuigsegur, 2025-05-08:
TP25-TP68  --> 3.03 V
TP25-TP69  --> 0 V

P33M, 2025-05-09:
Those are nominally correct voltages for low-speed device attach. In that case further debug isn't going to be useful without a bus analyzer trace. Most generic logic analysers can also decode USB1.1.

Is this something you have at your disposal?

jpuigsegur, 2025-05-09:
I am afraid I don't have this type of equipment.

P33M, 2025-05-12:
This is a stab in the dark, but is there any difference if you run these two commands

```

jpuigsegur, 2025-05-15:
No difference. Same behaviour as before