raspberrypi/linux #6834: PCI devices not part of IOMMU group (important for passthrough PCI devices into VMs)

AlbrechtL, 2025-05-05:
### Describe the bug

It seems that PCIe devices that are added to a Raspberry Pi 5 are not part of a IOMMU groups. But this is the prerequisite to passthrough PCI devices into a VM with qemu. More details you can find in the issue here https://github.com/AlbrechtL/openwrt-docker/issues/38.

My PCIe devices at the Raspberry Pi 5:
```
$ lspci
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30)
0001:01:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0001:01:00.1 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0001:01:00.2 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0001:01:00.3 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
0002:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30)
0002:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

The Intel I350 is missing in the IOMMU groups:
```
$ sudo find /sys/kernel/iommu_groups/ -type l
/sys/kernel/iommu_groups/1/devices/107c580000.hvs
/sys/kernel/iommu_groups/1/devices/axi:gpu
/sys/kernel/iommu_groups/0/devices/1000800000.codec
/sys/kernel/iommu_groups/0/devices/1000880000.pisp_be
```

Unfortunately, I didn't find any information if this is general possible or not. To test it, I compiled the kernel with `CONFIG_VFIO_PCI=m` support.

On the Internet, you can find many examples where users tried to use PCI passthrough with a Raspberry Pi 5.
* https://forums.raspberrypi.com/viewtopic.php?t=374670
* https://forums.raspberrypi.com/viewtopic.php?t=383922

### Steps to reproduce the behaviour

1. Add a PCIe device to the Raspberry Pi 5
2. Check `lspci` if the device is showing up
3. Check `find /sys/kernel/iommu_groups/ -type l` if the PCIe IDs of the added device are showing up

### Device (s)

Raspberry Pi 5

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage2
```

```
$ vcgencmd version
2025/01/22 00:16:51 
Copyright (c) 2012 Broadcom
version a7753063 (release) (embedded)
```

```
$ uname -a
Linux albrecht-rpi5 6.12.22-v8-16k-AlbrechtL+ #8 SMP PREEMPT Mon May  5 07:12:07 CEST 2025 aarch64 GNU/Linux
```

### Logs

```
$ dmesg | grep -i iommu
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=4 system_heap.max_order=0 smsc95xx.macaddr=2C:CF:67:7A:0F:14 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=13403702-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE iommu=on arm64_dma_use_iommu=on smmu.enable=1
[    0.000000] Unknown kernel command line parameters "iommu=on arm64_dma_use_iommu=on", will be passed to user space.
[    0.060875] iommu: Default domain type: Translated
[    0.060879] iommu: DMA domain TLB invalidation policy: strict mode
[    0.355890] bcm2712-iommu-cache 1000005b00.iommuc: bcm2712_iommu_cache_probe
[    0.644750] bcm2712-iommu 1000005100.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.645109] bcm2712-iommu 1000005100.iommu: bcm2712_iommu_probe: Success
[    0.645510] bcm2712-iommu 1000005200.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.645887] bcm2712-iommu 1000005200.iommu: bcm2712_iommu_probe: Success
[    0.646267] bcm2712-iommu 1000005280.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.646612] bcm2712-iommu 1000005280.iommu: bcm2712_iommu_probe: Success
[    3.288053]     iommu=on
[    3.288054]     arm64_dma_use_iommu=on
[    6.305228] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    6.305236] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_probe_device: MMU 1000005100.iommu
[    6.305241] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_device_group: MMU 1000005100.iommu
[    6.305246] rpi-hevc-dec 1000800000.codec: Adding to iommu group 0
[    6.305250] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_attach_dev: MMU 1000005100.iommu
[    6.315732] pispbe 1000880000.pisp_be: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    6.315741] pispbe 1000880000.pisp_be: bcm2712_iommu_probe_device: MMU 1000005100.iommu
[    6.315746] pispbe 1000880000.pisp_be: bcm2712_iommu_device_group: MMU 1000005100.iommu
[    6.315750] pispbe 1000880000.pisp_be: Adding to iommu group 0
[    6.315753] pispbe 1000880000.pisp_be: bcm2712_iommu_attach_dev: MMU 1000005100.iommu
[    6.413860] vc4_hvs 107c580000.hvs: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    6.413869] vc4_hvs 107c580000.hvs: bcm2712_iommu_probe_device: MMU 1000005200.iommu
[    6.413875] vc4_hvs 107c580000.hvs: bcm2712_iommu_device_group: MMU 1000005200.iommu
[    6.413880] vc4_hvs 107c580000.hvs: Adding to iommu group 1
[    6.413883] vc4_hvs 107c580000.hvs: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
[    6.415066] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    6.415073] vc4-drm axi:gpu: bcm2712_iommu_probe_device: MMU 1000005200.iommu
[    6.415077] vc4-drm axi:gpu: bcm2712_iommu_device_group: MMU 1000005200.iommu
[    6.415082] vc4-drm axi:gpu: Adding to iommu group 1
[    6.415084] vc4-drm axi:gpu: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
albrecht@albrecht-rpi5:~ $ 

```

### Additional context

Since the Raspberry Pi 5 is so powerful, it can run VMs seamless. To have a working PCI passthrough makes it possible to add any PCIe device into VMs. In my case it is a 4x Ethernet interface card, but other users would like to add GPUs or NVMe SSDs.

P33M, 2025-05-06:
What happens if you modify the devicetree to contain this patch? Build and install as per the usual instructions in the documentation.
```
diff --git a/arch/arm64/boot/dts/broadcom/bcm2712-rpi-5-b.dts b/arch/arm64/boot/dts/broadcom/bcm2712-rpi-5-b.dts
index 2939214f23ad..dd5e23b1cdf8 100644
--- a/arch/arm64/boot/dts/broadcom/bcm2712-rpi-5-b.dts
+++ b/arch/arm64/boot/dts/broadcom/bcm2712-rpi-5-b.dts
@@ -715,6 +715,7 @@ &hdmi1 {

 &pcie1 {
        brcm,clkreq-mode = "safe";
+       iommus = <&iommu2>;
 };

 &pcie2 {
diff --git a/arch/arm64/boot/dts/broadcom/bcm2712.dtsi b/arch/arm64/boot/dts/broadcom/bcm2712.dtsi
index 58d879004bd0..42fbe7d18274 100644
--- a/arch/arm64/boot/dts/broadcom/bcm2712.dtsi
+++ b/arch/arm64/boot/dts/broadcom/bcm2712.dtsi
@@ -513,7 +513,7 @@ pcie1: pcie@1000110000 {

                        dma-ranges =
                                /* 64GiB, 64-bit, non-prefetchable at PCIe 10_0000_0000 */
-                               <0x03000000 0x10 0x00000000 0x00 0x00000000 0x10 0x00000000>,
+                               <0x03000000 0x00 0x00000000 0x00 0x00000000 0x10 0x00000000>,
                                /* 4KiB, 64-bit, non-prefetchable at PCIe ff_ffff_f000 MIP1 */
                                <0x03000000 0xff 0xfffff000 0x10 0x00131000 0x00 0x00001000>;

```
There are several reasons why this isn't the default - mainly that arbitrary descriptor DMA from a PCIe endpoint will claim many TLB entries and potentially starve out real-time platform devices (ISP backend, HEVC codec).

AlbrechtL, 2025-05-06:
Thanks for responding.

I apply the patch and something changed now. The line `/sys/kernel/iommu_groups/0/devices/1000110000.pcie` is new. But it seems that only the first Ethernet controller `0001:01:00.0` is added.
```
albrecht@albrecht-rpi5:~/src/openwrt-docker $ sudo find /sys/kernel/iommu_groups/ -type l
/sys/kernel/iommu_groups/1/devices/107c580000.hvs
/sys/kernel/iommu_groups/1/devices/axi:gpu
/sys/kernel/iommu_groups/0/devices/1000800000.codec
/sys/kernel/iommu_groups/0/devices/1000880000.pisp_be
/sys/kernel/iommu_groups/0/devices/1000110000.pcie
```

But attaching the device `0001:01:00.0 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)` to `vfio-pci` is still not working.
```
[  421.937240] igb 0001:01:00.0: removed PHC on eth1
[  422.079527] VFIO - User Level meta-driver version: 0.3
[  422.112067] vfio-pci 0001:01:00.0: probe with driver vfio-pci failed with error -22

```

P33M, 2025-05-08:
PCI uses `iommu-map` to describe the relationship between a particular PCI bus master (RequesterID) and how the IOMMU uses this to index into a set of page tables. The hardware in bcm2712 doesn't pay attention to this information, so you cannot virtualise separate functions - they all share the same VA domain.

What you're asking for can't be supported.

closed 2025-05-08 by P33M

996refuse, 2025-05-15:
> PCI uses `iommu-map` to describe the relationship between a particular PCI bus master (RequesterID) and how the IOMMU uses this to index into a set of page tables. The hardware in bcm2712 doesn't [...]

appreciate your explanation, since it's the bcm2712 hardware issue that no one device multiple functions support, what if one device one function device, like NVMe SSD? @P33M 

to go further, as a workaround, we can disable the other functions (0001:01:00.1 0001:01:00.2 0001:01:00.3) first, then bind the vfio driver to 0001:01:00.0 to see if the corresponding port of 0001:01:00.0 is work. @AlbrechtL 

```
echo 0 > /sys/bus/pci/devices/0001:01:00.1/enable
echo 0 > /sys/bus/pci/devices/0001:01:00.2/enable
echo 0 > /sys/bus/pci/devices/0001:01:00.3/enable
```

thanks!