raspberrypi/linux #6840: Segfault in DRM framebuffer initialization with panel driver on Linux 6.13+ (Raspberry Pi CM5 Lite)

libbkmz, 2025-05-07:
### Describe the bug

When using the panel driver ([panel-cwu50.c](https://github.com/ak-rex/ClockworkPi-linux/blob/883135e4c7bde14cf5e59d8373042b15db0e61a2/drivers/gpu/drm/panel/panel-cwu50.c)) on Linux kernel 6.13 and 6.14, the system crashes with a segfault during DRM framebuffer initialization. The same driver works correctly on Linux kernel 6.12.
The crash happens during the framebuffer setup process, specifically in the `__drm_fb_helper_initial_config_and_unlock` and `drm_setup_crtcs_fb` functions.

This is all related to ClockworkPi uConsole device with CM4 adapter board which is handling CM5.

### Steps to reproduce the behaviour

Clone the Linux kernel repository and checkout the rpi-6.13.y branch.

Configure the kernel with the following commands:

make -j16 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2712_defconfig

Enable the following kernel build options:

CONFIG_REGMAP_I2C=y
CONFIG_INPUT_AXP20X_PEK=y
CONFIG_CHARGER_AXP20X=m
CONFIG_BATTERY_AXP20X=m
CONFIG_AXP20X_POWER=m
CONFIG_MFD_AXP20X=y
CONFIG_MFD_AXP20X_I2C=y
CONFIG_REGULATOR_AXP20X=y
CONFIG_AXP20X_ADC=m
CONFIG_TI_ADC081C=m
CONFIG_CRYPTO_LIB_ARC4=y
CONFIG_CRC_CCITT=y

Build the kernel:

make -j16 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image.gz modules dtbs

Deploy the kernel to a Raspberry Pi CM5 Lite and boot the system
Try to load the panel driver module panel-cwu50
Observe the kernel crash during DRM framebuffer initialization.

### Device (s)

Raspberry Pi CM5 Lite

### System

uname -a: 
```
Linux raspberrypi 6.13.12-v8-16k-bkmz1 #5 SMP PREEMPT Wed May 7 14:00:00 UTC 2025 aarch64 GNU/Linux
```

### Logs

Full kernel crash log: 
```
[  200.393257] drm-rp1-dsi 1f00130000.dsi: rp1dsi_host_attach: Attach DSI device name=cwu50 channel=0 lanes=4 format=0 flags=0x5 hs_rate=0 lp_rate=0
[  200.393780] [drm] Initialized drm-rp1-dsi 1.0.0 for 1f00130000.dsi on minor 0
[  200.393820] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000038
[  200.403692] Mem abort info:
[  200.403696]   ESR = 0x0000000096000045
[  200.403698]   EC = 0x25: DABT (current EL), IL = 32 bits
[  200.403701]   SET = 0, FnV = 0
[  200.403702]   EA = 0, S1PTW = 0
[  200.403703]   FSC = 0x05: level 1 translation fault
[  200.403705] Data abort info:
[  200.403706]   ISV = 0, ISS = 0x00000045, ISS2 = 0x00000000
[  200.403708]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[  200.403710]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  200.403712] user pgtable: 16k pages, 47-bit VAs, pgdp=00000000c0ff0000
[  200.403715] [0000000000000038] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  200.403723] Internal error: Oops: 0000000096000045 [#1] PREEMPT SMP
[  200.403731] Modules linked in: panel_cwu50(O+) rfcomm algif_hash algif_skcipher af_alg bnep binfmt_misc cdc_acm joydev brcmfmac_wcc vc4 snd_soc_hdmi_codec spidev brcmfmac snd_soc_core hci_uart btbcm bluetooth snd_pcm_dmaengine cfg80211 snd_pcm rpi_hevc_dec aes_ce_blk pisp_be aes_ce_cipher ghash_ce snd_timer gf128mul snd ecdh_generic sha2_ce sha256_arm64 ecc videobuf2_dma_contig sha1_ce v4l2_mem2mem brcmutil videobuf2_memops rfkill videobuf2_v4l2 v3d videodev libaes drm_display_helper sha1_generic cec raspberrypi_hwmon drm_rp1_dsi videobuf2_common spi_bcm2835 mc gpio_keys drm_shmem_helper dwc2 i2c_designware_platform i2c_brcmstb drm_client_lib gpu_sched i2c_gpio rp1_pio i2c_designware_core drm_dma_helper squashfs i2c_algo_bit drm_kms_helper raspberrypi_gpiomem nvmem_rmem rp1_mailbox rp1_adc rp1 uio_pdrv_genirq uio ocp8178_bl(O) axp20x_usb_power axp20x_battery axp20x_ac_power axp20x_adc ti_adc081c industrialio_triggered_buffer kfifo_buf industrialio uhid drm fuse drm_panel_orientation_quirks backlight dm_mod
[  200.403833]  ip_tables x_tables ipv6
[  200.403841] CPU: 2 UID: 0 PID: 1816 Comm: modprobe Tainted: G           O       6.13.12-v8-16k-bkmz1-bkmz1-bkmz1-bkmz1-bkmz1+ #5
[  200.403846] Tainted: [O]=OOT_MODULE
[  200.403849] Hardware name: Raspberry Pi Compute Module 5 Lite Rev 1.0 (DT)
[  200.403851] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  200.403855] pc : __drm_fb_helper_initial_config_and_unlock+0x2c4/0x520 [drm_kms_helper]
[  200.403902] lr : __drm_fb_helper_initial_config_and_unlock+0x230/0x520 [drm_kms_helper]
[  200.403928] sp : ffffc0008462b740
[  200.403929] x29: ffffc0008462b770 x28: 0000000034325258 x27: ffffc0008462b750
[  200.403934] x26: ffff800006f6b030 x25: ffff800180c83000 x24: ffff800180c83000
[  200.403938] x23: ffffd06f9bffc190 x22: ffff800180c83000 x21: ffffd06f9c0d8db8
[  200.403942] x20: ffff800006f6b000 x19: ffffd06f9bffc000 x18: ffffffffffffffff
[  200.403946] x17: 20726f6e696d206e x16: ffffd06fcf18dce8 x15: ffff8000029166d4
[  200.403950] x14: 0000000000000000 x13: 0000000000000000 x12: ffff800002914900
[  200.403954] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffd06f9c0d1458
[  200.403958] x8 : ffff800002915d00 x7 : ffff800002915c80 x6 : ffff800002914930
[  200.403962] x5 : 00000000000002d0 x4 : 0000000000001000 x3 : 0000000000000001
[  200.403966] x2 : 005d6e6f6362665b x1 : 0000000000000000 x0 : ffff800006f6b000
[  200.403971] Call trace:
[  200.403973]  __drm_fb_helper_initial_config_and_unlock+0x2c4/0x520 [drm_kms_helper] (P)
[  200.403999]  drm_fb_helper_initial_config+0x4c/0x68 [drm_kms_helper]
[  200.404023]  drm_fbdev_client_hotplug+0x84/0xe8 [drm_client_lib]
[  200.404028]  drm_client_register+0x60/0xb0 [drm]
[  200.404122]  drm_fbdev_client_setup+0xac/0x3c98 [drm_client_lib]
[  200.404126]  drm_client_setup+0x20/0x60 [drm_client_lib]
[  200.404129]  rp1dsi_host_attach+0x1c4/0x2148 [drm_rp1_dsi]
[  200.404138]  mipi_dsi_attach+0x38/0x68
[  200.404154]  cwu50_probe+0x120/0x208 [panel_cwu50]
[  200.404164]  mipi_dsi_drv_probe+0x28/0x40
[  200.404170]  really_probe+0xc4/0x2d0
[  200.404178]  __driver_probe_device+0x80/0x130
[  200.404184]  driver_probe_device+0x44/0x168
[  200.404188]  __driver_attach+0x98/0x1b0
[  200.404193]  bus_for_each_dev+0x84/0x100
[  200.404199]  driver_attach+0x2c/0x40
[  200.404203]  bus_add_driver+0xec/0x220
[  200.404207]  driver_register+0x70/0x138
[  200.404212]  mipi_dsi_driver_register_full+0x60/0x78
[  200.404217]  cwu50_driver_init+0x28/0x3ff8 [panel_cwu50]
[  200.404224]  do_one_initcall+0x60/0x2a0
[  200.404233]  do_init_module+0x5c/0x230
[  200.404238]  load_module+0x187c/0x1e60
[  200.404242]  __do_sys_init_module+0x150/0x200
[  200.404246]  __arm64_sys_init_module+0x24/0x40
[  200.404249]  invoke_syscall+0x50/0x120
[  200.404253]  el0_svc_common.constprop.0+0x48/0xf0
[  200.404257]  do_el0_svc+0x24/0x38
[  200.404260]  el0_svc+0x30/0xd0
[  200.404267]  el0t_64_sync_handler+0x10c/0x138
[  200.404272]  el0t_64_sync+0x198/0x1a0
[  200.404279] Code: aa1403e0 f2cdcde2 f0fff953 f2e00ba2 (f9001c22) 
[  200.404281] ---[ end trace 0000000000000000 ]---
```

__drm_fb_helper_initial_config_and_unlock+0x2c4/0x520 represents this line: 
https://github.com/raspberrypi/linux/blob/051386c64d108aa96f397928e11bd1cd7293a93a/drivers/gpu/drm/drm_fb_helper.c#L1645 

```
1645:    strcpy(fb_helper->fb->comm, "[fbcon]");
```

### Additional context

After commenting strcpy function call, I observe segfault in that line: https://github.com/raspberrypi/linux/blob/051386c64d108aa96f397928e11bd1cd7293a93a/drivers/gpu/drm/drm_fb_helper.c#L1778

```
			info->var.width = connector->display_info.width_mm;
```

which a little bit weird... 

Also, I tried this driver for panel-cwu50: https://github.com/qkdxorjs1002/linux/blob/f0595b7f739b3c54e69225a65bb4a4ea76cb4139/drivers/gpu/drm/panel/panel-cwu50.c

with the same result =\

6by9, 2025-05-08:
I would normally say that reporting issues with a panel driver that isn't in this kernel tree is out of scope for us to investigate.

However there appears to be a problem more generally with the RP1 DSI driver that causes this, as it fails with the Pi Touch Display 2 as well. I'm investigating.
I am aware that upstream reworked a load of the fb helper code around that time, so it's likely to be linked to that. The equivalent patch for vc4 was https://github.com/torvalds/linux/commit/45903624e9fc57e38eb7f023717205cce2d5e4a3

6by9, 2025-05-08:
#6841 should resolve this.

6.13 is EOL already as 6.14 is released, so I don't propose backporting it.

pelwell, 2025-05-08:
#6841 is now merged, but rpi-6.13.y is obsolete.

PR #6841 (2025-05-08): rpi-6.14: Add DRM default client setup to RP1 DRM devices
Fixes #6840