raspberrypi/linux #6847: EGLFS appllication (Falkon) fails to run on a CM5 (v6.12.25)

JinShil, 2025-05-12:
### Describe the bug

Running `falkon -platform eglfs` on a CM5 fails, but runs fine on a CM4.

The error messages printed to the terminal are:
```
$ falkon -platform eglfs --fullscreen https://raspberrypi.com
drmModeGetResources failed (Operation not supported)
no screens available, assuming 24-bit color
Falkon: 1 extensions loaded
Fatal: Cannot create window: no screens available
Aborted
```

I do not notice anything out of the ordinary printed to dmesg or in the journal.  Given that, I would have assumed this is an application or dependency problem, but the fact that the same thing works on a CM4 tells me this may be a kernel problem.

### Steps to reproduce the behaviour

1. Install the Falkon browser with `sudo apt install falkon`
2. Configure OS to boot to a console
3. Boot to the console
4. Run `falkon -platform eglfs --fullscreen https://raspberrypi.com`
5. Observe errors printed to the terminal.

Follow the same procedure on a CM4 and notice it works fine.

### Device (s)

Raspberry Pi CM5 Lite

### System

```
$cat /etc/rpi-issue
Raspberry Pi reference 2025-05-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 49e1078c1b07ff4948d304e0149d3bc698e0006a, stage4
```
```
$ vcgencmd version
2025/03/19 13:41:26
Copyright (c) 2012 Broadcom
version cec1d3ae (release) (embedded)
```
```
$ uname -a
Linux comfilepi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

### Logs

No anomalies observed in the logs

popcornmix, 2025-05-12:
Are you booting from sdcard or emmc.
If sdcard are you testing exactly the same sdcard?
If not are you sure both systems are full updated (sudo apt update && sudo apt dist-upgrade).

Do you get the same versions for the three commands you reported on the CM4 too?

pelwell, 2025-05-12:
In what way is this a kernel issue?

XECDesign, 2025-05-12:
I believe it's a generic Qt issue. Maybe try the suggestion [here](https://forums.raspberrypi.com/viewtopic.php?t=287800#p1742985).

But yeah, I don't think this is a kernel or even a Pi issue at all.

JinShil, 2025-05-12:
> Are you booting from sdcard or emmc.

SD Card

> If sdcard are you testing exactly the same sdcard?

Yes

> If not are you sure both systems are full updated (sudo apt update && sudo apt dist-upgrade).

I just downloaded the OS today and updated it immediately after the first boot.

> Do you get the same versions for the three commands you reported on the CM4 too?

I need to run the commands again when I get back to the office, but given that it is the same SD card, I believe only the `uname -a` returns a different value given that the kernels are different for the CM4 and CM5.

> In what way is this a kernel issue?

I believe this is a kernel issue because it is the only difference between the two subjects (CM4 and CM5), other than the compute module itself. 

The above statements are made from memory, but I will confirm tomorrow.

JinShil, 2025-05-12:
> I believe it's a generic Qt issue. Maybe try the suggestion [here](https://forums.raspberrypi.com/viewtopic.php?t=287800#p1742985).

That solution looks promising, but if that is indeed the solution, it begs the question why that is needed.  Is the selection of `card0` or `card1` not deterministic?

popcornmix, 2025-05-12:
> That solution looks promising, but if that is indeed the solution, it begs the question why that is needed. Is the selection of card0 or card1 not deterministic?

Correct. Pretty much all kernel device nodes do not guarantee the numbering. It depends on the order they are probed. And as kernel boot is multi-threaded it may differ from one boot to another.

User code should never hard code which device node to use - it should iterate over then and pick the node with the appropriate properties.

The linked solution may work, but is not a robust long term fix. It sounds like the fix should go in eglfs (whatever code it is that tried to open dev/dri/card0 or dev/dri/card1.

closed 2025-05-13 by JinShil