raspberrypi/linux #6849: USB issues (renesas UPD720201) since kernel 6.12

houkhouk, 2025-05-13:
### Describe the bug

I am using a custom board for my Rpi CM5. USB3 connections are provided through Renesas UPD720201 and PCI bus. The firmware of teh chip is the latest provided by Renesas: K2026090.mem

I had no issues when using Kernel 6.6. Since update to 6.12, the xhci-pi-renesas kernel module seems unstable. It either won't work, or only for a few seconds after boot. 

lspci is consistent and outputs:
```
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30)
0001:01:00.0 USB controller: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller (rev 03)
0002:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30)
0002:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

It is sometimes possible to revive the usb ports temporarily by suppressing the pci peripheral and rescanning
```
echo 1 > /sys/bus/pci/devices/0001:01:00.0/remove 
echo 1 > /sys/bus/pci/rescan
```

dmesg output is provided under. 

Maybe similar to https://github.com/raspberrypi/linux/issues/6755 ?

### Steps to reproduce the behaviour

I am unsure whether the problem is related to my hardware (Renesas UPD720201) or if there is an issue with the xhci management. Reproducing might require the use the same USB integrated circuit.

### Device (s)

Raspberry Pi CM5

### System

```
cat /etc/rpi-issue
Raspberry Pi reference 2025-05-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 49e1078c1b07ff4948d304e0149d3bc698e0006a, stage4
```
```
vcgencmd version
2025/04/08 00:16:47 
Copyright (c) 2012 Broadcom
version 50beecf0 (release) (embedded)
```

```
uname -a
Linux raspberrypi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

### Logs

```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.25+rpt-rpi-2712 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 5 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000003a000000..0x000000003dffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165c0..0x000000003fd16629 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000003fffffff] (1024MB)
[    0.000000] Faking node 1 at [mem 0x0000000040000000-0x000000007fffffff] (1024MB)
[    0.000000] Faking node 2 at [mem 0x0000000080000000-0x00000000bfffffff] (1024MB)
[... 628 lines cut ...]
[    3.296299] usb usb7: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.12
[    3.296301] usb usb7: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.296303] usb usb7: Product: xHCI Host Controller
[    3.296305] usb usb7: Manufacturer: Linux 6.12.25+rpt-rpi-2712 xhci-hcd
[    3.296306] usb usb7: SerialNumber: 0001:01:00.0
[    3.296396] hub 7-0:1.0: USB hub found
[    3.296407] hub 7-0:1.0: 4 ports detected
[    3.574741] Bluetooth: RFCOMM TTY layer initialized
[    3.574754] Bluetooth: RFCOMM socket layer initialized
[    3.574765] Bluetooth: RFCOMM ver 1.11
[    3.577742] usb 6-1: new low-speed USB device number 2 using xhci-pci-renesas
[    3.753859] usb 6-1: New USB device found, idVendor=046d, idProduct=c31c, bcdDevice=64.02
[    3.753866] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.753869] usb 6-1: Product: USB Keyboard
[    3.753870] usb 6-1: Manufacturer: Logitech
[    3.791761] usb 6-1: ERROR: Endpoint drop flag = 0, add flag = 1, and endpoint is not disabled.
[    3.791769] xhci-pci-renesas 0001:01:00.0: xhci_fixup_endpoint: Configure endpoint failed: -22
[    3.791856] input: Logitech USB Keyboard as /devices/platform/axi/1000110000.pcie/pci0001:00/0001:00:00.0/0001:01:00.0/usb6/6-1/6-1:1.0/0003:046D:C31C.0001/input/input5
[    3.960551] hid-generic 0003:046D:C31C.0001: input,hidraw0: USB HID v1.10 Keyboard [Logitech USB Keyboard] on usb-0001:01:00.0-1/input0
[    5.855608] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
[   14.301697] xhci-pci-renesas 0001:01:00.0: xHCI host not responding to stop endpoint command
[   14.301709] xhci-pci-renesas 0001:01:00.0: xHCI host controller not responding, assume dead
[   14.301720] xhci-pci-renesas 0001:01:00.0: HC died; cleaning up
[   14.301830] usb 6-1: USB disconnect, device number 2
```

P33M, 2025-05-13:
I have a uPD720202 expansion board and don't get a controller failure. I've copied said firmware file to /lib/firmware/renesas_usb_firmware.mem (my board has an EEPROM but the kernel preferentially loads the mem file). The error message for xhci_fixup_endpoint is spurious - add and drop bits are both set to 1, the xhc just doesn't like the command for whatever reason.

What happens if you add `pcie_aspm=off` to /boot/firmware/cmdline.txt?

houkhouk, 2025-05-13:
Adding `pcie_aspm=off` to /boot/firmware/cmdline.txt makes everything work again, thanks !

I am not entirely sure to understand what is happening, ASPM seems to manage the power supply when a device is marked as active. So having both add and drop bits set to confused the driver on whether it should power the device ?

P33M, 2025-05-13:
No, the endpoint configure message is unrelated to the controller failure.

Between 6.6. and 6.12, we swapped to an upstream version of the pcie-brcmstb driver for bcm2712.

One part of the driver that changed was around how the CLKREQ# line was handled. Downstream had a heuristic to detect whether or not to enable CLKREQ# (permitting ClockPM or L1 sub-states), upstream has a DT property that specifies the behaviour. On CM4/5 it's generally assumed that board designers implement the interconnect correctly, i.e. connect this signal up.

One or both of your M.2 to edge-connector adapter/uPD boards don't connect it.

houkhouk, 2025-05-13:
Ok, thanks for the explanation. The CLKREQ signal is actually connected to pin 102 (PCIE_CLK_nREQ) of my CM5. I'll investigate, maybe something went wrong there, or missing a pull-up/pull-down resistor. 

I'll mark the issue as resolved since the issue seems to be hardware and not software.

closed 2025-05-13 by houkhouk