raspberrypi/linux #6853: Need compilation for 2 drivers turned on and updates to firmware

morrownr, 2025-05-14:
### Describe the bug

Describe the bug

I am adding 4 very closely related issues and the fixes:

```
Issue #1: Support for Mediatek (MTK) bluetooth is not compiling. I
am testing two USB WiFi adapters that support wifi and bluetooth
and need this support as bluetooth will not work without it.

From .config

#
# Bluetooth device drivers
#
CONFIG_BT_INTEL=m
CONFIG_BT_BCM=m
CONFIG_BT_RTL=m
CONFIG_BT_HCIBTUSB=m
# CONFIG_BT_HCIBTUSB_AUTOSUSPEND is not set
CONFIG_BT_HCIBTUSB_POLL_SYNC=y
CONFIG_BT_HCIBTUSB_BCM=y
# CONFIG_BT_HCIBTUSB_MTK is not set
CONFIG_BT_HCIBTUSB_RTL=y

Corrective action:

Change "# CONFIG_BT_HCIBTUSB_MTK is not set"
to     "CONFIG_BT_HCIBTUSB_MTK=y"

Issue #2: Firmware files for the mt7921u driver are very old and need to
be updated. I have a guide that helps with issue.

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

The following shows the date of the very old firmware files:

$ ethtool -i wlan0
driver: mt7921u
version: 6.12.25+rpt-rpi-v8
firmware-version: ____010000-20240219111038

Issue #3: Support for Mediatek MT7925U is not compiling but needs to
be compiling.

From .config

Change "# CONFIG_MT7925U is not set"
to "CONFIG_MT7925U=m"

Issue #4: Mediatek MT7925 firmware files need to be updated.

See Issue #2 for the guide that will walk you through how to update these firmware
files to the latest version.
```

### Steps to reproduce the behaviour

The behavior doesn't need to be reproduced as the fixes are included above

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2025-05-06
Apr 30 2025 13:33:39
6.12.25+rpt-rpi-v8

### Logs

Not needed. This is straightforward set of related issues.

### Additional context

Your support is appreicated.

pelwell, 2025-05-14:
I don't mind enabling a few more modules (unless they are enormous), but the firmware updates are more problematic. We get most of our firmware files from the Debian firmware packages, in this case it would be firmware-mediatek. We are currently shipping the most up-to-date version of that package, but the version is trixie is more recent - the package is currently dated 20241210-1+rpt2, and strings in the files suggests they are from 2024/11/06.

morrownr, 2025-05-14:
Hi @pelwell 

Thanks for enabling the 2 modules. They are definitely needed:

The below module is needed for the Alfa AXML and AXM USB WiFi adapters that are popular:

CONFIG_BT_HCIBTUSB_MTK=y

The below module is needed for several adapters that are in testing and will be on the market very soon:

CONFIG_MT7925U=m

I run the github.com/morrownr/USB-WiFi site. I get reports from Linux USB WiFi users from around the world and reports for the 4 issues here are building.

Regarding the firmware files. I use Mint (LMDE) as my daily driver. I also use Debian. Debian is a very good distro but timely firmware updates for WiFi devices has become more important for WiFi 6/7 devices as the complexity of the drivers and firmware has led to the need, at least over the last 2 years, for more frequent firmware updates. This is true for all WiFi device makers and not just Mediatek, which is the subject here.

Regarding the mt7921 firmware:

> the version is trixie is more recent - the package is currently dated 20241210-1+rpt2, and strings in the files suggests they are from 2024/11/06.

That is the firmware we need now. The version currently in RasPiOS is several versions old and has several bugs that are causing problems.

Regarding the mt7925 firmware:

The latest firmware was logged into the official kernel site on 2025-04-30:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek/mt7925

We need that new mt7925 firmware as it fixes some serious bugs. I am capable of updating firmware on my own but many RasPiOS users do not, even if a guide is provided, so I am speaking for them when I mention this need. Maybe the need to speed up the process can be brought to the attention of the Debian maintainer? 

Your help is greatly appreciated.

pelwell, 2025-05-15:
The additional drivers are enabled by #6857. If you wait about half an hour you will be able to install a trial version using `sudo rpi-update pulls/6857`, backing up any important data first.

I'm not sure about hosting more third party firmwares in our firmware repo though - it's more packaging work and maintenance. I think you would be better off writing a simple command-line tool that scrapes the Linux kernel firmware repo for the drivers your users want. It could even automatically detect those which are in use on a user's device.

morrownr, 2025-05-15:
@pelwell 

> The additional drivers are enabled by #6857.

This is very much appreciated. Thank you.

> I'm not sure about hosting more third party firmwares in our firmware repo though...

I am not asking for more firmwares to be hosted, I am respectfully asking that the right ones be hosted. If my understanding of how  firmwares are selected for RasPiOS is correct, the firmware packages from Debian stable are used. Back in the days of WiFi 4 and WiFi 5, things were somewhat more simple and bugs in firmware were less common. Those days are long gone.  If my memory serves me correctly, the firmware files for the mt7921u driver have been updated 9 times in the last 2 years and 5 times since the version of the firmware that is shipping with the current RasPiOS. The new firmware files represent bug fixes. Is the open source part of wifi drivers left static? No. regular patches go into supported kernels that then flow into RasPiOS and on to users as kernel updates. 

The best short term fix that should cause the smallest amount of change is to use Debian testing for firmware-mediatek. As you noted, the files in firmware-mediatek in trixie are much more current. Once Trixie is released then moving to use the firmware-mediatek from the next testing can be done. Ultimately, a better system needs to be used... such as a script that scrapes the newest on a weekly basis so that they can flow out to users automatically.

> I think you would be better off writing a simple command-line tool that scrapes the Linux kernel firmware repo for the drivers your users want.

This is not about my users. This is about RasPiOS users and their ability to install an OS that is as trouble free as possible. I can write a script and I am willing to do so if it is used by RasPiOS to keep its wifi firmware current in its dev environment. 

This is an issue that has come about due to the increased complexity of wireless drivers and firmware with WiFi 6 and 7. Right now the issue is most apparent with Mediatek because they are at forefront of having chips and drivers that are Linux Wireless Standards (mac80211) compliant and the drivers/firmware are full featured and in demand. By the time RasPiOS is updated to the LTS kernel that will be released later this year, this is going to be an issue with Realtek wireless drivers. It is an issue with Intel and Qualcomm as well but those two companies do not produce USB chips so it is not something that comes up as much with RasPi hardware.

Remember that wifi drivers consist of 2 parts: The open source part that is in the kernel and the closed source part that is in the firmware files. The open source part is regularly updated and those updates show up in RasPiOS. The firmware files are also regularly updated but as it stands right now, there is no system to keep them updated and that leads to user problems.

Sorry for the length of the post. Thanks for your help.

Regards,

@morrownr

morrownr, 2025-05-31:
@eatingmangoes 

> The card works on the more recent kernel versions on a normal x86 laptop normally.

This may indicate that the problem is with the M.2 hat. It could be a driver error or one of many things.

>  (connected using a m.2 hat)

I am not familiar with the M.2 hat or any of the details of using it with a RasPi. Have you used this hat successfully with other cards?

eatingmangoes, 2025-05-31:
> @eatingmangoes
>
>
> This may indicate that the problem is with the M.2 hat. It could be a driver error or one of many things.
>
>
> I am not familiar with the M.2 hat or any of the details of using it [...]

The hat does work fine with kernel 6.11 and under with the card.