raspberrypi/linux #6856: lanes parameter missing for a few ili9881c_desc like nwe080

berkutta, 2025-05-15:
### Describe the bug

On rpi-6.6.y the nwe080 description have the lanes parameter set to 4:
https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L2044-L2050

On rpi-6.12.y, this parameters is missing:
https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L2506-L2511

Without lanes being specified, the display doesn't work.

### Steps to reproduce the behaviour

- rpi-6.6.y working
- rpi-6.12.y not working
- rpi-6.6.y with added lanes working

### Device (s)

Raspberry Pi CM4

### System

Raspbian

6by9, 2025-08-04:
Sorry, I missed this one when it was first opened.

Whilst Raspberry Pi accept PRs for third party hardware (nwe080 was added for CutiePi https://github.com/raspberrypi/linux/pull/4742) and will do the basic forward porting of the drivers, we do NOT take on full responsibility for maintaining or necessarily testing it (in many cases we don't have hardware to test with).
If there are major issues in forward porting to a new kernel version, then drivers will get dropped and kicked back to the original contributor to fix.

PRs to fix up forward porting errors are welcome, as is upstreaming of patches (there's no reason nwe080 couldn't go upstream).

jack-obrien, 2025-08-05:
Yes, I have tested on my panel that the lanes parameter is still needed. Will get a PR ready soon!