raspberrypi/linux #6864: Audio stuttering and crackling when transferring data through pipe on Raspberry Pi OS Bookworm

zxmak, 2025-05-21:
### Describe the bug

Audio playback is intermittently interrupted by stuttering and crackling noises when a data stream is transferred through a pipe using standard Linux utilities. The issue is especially noticeable when performing data-heavy operations such as piping ```/dev/zero``` through ```pv``` to ```/dev/null```, and appears to be related to resource contention or scheduling.

Additional Information:
-    The issue was first noticed during debugging of a custom application that processes data from a pipe.
-    It was later confirmed to occur using only standard tools (cat, pv, mpg123).
-    The CPU load remains relatively low when the issue occurs.
-    Appears to be either a scheduler latency issue, audio buffer underrun, or an I/O-related bottleneck affecting audio playback.

### Steps to reproduce the behaviour

1) Open a terminal and start audio playback (e.g., internet radio):
```
mpg123 -f 6554 https://strm112.1.fm/onelive_mobile_mp3
```

2) In another terminal, run:
```
cat /dev/zero | pv -r > /dev/null
```

3) Optionally, open Chromium and navigate to a simple web page, for example https://www.google.com to increase likelihood of issue manifesting.

**Expected Behavior:** 
Audio should continue playing smoothly without interruptions, regardless of background data transfers or other light system activity.

**Actual Behavior:**
-    Audio crackles intermittently or stutters heavily.
-    In some cases, the audio becomes distorted and jittery immediately; in others, only mild crackling is noticed after several seconds.
-    Severity varies based on system state and seems to be influenced by non-deterministic factors.

**Note:** The issue does not always manifest immediately and may require restarting the ```cat /dev/zero | pv -r > /dev/null``` command multiple times under different conditions. I have not identified a clear pattern or trigger, but once the audio disruption begins, it tends to persist consistently until the command ```cat /dev/zero | pv -r > /dev/null``` is restarted. Interestingly, restarting the same command - under seemingly identical system conditions (no change in background processes or overall system load), can result in normal, uninterrupted audio playback.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

vcgencmd version
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)

uname -a
Linux zx80 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

### Logs

No relevant error messages appear in the dmesg log when the issue occurs.

6by9, 2025-05-21:
Are you using HDMI audio or the headphone jack? The two involve processing at different levels (ALSA for HDMI and the firmware for the headphone jack).

zxmak, 2025-05-21:
I'm using USB DAC

pelwell, 2025-05-21:
Synchronous or asynchronous? With what kind of quality settings?

zxmak, 2025-05-22:
> Synchronous or asynchronous? 

          Transfer Type            Isochronous
          Synch Type               Synchronous

> With what kind of quality settings?

    Packet Size = 72
    Momentary freq = 48000 Hz (0x6.0000)

    Format: S32_LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (SYNC)
    Rates: 44100, 48000, 96000, 192000, 384000
    Data packet interval: 125 us
    Bits: 32
    Channel map: FL FR

pelwell, 2025-05-22:
I don't know what kind of latency tolerance that give you, but obviously not enough. Also, 32-bit samples are a bit excessive, but less so than 24-bit stereo at 384kHz.

If you aren't actively using the initramfs (e.g. to load an exotic network interface driver before using an NFS rootfs), you might want to try one of the new -rt (realtime) kernel builds (althought apart from booting it and seeing that normal things seem to work, I have no real experience of it).

To install the RT kernel:
```
$ sudo WANT_32BIT=0 WANT_64BIT=0 WANT_PI4=0 WANT_PI5=0 WANT_64BIT_RT=1 rpi-update rpi-6.12.y
```
Note that you will need to specify the branch name (`rpi-6.12.y`) - the normal rpi-update releases don't currently include the -rt builds. Also, back up any important data first - this is beta software.

You can switch to the new kernel using "kernel=kernel8_rt.img". And, if something goes horribly wrong, you should be able to revert to the previous kernel by commenting out or deleting that line.

pelwell, 2025-05-22:
And that's with the default 64-bit kernel.

P33M, 2025-05-22:
@pelwell does the DAC also work fine under those conditions if plugged into one of the USB2.0 ports?

@zxmak please post a full `sudo lsusb -v` for your DAC.

pelwell, 2025-05-22:
Yes.