raspberrypi/linux #6866: edt-ft5x06 display identification

JanKehren, 2025-05-21:
### Describe the bug

When using the edt-ft5x06 driver for the official Raspberry Pi Touchscreen, often the firmware doesn't put proper values in the identification registers. This leads to having random names for the device as the name will be "generic ft5x06 (%02x)" where %02x will be the value of the register. 
This is an issue under Android as the OS will use the device name to identify and set up input configuration. 

### Steps to reproduce the behaviour

Connect a touchscreen with a bad value in the identification registers.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 5

### System

Android OS

### Additional context

If this is not relevant for other OS's I am happy to resolve this on my end otherwise I can provide a patch for this

6by9, 2025-05-21:
The Pi firmware doesn't write that register at all.
It's defined in the datasheet as "CTPM Vendor ID" and is read only. It should be written into the panel configuration by the display vendor during manufacture.

Raspberry Pi displays will always report it as either 0x00 or 0x79.
Third party manufacturers who have cloned the Pi display may return other values.

Raspberry Pi OS handles either of those within the default config for [Wayfire](https://github.com/raspberrypi-ui/raspberrypi-ui-mods/blob/master/etc/wayfire/template.ini#L25-L47) or [labwc](https://github.com/raspberrypi-ui/raspberrypi-ui-mods/blob/master/etc/xdg/labwc/rc.xml#L161-L168) (there was a matching script for X11, but that is now deprecated).

JanKehren, 2025-05-21:
Ah okay, I guess then this issue could be closed. Thank you for the quick answer.

closed 2025-05-21 by 6by9