raspberrypi/linux #6871: Hyperpixel 4.0 black screen after boot since 6.12.30

LaaZa, 2025-05-27:
### Describe the bug

While not the same as #6721 basically the same setup still.

Hyperpixel 4.0 display goes to black screen after boot.(boot text is seen) The display is detected by the OS and otherwise works including touch but no picture, backlight is still on. This happens on 6.12.30, I'm not sure about .29 but .28 works as normal and downgrading to it fixes the issue. 

### Steps to reproduce the behaviour

1. Rpi 4(00) on Arch based OS or something with 6.12.30.
2. Boot with Hyperpixel 4.0 attached
3. See boot text scroll on the display
4. At certain point in the boot process the display goes black and will not show any picture but still has backlight and is detected.

### Device (s)

Raspberry Pi 400

### System

- EndeavourOS
- Firmware 17084b403fb60475b8ee2641c2641c26049a7d54bf153 (clean) (release) (start)
- kernel 6.12.30

### Additional context

Updated from 6.12.28(works) to 6.12.30(broken) so 6.12.29 status is unknown.

6by9, 2025-05-29:
Debug avenues are limited as it is working fine with Raspberry Pi OS with 6.12.30 from rpi-6.12.y.
I've had a quick look through the kernel commit logs, and there is nothing obvious between 6.12.28 and 6.12.30 that would affect the HyperPixel.

Do you know the commit hash that your 6.12.30 kernel has been built from? I can try recreating that build, but that's really the end of the line as we don't officially support Arch or EndeavourOS.

LaaZa, 2025-05-29:
Should be 1ec873f 

https://github.com/archlinuxarm/PKGBUILDs/commit/8ba59eb9ef84617ddc2bd8fa32d47c1e03594775

And it's fine, I appreciate taking a look at this. It's not as big deal as the previous issue. I wonder if any of those flags in the PKGBUILD cause it.

charel, 2025-06-01:
> Raspberry Pi OS with the kernel from rpi-update (6.12.30 Fri May 23 13:29:06) has just booted fine with a Pi4 and HyperPixel4 (480x800) panel configured.

I've tried this on Raspberry Pi OS with kernel 6.12.30 from rpi-update and the screen boots blank (not even boot messages) on my cm5 with a hyperpixel 4.0 square screen. 
I'm about to recreate the setup to test it further and collect the logs/information asked in this issue. I can also test on a rpi5 with the same screen.

LaaZa, 2025-06-01:
I should perhaps say that for me kmsprint did show the display. And it is definitely detected by the system. I can get the picture to show after seemingly random toggling between ttys with ctrl+alt+f#, it shows at f1 obviously, but others don't show anything before it starts working. It doesn't seem to be strictly time based but just randomly at some point it will work but not by itself.

charel, 2025-06-01:
I tried changing tty, but the display was never showing anything.