raspberrypi/linux #6872: complete kernel hung on starting GUI app: kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000

zxmak, 2025-05-27:
### Describe the bug

I recently encountered an unexpected crash of the Octave GUI, which I documented in [this issue](https://github.com/raspberrypi/bookworm-feedback/issues/417). After some investigation, I found that the problem was caused by the contents of the file ```~/.local/share/octave/history```. In that case, simply deleting the file resolved the issue. However, I wanted to understand what specifically in the file caused the crash, so I began inspecting its contents.

It turns out the problem was triggered by a very long line - approximately 28'000 characters, representing an array initialization. I then tried reducing the line length to determine the threshold at which the issue occurs. I found that Octave GUI crashes somewhere between 4'000 and 5'000 characters. 

More critically, while testing within that range, I encountered an even more severe issue: placing a history file with a line of a certain length causes a complete system hang, requiring a hard reboot. 

The kernel completely hang, GUI stops to respond and the system didn't respond even on network connections. After reboot you can find 4 log records with error before kernel hang:

journalctl log:
```
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
```

I can stable reproduce this issue simply by placing the attached ```history``` file to ```~/.local/share/octave/``` folder and launching Octave GUI.

I suspect there is some issue with V3D video driver and/or possibly Qt5 libraries, probably some kind of security issue with buffer overflow.

The issue has been reproduced on Raspberry Pi OS Bookworm (with all latest APT updates), running under the Wayland display server with the Labwc compositor. Based on testing for mentioned issue with octave crash, the same behavior is also observed when using the Wayfire compositor under Wayland. When running under X11, the behavior differs slightly but still results in problematic behavior.

### Steps to reproduce the behaviour

1) Build and install octave

2) unzip history file from attached ZIP archive to the folder ```~/.local/share/octave/```:

[history.zip](https://github.com/user-attachments/files/20454289/history.zip)

3) Launch Octave GUI (from start menu, or just with terminal ```octave --gui```)

**Expected result:** The Linux kernel continue running with no issue

**Actual result:** The Linux kernel completely hang and stops to work.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)

$ uname -a
Linux rzx 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

### Logs

```
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
```

### Additional context

I can stable reproduce this issue on Octave 10.1.0 build from [official sources](https://ftp.gnu.org/gnu/octave/). But since original issue also was reproduced on Octave 9.4.0, I think it should be reproducible on 9.4.0 and older versions of octave either, but I didn't tested it with older versions.

you can build and install octave in the following way:
```
# prerequisites
sudo apt-get install gcc g++ gfortran make libopenblas-dev liblapack-dev libpcre3-dev libarpack2-dev libcurl4-gnutls-dev epstool libfftw3-dev fig2dev libfltk1.3-dev libfontconfig1-dev libfreetype6-dev libgl2ps-dev libglpk-dev libreadline-dev gnuplot libgraphicsmagick++1-dev libhdf5-dev libsndfile1-dev llvm-dev texinfo libgl1-mesa-dev libosmesa6-dev pstoedit portaudio19-dev libjack-jackd2-dev libqhull-dev libqrupdate-dev libqt5core5a qtbase5-dev qttools5-dev qttools5-dev-tools libqscintilla2-qt5-dev libsuitesparse-dev texlive texlive-latex-extra libxft-dev zlib1g-dev autoconf automake bison flex gperf gzip icoutils librsvg2-bin libtool perl rsync tar libsundials-dev git rapidjson-dev
sudo apt-get install openjdk-17-jdk

# download source, build and install
wget https://ftp.gnu.org/gnu/octave/octave-10.1.0.tar.gz
tar -xzf octave-10.1.0.tar.gz 
cd octave-10.1.0

mkdir .build && cd .build && ./../configure
make -j2
sudo make install
```

I may miss some prerequisite library, so if something is missing just add.

popcornmix, 2025-05-28:
Yeah - I got the same result.
Tested Octave from apt and custom built one.
Tested apt kernel (6.12.25), latest 6.12 and 6.15.
Tested on Pi4 and Pi5.

If the large history file is present Octave crashes on launch. Without the history file it loads okay.
No messages in dmesg, and system still usable.

zxmak, 2025-05-28:
Hmmm...

when history contains line more than 5000 letters - yes, it just crash the Octave, but the kernel still working ok.

But with attached history file it leads to stable kernel crash on my machine. Hang to be more clear - all processes are stopped and kernel doesn't respond, the system don't respond even on network pings. 

And stable 4 log messages in the log before hang:
```
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:10 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* Resetting GPU for hang.
May 27 11:59:11 rzx kernel: v3d fec00000.v3d: [drm:v3d_reset [v3d]] *ERROR* V3D_ERR_STAT: 0x00001000
```

zxmak, 2025-05-29:
I performed a clean installation of the latest image using the default settings and confirmed that it results in just an Octave crash, as previously noted by @popcornmix. To further investigate, I reapplied my typical configuration and customizations incrementally, testing each step to identify the cause of the kernel crash.

Through this process, I determined that the issue is triggered by installing the Consolas font in the ```~/.fonts``` directory. Once the font is installed, launching Octave with a specific history file consistently causes a kernel crash. If the font file is removed, launching Octave results in a regular application crash only - the Linux kernel remains unaffected and continues to operate normally.

I attached the font file, which should be extracted to the ```~/.fonts``` folder to reproduce the kernel crash.

So, steps to reproduce with clean latest Raspberry Pi OS arrch64 install:

1) Build and install octave 10.1.0 (see original issue for details)

2) Unzip font file from attached archive to ```~/.fonts``` folder

[fonts.zip](https://github.com/user-attachments/files/20510984/fonts.zip)

3) Unzip history file to ```~/.local/share/octave/``` (see archive in original issue steps-to-reproduce)

4) Remove octave ini file: ```rm ~/.config/octave/octave-gui.ini ~/.config/octave/octave-gui.ini.lock```

5) Start ```octave --gui```

I have omitted some additional settings that I believe are unrelated to the issue. However, if you're still unable to reproduce the problem, please let me know, and I will review those settings to determine if any of them might also contribute to the kernel crash.

At a glance it appears to be a GPU driver-related issue that occurs under specific conditions and results in a complete kernel crash.

zxmak, 2025-05-29:
I performed a clean installation of the latest Raspberry Pi OS image using the default settings and confirmed that it results in just an Octave crash, as previously noted by @popcornmix. To further investigate, I reapplied my typical configuration and customizations incrementally, testing each step to identify the cause of the kernel crash.

Through this process, I determined that the issue is triggered by installing the Consolas font in the ```~/.fonts``` directory. Once the font is installed, launching Octave with a specific history file consistently causes a kernel crash. If the font file is removed, launching Octave results in a regular application crash only - the Linux kernel remains unaffected and continues to operate normally.

But since its not open source font, I'm not sure if its allowed to share it. As I read it is available for download and use with different products and packages, but as I understand not allowed to redistribute.

So, steps to reproduce with clean latest Raspberry Pi OS arrch64 install:

1) Build and install octave 10.1.0 (see original issue for details)

2) Copy Consolas font file to ```~/.local/share/octave/``` folder, the file md5sum is fe4a6c771135c7d60c684b564466e82a.

3) Unzip history file to ```~/.local/share/octave/``` (see archive in original issue steps-to-reproduce)

4) Remove octave ini file: ```rm ~/.config/octave/octave-gui.ini ~/.config/octave/octave-gui.ini.lock```

5) Start ```octave --gui```

I have omitted some additional settings that I believe are unrelated to the issue. However, if you're still unable to reproduce the problem, please let me know, and I will review those settings to determine if any of them might also contribute to the kernel crash.

At a glance it appears to be a GPU driver-related issue that occurs under specific conditions and results in a complete kernel crash.

zxmak, 2025-06-03:
you need this file to reproduce kernel crash:
```
$ md5sum consola.ttf 
fe4a6c771135c7d60c684b564466e82a  consola.ttf

$ sha256sum consola.ttf 
45c14a49e0ba2edc00b72afad9a930cad5c1b9a93323b239a8c308efc5a65e8e  consola.ttf

$ crc32 consola.ttf 
cf7d826f
```

As I remember it was distributed from MS official site as separate product, but now I don't see such link. It also included in many other products, but I'm not sure if its allowed to share it.

AlexGx, 2026-02-27:
I’m getting a similar error on embedded Linux (Pi 4B)

$kmsg BUG: scheduling while atomic: irq/55-v3d/349/0x00000002
$kmsg BUG: scheduling while atomic: irq/55-v3d/349/0x00000000
$kmsg v3d fec00000.v3d: [drm:v3d_reset [v3d]] ERROR Resetting GPU for hang.
$kmsg v3d fec00000.v3d: [drm:v3d_reset [v3d]] ERROR V3D_ERR_STAT: 0x00001000