raspberrypi/linux #6873: rpivid log spam in kernel ring buffer

andrum993, 2025-05-28:
### Describe the bug

Yesterday evening I watched an H.265 video using Kodi, on Raspberry Pi OS (fully up to date). Today I noticed that the kernel ring buffer is full of log spam from rpivid. Video playback was absolutely fine. Can these messages be changed to debug only, to get rid of the log spam?

```
andrew@tarmachan:~ $ dmesg -e
[May27 19:38] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000009] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.011766] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000007] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000003] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.051173] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000007] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.054399] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000006] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.013309] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000006] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.012185] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000005] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
...
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.051314] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000008] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB AUX ent 3, timestamp=29950000, index=10
[  +0.054264] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000006] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.000534] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
...
[  +0.000002] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[  +0.051069] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
[  +0.000015] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
[  +0.000003] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
[  +0.000001] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
[May27 20:20] rpi-hevc-dec 1000800000.codec: phase1_cb: Post wait: 0x8
[  +0.002061] rpi-hevc-dec 1000800000.codec: phase1_thread: Coeff realloc (600000) OK
```

### Steps to reproduce the behaviour

Play an H.265 video using Kodi on Raspberry Pi OS running on a Raspberry Pi which has H.265 hardware decode built in (i.e. Pi 4 and Pi 5 family).

### Device (s)

Raspberry Pi 5

### System

```
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
[... 1165 lines cut ...]
3121895.083: sdram: sdram refresh 2081->8324 (1)
3124922.286: sdram: sdram refresh 2081->4162 (2)
3317745.343: sdram: sdram refresh 2081->8324 (1)
3370762.497: sdram: sdram refresh 2081->4162 (2)
3376002.347: sdram: sdram refresh 2081->8324 (1)
3377013.612: sdram: sdram refresh 2081->4162 (2)
3223559.576: sdram: sdram refresh 2081->8324 (1)
4091045.275: sdram: sdram refresh 2081->4162 (2)

dmesg log
---------

[877850.788494] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
snip
[880403.904186] rpi-hevc-dec 1000800000.codec: phase1_thread: Coeff realloc (600000) OK

EEPROM
------

BOOTLOADER: up to date
   CURRENT: Thu  8 May 14:13:17 UTC 2025 (1746713597)
    LATEST: Thu  8 May 14:13:17 UTC 2025 (1746713597)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

popcornmix, 2025-05-28:
Ping @jc-kynesim

andrum993, 2025-05-29:
> Its a fair comment. Those messages have been helpful when wondering why things are going wrong (and missing DPB entries normally shows something unhappy with the stream / decode process) but they could reasonably be put under a light debug switch. I'll look into it.

Thanks. I don't have a problem with the occasional message, but for this particular video then entire message buffer was filled with these extraneous messages.

(Having said that, I don't think it is 100% correct to be logging these messages by default when video playback looks completely fine to the end user).

jc-kynesim, 2025-05-29:
The kernel module cannot tell that the pici looks fine. It can tell that a frame it has been asked to use as a reference frame doesn't exist and that this will potentially cause corruption. Reasons it might not be visible include:
(a) the frame is in the list but actually never used (the s/w cannot detect this)
(b) when a frame is missing the s/w picks a nearby frame to use as a substitute and this will often produce output that is very close to what is required.
On the other hand if corruption is seen it can be very handy to have a log of possible causes.

Either way I'll see what I can do to hide these unless the user asks for them

andrum993, 2025-07-08:
Just had another look at this. It seems these messages are the result of my using the fast forward feature in Kodi. Example:

```
<4>[542105.211895] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
<4>[542105.211920] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
<4>[542105.211924] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
<4>[542105.211928] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
<4>[542107.511763] rpi-hevc-dec 1000800000.codec: Missing DPB ent 0, timestamp=0
<4>[542107.511791] rpi-hevc-dec 1000800000.codec: Missing DPB ent 1, timestamp=0
<4>[542107.511794] rpi-hevc-dec 1000800000.codec: Missing DPB ent 2, timestamp=0
<4>[542107.511797] rpi-hevc-dec 1000800000.codec: Missing DPB ent 3, timestamp=0
```

These messages are getting logged at log level 4 - warning - which seems a bit excessive for something that is not actually a problem with the running software, but rather a potential problem with a video file. What about perhaps changing the log level to 6 (informational), or 7 (debug), and also rate limiting it using printk_ratelimit() ? The default rate limit is 10 messages in 5 seconds, which should be enough to keep the ring buffer from getting completely filled up.

jc-kynesim, 2025-07-09:
Sorry - should have commented sooner. Those messages will be behind a "debug" parameter. The fix is in the driver version we are attempting to upstream but hasn't percolated down to the current release version yet.  It will happen but it doesn't have a very high priority.

jc-kynesim, 2025-07-09:
Not a useful one at the moment! Whilst we do have a branch that holds the upstream effort (not yet merged - state of the argument) it is in the mainline tree and so lacks all other downstream drivers. Also downstream wants some extra bits to keep backward compatibility (upstream has a new method of describing sand which isn't well supported elsewhere yet). 
If you want a kernel without those debug lines right now then I suggest that you just comment them out.
The backport into the PiOS kernel will happen, it just hasn't happened yet.

6by9, 2025-07-09:
If you wanted my dev branch for it, then it's https://github.com/6by9/linux/tree/mainline_2712_hevc_dec_v4

I will look to do a backport, but it's likely to be a fixup patch on the current rpi-6.12.y rather than a revert and reapply. If we can get it merged properly before the last kernel release of the year, then it will just be in the next LTS kernel (we can hope).

andrum993, 2025-07-09:
> If you wanted my dev branch for it, then it's https://github.com/6by9/linux/tree/mainline_2712_hevc_dec_v4
>
> I will look to do a backport, but it's likely to be a fixup patch on the current [...]

I'm not looking for a backport - I just wanted to see what the fix looked like.

andrum993, 2026-03-20:
I've had a look at the fix, using rpi-update to get 6.18, but it's not really a fix. It only masks out some of the log spam. To me, none of the messages I'm seeing should be in the kernel log by default, and if you're adding a debug switch, then none of the messages that are emitted that are to do with the video apparently being not quite right should be in the kernel log - they should all be behind the debug switch. It's also not clear why this even needed any changes to the driver, when you could just set the system's loglevel to 7 (debug). While the messages I'm sure were invaluable during development of the driver and hardware, they should be debug only, not info or warning. But I'm not a kernel dev.

However, I'm now not bothered by this issue, since the Pi these messages appear on is now solely dedicated to Kodi. I will, however, leave this issue open in case anyone else is affected by it.

pelwell, 2026-03-20:
> It's also not clear why this even needed any changes to the driver, when you could just set the system's loglevel to 7 (debug)

Or raising an issue, if you're going down that route.