raspberrypi/linux #6878: Kernel 6.12.30 breaks HyperPixel 4.0 Square screen on rpi5 and cm5

charel, 2025-06-02:
### Describe the bug

When installing the 6.12.30 kernel, the hyperpixel 4.0 square screen stops working. It was working previously on 6.12.25 (current version in repos). 

Similar, but not identical to #6871.

### Steps to reproduce the behaviour

1) Install new raspberry pi OS on uSD/NVMe on a cm5 or rpi5 with a Hyperpixel 4.0 square screen (e.g. from Pimoronni or on a Hackberry)
2) Adjust config.txt to work with the screen (see below)
3) Confirm it works 
4) Update kernel to 6.12.30 using `rpi-update`
5) Screen stops working

### Device (s)

Raspberry Pi CM5 Lite

### System

##### uname -a
Before update:
`Linux raspberrypi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux`
After switching to newer kernel:
 `Linux raspberrypi 6.12.30-v8-16k+ #1879 SMP PREEMPT Fri May 23 13:30:44 BST 2025 aarch64 GNU/Linux`

### Logs

#### config.txt
```config.txt
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1
[... 751 lines cut ...]
130: a1    pn | hi // WIFI_SDIO_CLK/GPIO30 = SD2_CLK
131: a1    pu | hi // WIFI_SDIO_CMD/GPIO31 = SD2_CMD
132: a1    pu | hi // WIFI_SDIO_D0/GPIO32 = SD2_DAT0
133: a1    pu | hi // WIFI_SDIO_D1/GPIO33 = SD2_DAT1
134: a1    pu | hi // WIFI_SDIO_D2/GPIO34 = SD2_DAT2
135: a1    pu | hi // WIFI_SDIO_D3/GPIO35 = SD2_DAT3
200: ip    pd | hi // RP1_SDA/AON_GPIO0 = input
201: ip    pd | hi // RP1_SCL/AON_GPIO1 = input
202: op dh pd | hi // RP1_RUN/AON_GPIO2 = output
203: op dh pd | hi // SD_IOVDD_SEL/AON_GPIO3 = output
204: op dh pd | hi // SD_PWR_ON/AON_GPIO4 = output
205: op dh pd | hi // ANT1/AON_GPIO5 = output
206: op dl pd | lo // ANT2/AON_GPIO6 = output
208: ip    pd | lo // 2712_WAKE/AON_GPIO8 = input
209: op dh pd | hi // 2712_STAT_LED/AON_GPIO9 = output
212: ip    pd | lo // PMIC_INT/AON_GPIO12 = input
213: a2    pu | hi // UART_TX_FS/AON_GPIO13 = VC_TXD0
214: a3    pu | hi // UART_RX_FS/AON_GPIO14 = VC_RXD0
232: a1    -- | hi // HDMI0_SCL/AON_SGPIO0 = HDMI_TX0_BSC_SCL
233: a1    -- | hi // HDMI0_SDA/AON_SGPIO1 = HDMI_TX0_BSC_SDA
234: a1    -- | hi // HDMI1_SCL/AON_SGPIO2 = HDMI_TX1_BSC_SCL
235: a1    -- | hi // HDMI1_SDA/AON_SGPIO3 = HDMI_TX1_BSC_SDA
236: a2    -- | hi // PMIC_SCL/AON_SGPIO4 = BSC_M2_SCL
237: a2    -- | hi // PMIC_SDA/AON_SGPIO5 = BSC_M2_SDA
```

charel, 2025-06-02:
#### DRI state
`sudo cat /sys/kernel/debug/dri/1/state`
```
plane[46]: plane-0
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[65]: plane-1
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[81]: plane-2
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[93]: plane-3
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=0
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[105]: plane-4
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=1
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[116]: plane-5
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=2
	color-encoding=ITU-R BT.709 YCbCr
	color-range=YCbCr limited range
	color_mgmt_changed=0
plane[127]: plane-6
	crtc=(null)
	fb=0
	crtc-pos=0x0+0+0
	src-pos=0.000000x0.000000+0.000000+0.000000
	rotation=1
	normalized-zpos=3
[... 562 lines cut ...]
	output_bpc=0
	output_format=RGB
	tmds_char_rate=0
connector[63]: Writeback-1
	crtc=(null)
	self_refresh_aware=0
	max_requested_bpc=0
	colorspace=Default
connector[79]: Writeback-2
	crtc=(null)
	self_refresh_aware=0
	max_requested_bpc=0
	colorspace=Default
HVS State
	Core Clock Rate: 0
	Channel 0
		in use=0
		load=0
	Channel 1
		in use=0
		load=0
	Channel 2
		in use=0
		load=0
```

mdromjanmdsiddikul4-creator, 2026-10-10:
> ### Describe the bug
> When installing the 6.12.30 kernel, the hyperpixel 4.0 square screen stops working. It was working previously on 6.12.25 (current version in repos).
> 
> Similar, but not identical to [#6871](https://github.com/raspberrypi/linux/issues/6871).
> 
> ### Steps to reproduce the behaviour
> 1. Install new raspberry pi OS on uSD/NVMe on a cm5 or rpi5 with a Hyperpixel 4.0 square screen (e.g. from Pimoronni or on a Hackberry)
> 2. Adjust config.txt to work with the screen (see below)
> 3. Confirm it works
> 4. Update kernel to 6.12.30 using `rpi-update`
> 5. Screen stops working
> 
> ### Device (s)
> Raspberry Pi CM5 Lite
> 
> ### System
> ##### uname -a
> Before update: `Linux raspberrypi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux` After switching to newer kernel: `Linux raspberrypi 6.12.30-v8-16k+ #1879 SMP PREEMPT Fri May 23 13:30:44 BST 2025 aarch64 GNU/Linux`
> 
> ### Logs
> #### config.txt
> # For more options and information see
> # http://rptl.io/configtxt
> # Some settings may impact device functionality. See link above for details
> 
> # Uncomment some or all of these to enable the optional hardware interfaces
> #dtparam=i2c_arm=on
> #dtparam=i2s=on
> #dtparam=spi=on
> 
> # Enable audio (loads snd_bcm2835)
> dtparam=audio=on
> 
> # Additional overlays and parameters are documented
> # /boot/firmware/overlays/README
> 
> # Automatically load overlays for detected cameras
> camera_auto_detect=1
> 
> # Automatically load overlays for detected DSI displays
> display_auto_detect=1
> 
> # Automatically load initramfs files, if found
> auto_initramfs=1
> 
> # Enable DRM VC4 V3D driver
> dtoverlay=vc4-kms-v3d
> max_framebuffers=2
> 
> # Don't have the firmware create an initial video= setting in cmdline.txt.
> # Use the kernel's default instead.
> disable_fw_kms_setup=1
> 
> # Run in 64-bit mode
> arm_64bit=1
> 
> # Disable compensation for displays with overscan
> disable_overscan=1
> 
> # Run as fast as firmware / board allows
> arm_boost=1
> 
> [cm4]
> # Enable host mode on the 2711 built-in XHCI USB controller.
> # This line should be removed if the legacy DWC2 controller is required
> # (e.g. for USB device mode) or if USB support is not required.
> otg_mode=1
> 
> [cm5]
> dtoverlay=dwc2,dr_mode=host
[... 748 lines cut ...]
> 133: a1    pu | hi // WIFI_SDIO_D1/GPIO33 = SD2_DAT1
> 134: a1    pu | hi // WIFI_SDIO_D2/GPIO34 = SD2_DAT2
> 135: a1    pu | hi // WIFI_SDIO_D3/GPIO35 = SD2_DAT3
> 200: ip    pd | hi // RP1_SDA/AON_GPIO0 = input
> 201: ip    pd | hi // RP1_SCL/AON_GPIO1 = input
> 202: op dh pd | hi // RP1_RUN/AON_GPIO2 = output
> 203: op dh pd | hi // SD_IOVDD_SEL/AON_GPIO3 = output
> 204: op dh pd | hi // SD_PWR_ON/AON_GPIO4 = output
> 205: op dh pd | hi // ANT1/AON_GPIO5 = output
> 206: op dl pd | lo // ANT2/AON_GPIO6 = output
> 208: ip    pd | lo // 2712_WAKE/AON_GPIO8 = input
> 209: op dh pd | hi // 2712_STAT_LED/AON_GPIO9 = output
> 212: ip    pd | lo // PMIC_INT/AON_GPIO12 = input
> 213: a2    pu | hi // UART_TX_FS/AON_GPIO13 = VC_TXD0
> 214: a3    pu | hi // UART_RX_FS/AON_GPIO14 = VC_RXD0
> 232: a1    -- | hi // HDMI0_SCL/AON_SGPIO0 = HDMI_TX0_BSC_SCL
> 233: a1    -- | hi // HDMI0_SDA/AON_SGPIO1 = HDMI_TX0_BSC_SDA
> 234: a1    -- | hi // HDMI1_SCL/AON_SGPIO2 = HDMI_TX1_BSC_SCL
> 235: a1    -- | hi // HDMI1_SDA/AON_SGPIO3 = HDMI_TX1_BSC_SDA
> 236: a2    -- | hi // PMIC_SCL/AON_SGPIO4 = BSC_M2_SCL
> 237: a2    -- | hi // PMIC_SDA/AON_SGPIO5 = BSC_M2_SDA
> ```
> 
> ### Additional context
> _No response_