raspberrypi/linux #6887: kernel: Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP

zxmak, 2025-06-09:
### Describe the bug

It happened while browsing the web in Chromium.
When it occurred, the system was unable to kill the Chromium process and couldn't complete a sudo reboot.
I had never encountered such an issue before.

### Steps to reproduce the behaviour

-

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)
Linux rzx 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

### Logs

Jun 09 11:11:19 rzx kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
Jun 09 11:11:19 rzx kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed disabling i/p port, ret -62
Jun 09 11:11:21 rzx kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 8 outstanding
Jun 09 11:11:21 rzx kernel: ------------[ cut here ]------------
Jun 09 11:11:21 rzx kernel: WARNING: CPU: 2 PID: 336013 at drivers/media/common/videobuf2/videobuf2-core.c:2215 __vb2_queue_cancel+0x238/0x2d8 [videobuf2_common]
Jun 09 11:11:21 rzx kernel: Modules linked in: tun snd_seq_dummy snd_hrtimer snd_seq zram lz4_compress brcmfmac_wcc binfmt_misc snd_usb_audio brcmfmac snd_hwdep snd_usbmidi_lib snd_rawmidi bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) sg snd_seq_device joydev brcmutil bcm2835_mmal_vchiq(C) rpi_hevc_dec cfg80211 videobuf2_vmalloc vc_sm_cma(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 rfkill videodev videobuf2_common raspberrypi_hwmon mc snd_bcm2835(C) raspberrypi_gpiomem nvmem_rmem uio_pdrv_genirq uio i2c_dev fuse dm_mod ip_tables x_tables ipv6 v3d vc4 snd_soc_hdmi_codec drm_display_helper cec gpu_sched drm_shmem_helper drm_dma_helper drm_kms_helper drm drm_panel_orientation_quirks i2c_brcmstb snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd backlight
Jun 09 11:11:21 rzx kernel: CPU: 2 UID: 1000 PID: 336013 Comm: ThreadPoolSingl Tainted: G        WC         6.12.25+rpt-rpi-v8 #1  Debian 1:6.12.25-1+rpt1
Jun 09 11:11:21 rzx kernel: Tainted: [W]=WARN, [C]=CRAP
Jun 09 11:11:21 rzx kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jun 09 11:11:21 rzx kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jun 09 11:11:21 rzx kernel: pc : __vb2_queue_cancel+0x238/0x2d8 [videobuf2_common]
Jun 09 11:11:21 rzx kernel: lr : __vb2_queue_cancel+0x34/0x2d8 [videobuf2_common]
Jun 09 11:11:21 rzx kernel: sp : ffffffc08f223b00
Jun 09 11:11:21 rzx kernel: x29: ffffffc08f223b00 x28: ffffffe7b5511d18 x27: ffffffe7b536b0f8
Jun 09 11:11:21 rzx kernel: x26: 0000000000000000 x25: ffffff8044c5f548 x24: ffffff80021889c0
Jun 09 11:11:21 rzx kernel: x23: 0000000000000000 x22: ffffff800cd009a8 x21: ffffff800cd009a8
Jun 09 11:11:21 rzx kernel: x20: 000000000000000a x19: ffffff800cd009a8 x18: ffffffffffffffff
Jun 09 11:11:21 rzx kernel: x17: 756f656d6954203a x16: ffffffe7df35c8f0 x15: 6873756c665f6365
Jun 09 11:11:21 rzx kernel: x14: 0000000000000004 x13: ffffff8042640028 x12: 0000000000000000
Jun 09 11:11:21 rzx kernel: x11: ffffff80053c25c0 x10: ffffff80053c2480 x9 : ffffffe7df35cab8
Jun 09 11:11:21 rzx kernel: x8 : ffffffc08f223a10 x7 : 0000000000000000 x6 : 0000000000000228
Jun 09 11:11:21 rzx kernel: x5 : 0000000000150010 x4 : fffffffec1a75c20 x3 : 0000000000150010
Jun 09 11:11:21 rzx kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000008
Jun 09 11:11:21 rzx kernel: Call trace:
Jun 09 11:11:21 rzx kernel:  __vb2_queue_cancel+0x238/0x2d8 [videobuf2_common]
Jun 09 11:11:21 rzx kernel:  vb2_core_streamoff+0x2c/0xd0 [videobuf2_common]
Jun 09 11:11:21 rzx kernel:  vb2_streamoff+0x20/0x78 [videobuf2_v4l2]
Jun 09 11:11:21 rzx kernel:  v4l2_m2m_streamoff+0x50/0x160 [v4l2_mem2mem]
Jun 09 11:11:21 rzx kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x38 [v4l2_mem2mem]
Jun 09 11:11:21 rzx kernel:  v4l_streamoff+0x2c/0x40 [videodev]
Jun 09 11:11:21 rzx kernel:  __video_do_ioctl+0x170/0x3e0 [videodev]
Jun 09 11:11:21 rzx kernel:  video_usercopy+0x1bc/0x780 [videodev]
Jun 09 11:11:21 rzx kernel:  video_ioctl2+0x20/0x38 [videodev]
Jun 09 11:11:21 rzx kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Jun 09 11:11:21 rzx kernel:  __arm64_sys_ioctl+0xb4/0x100
Jun 09 11:11:21 rzx kernel:  invoke_syscall+0x50/0x120
Jun 09 11:11:21 rzx kernel:  el0_svc_common.constprop.0+0x48/0xf0
Jun 09 11:11:21 rzx kernel:  do_el0_svc+0x24/0x38
Jun 09 11:11:21 rzx kernel:  el0_svc+0x30/0xd0
Jun 09 11:11:21 rzx kernel:  el0t_64_sync_handler+0x100/0x130
Jun 09 11:11:21 rzx kernel:  el0t_64_sync+0x190/0x198
Jun 09 11:11:21 rzx kernel: ---[ end trace 0000000000000000 ]---
Jun 09 11:11:21 rzx kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buffer 0 in active state
Jun 09 11:11:21 rzx kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buffer 1 in active state
Jun 09 11:11:21 rzx kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buffer 2 in active state
[... 50 lines cut ...]
Jun 09 11:11:21 rzx kernel:  __fput+0xd0/0x2e0
Jun 09 11:11:21 rzx kernel:  __fput_sync+0x50/0x68
Jun 09 11:11:21 rzx kernel:  __arm64_sys_close+0x40/0x90
Jun 09 11:11:21 rzx kernel:  invoke_syscall+0x50/0x120
Jun 09 11:11:21 rzx kernel:  el0_svc_common.constprop.0+0x48/0xf0
Jun 09 11:11:21 rzx kernel:  do_el0_svc+0x24/0x38
Jun 09 11:11:21 rzx kernel:  el0_svc+0x30/0xd0
Jun 09 11:11:21 rzx kernel:  el0t_64_sync_handler+0x100/0x130
Jun 09 11:11:21 rzx kernel:  el0t_64_sync+0x190/0x198
Jun 09 11:11:21 rzx kernel: Code: 5400018c b9400843 6b01007f 5400012b (b9400282) 
Jun 09 11:11:21 rzx kernel: ---[ end trace 0000000000000000 ]---
Jun 09 11:17:01 rzx CRON[336889]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun 09 11:17:01 rzx CRON[336890]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Jun 09 11:17:01 rzx CRON[336889]: pam_unix(cron:session): session closed for user root
Jun 09 11:42:58 rzx kernel: bcm2835-codec bcm2835-codec: Mutex fail
Jun 09 11:43:01 rzx dbus-daemon[761]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.594' (uid=1000 pid=339828 comm="/usr/lib/firefox/firefox")
Jun 09 11:43:01 rzx systemd[1]: Starting systemd-timedated.service - Time & Date Service...
Jun 09 11:43:01 rzx dbus-daemon[761]: [system] Successfully activated service 'org.freedesktop.timedate1'
Jun 09 11:43:01 rzx systemd[1]: Started systemd-timedated.service - Time & Date Service.
Jun 09 11:43:07 rzx rtkit-daemon[956]: Supervising 6 threads of 3 processes of 1 users.
Jun 09 11:43:07 rzx rtkit-daemon[956]: Supervising 6 threads of 3 processes of 1 users.
Jun 09 11:43:30 rzx dbus-daemon[761]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.598' (uid=1000 pid=340047 comm="/usr/lib/chromium/chromium --force-renderer-access")
Jun 09 11:43:30 rzx systemd[1]: bluetooth.service - Bluetooth service was skipped because of an unmet condition check (ConditionPathIsDirectory=/sys/class/bluetooth).
Jun 09 11:43:31 rzx systemd[1]: systemd-timedated.service: Deactivated successfully.
Jun 09 11:43:34 rzx kernel: bcm2835-codec bcm2835-codec: Mutex fail

zxmak, 2025-06-09:
I don't have exact steps to reproduce, but I never had such issues on the same config before latest apt upgrade. So it may be related with some changes in the latest update.

zxmak, 2025-06-09:
> Was your previous good system running on a 6.6 kernel, so it's the jump to 6.12 that has exposed issues?

I've been running version 6.12.25 for quite some time now - pretty much since it was first released—and I haven't encountered this issue at all.

After rebooting, I tried browsing through sites from my history, but the issue didn’t occur again.

6by9, 2025-06-12:
`alloc_compact_internal` is due to being out of gpu_mem. That may be symptom rather than cause, but checking `vcgencmd get_mem gpu` for the memory split configuration would be worthwhile.