raspberrypi/linux #6895: VL805 Chip Version Appears Blank In "sudo rpi-eeprom-update"

OperationOfTheCloud, 2025-06-11:
### Describe the bug

The VL805 chip of the RPi4 could possibly be unsupported in new EEPROM updates? (my best guess)

### Steps to reproduce the behaviour

Execute sudo rpi-eeprom update and look in the output:
```
BOOTLOADER: up to date
   CURRENT: Thu  8 May 15:21:35 UTC 2025 (1746717695)
    LATEST: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 
    LATEST: 
```
Executing lsusb also does not show anything, not even a single line. It just does like this:
```
chase@chase:~ $ lsusb
chase@chase:~ $:
```
USB devices also do not work.

### Device (s)

Raspberry Pi 4 Mod. B

### System

I am using the Raspberry Pi OS (64-bit) released on 2025-05-13 in the official Raspberry Pi Imager.

### Logs

DMESG OUTPUT:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.12.25+rpt-rpi-v8 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef667c0..0x000000003ef66820 (0 KiB) nomap non-reusable nvram@0
[    0.000000] OF: reserved mem: 0x000000003ef66580..0x000000003ef6677f (0 KiB) nomap non-reusable nvram@1
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x00000000fbffffff] (4032MB)
[    0.000000] NUMA: Initialized distance table, cnt=1
[    0.000000] NODE_DATA(0) allocated [mem 0xfb7fa300-0xfb7fcfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 33 pages/cpu s95064 r8192 d31912 u135168
[    0.000000] pcpu-alloc: s95064 r8192 d31912 u135168 alloc=33*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[... 401 lines cut ...]
[    4.839085] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    4.839614] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    5.190627] Adding 524284k swap on /var/swap.  Priority:-2 extents:4 across:5709824k SS
[    5.352752] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.352769] Bluetooth: BNEP filters: protocol multicast
[    5.352782] Bluetooth: BNEP socket layer initialized
[    5.424239] Bluetooth: hci0: BCM: features 0x2f
[    5.425497] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    5.425505] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    5.443503] Bluetooth: MGMT ver 1.23
[    5.455722] NET: Registered PF_ALG protocol family
[    6.384618] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    6.386400] bcmgenet fd580000.ethernet eth0: Link is Down
[    6.406307] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.390996] Bluetooth: RFCOMM TTY layer initialized
[    9.391030] Bluetooth: RFCOMM socket layer initialized
[    9.391051] Bluetooth: RFCOMM ver 1.11
[  183.159374] ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
```
lsusb command also appears blank output.

### Additional context

USB devices do not work;
Version is blank for VL805;

pelwell, 2025-06-11:
Is this a new device? Has it worked with previous images?

This line shows that the PCIe link to the VL805 is not functioning, which probably explains the symptoms you are seeing:
```
[    0.823453] brcm-pcie fd500000.pcie: link down
```

OperationOfTheCloud, 2025-06-12:
Hm, true. This device is 3 years, as I bought it in 2022. It has worked when I first got it.