raspberrypi/linux #6899: Please consider enabling DSA subsystem (CONFIG_NET_DSA=m)

equinox0815, 2025-06-14:
### Describe the bug

I have a carrier board for CM4 and CM5 from Axzez ([Interceptor](https://www.axzez.com/product-page/interceptor-carrier-board)) which has an onboard Ethernet switch from Realtek. Axzez provides firmware images that contain a custom kernel that allows configuration of the switch using the [DSA subsystem](https://www.kernel.org/doc/html/latest/networking/dsa/index.html). Unfortunately for my use-case these images are less than ideal and i would like to use standard RaspiOS images instead.
Because i don't want to recompile the whole kernel every time the RaspiOS kernel gets updated i looked into the possibility to only compile the relevant modules. Unfortunately this does not work because the DSA subsystem depends on Switchdev (CONFIG_NET_SWITCHDEV=y) which cannot be compiled as a module but rather must be part of the main kernel image. Also some symbols of the DSA module need to be built-in as well (https://github.com/raspberrypi/linux/blob/rpi-6.12.y/net/dsa/Makefile#L3) when DSA is enabled no matter if DSA itself is compiled as a module or built-in.

I did some testing and i looks to me as if the official kernel packages would be shipped with `CONFIG_NET_DSA=m`.  I could use DKMS or something similar to just build the missing modules. This would make my life much easier. Also i think there a number of other carrier boards that use Ethernet switches which could benefit from the change.

### Steps to reproduce the behaviour

I wrote a little test [script](https://gist.github.com/equinox0815/d9329170db8eab9b6f69829c59a5a56c) that tries to build the modules manually. This is script is not very well optimized but can be used to reproduce the problem. The patch file that is referenced in the script can be found [here](https://github.com/niflostancu/rpi-debian-build-scripts/blob/master/configs/interceptor-6.x/files/kernel-6.5/rtl8365mb.patch).

### Device (s)

Raspberry Pi CM4

### System

* OS Version
```
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage2
```
* Firmware Version
```
Apr 30 2025 13:33:39 
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)
```
* Kernel Version
```
Linux dsa-test 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

pelwell, 2025-06-18:
...and if it doesn't, say which extra Kconfig settings you need to build the required driver modules.

pelwell, 2025-06-19:
Thanks for going through the awkward testing process - I've now merged the pull request, and the resulting kernel will eventually reach the relevant apt package.

closed 2025-06-19 by equinox0815