raspberrypi/linux #6908: [PRi6]WiFi not work

LubinLew, 2025-06-18:
### Describe the bug

wlan0 interface always down,
I tried **ifconfig wlan0 up** and **ip link set dev wlan0 up**, not work.

### Steps to reproduce the behaviour

1. Burn the Image(I tried ubuntu 20.04-server/24.04-server/25.04-server and raspberrypi os).
2. Startup

### Device (s)

Raspberry Pi CM5

### System

```bash
cat /etc/rpi-issue
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage4
```

```bash
vcgencmd version
2025/02/12 10:51:52 
Copyright (c) 2012 Broadcom
version f788aab6 (release) (embedded)
```

```bash
uname -a
Linux raspberrypi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

```

### Logs

[dmesg.log](https://github.com/user-attachments/files/20798220/dmesg.log)

```log
root@raspberrypi:~# dmesg |grep brcmf
[ 2.170084] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 2.173117] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[ 2.174012] usbcore: registered new interface driver brcmfmac
[ 2.461386] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[ 2.461634] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[ 3.774084] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 92.025612] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -52
[ 95.905651] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -52
```

pelwell, 2025-06-18:
What do the following commands report? You _may_ need to use RPiOS.
```
$ ifconfig wlan0
$ iwconfig wlan0
$ rfkill
$ iwlist scan
```

pelwell, 2025-06-19:
Have you attempted to connect to an access point? If so, which one(s) from that list?

LubinLew, 2025-06-19:
@pelwell Thanks.

wpa_cli command works.

```cmd
add_network
set_network 0 ssid "AP"
set_network 0 psk "passwd"
enable_network 0
```

closed 2025-06-19 by LubinLew