raspberrypi/linux #6911: Storage Broken After Upgrading from 6.12.32 to 6.12.33

ak-rex, 2025-06-19:
### Describe the bug

# Describe the bug

After upgrading the Linux kernel from 6.12.32 to 6.12.33, USB storage devices no longer function when connected to the OTG port using the dwc2 driver. Possibly related to recent changes in dwc2.

### Steps to reproduce the behaviour

# Reproduction Steps:
Boot kernel 6.12.33 with dwc2 enabled.
Plug in a USB flash drive to the OTG port.
Observe dmesg and /dev/ for device activity.
Compare behavior with kernel 6.12.32.

### Device (s)

Raspberry Pi CM5 Lite

### System

# System Info:
Model: Raspberry Pi CM5 / Custom IO Board 
OS: Debian 12 Bookworm
USB port used: OTG port
Driver in use: dwc2, Device operates in host mode; not using gadget functionality.

# Kernel versions:
Working: 6.12.32
Broken: 6.12.33

### Additional context

# Expected Behavior:
When a USB storage device is plugged into the OTG port, it should enumerate and mount as a block device (e.g., /dev/sda), as it did under 6.12.32.

# Actual Behavior:
Under 6.12.33, plugging in a USB storage device results in:
No enumeration
No device node (/dev/sdX) created

ak-rex, 2025-06-19:
<details>
  <summary>config.txt</summary>
  
```
[pi3]
#dtoverlay=clockworkpi-devterm-cm3
dtoverlay=clockworkpi-uconsole-cm3
dtoverlay=vc4-kms-v3d
dtparam=spi=on
dtoverlay=spi-gpio35-39
gpio=11=op,dh
enable_uart=1

[pi4]
#dtoverlay=clockworkpi-devterm
dtoverlay=clockworkpi-uconsole
dtoverlay=vc4-kms-v3d-pi4,cma-384
dtparam=spi=on
enable_uart=1

[pi5]
#dtoverlay=clockworkpi-devterm-cm5
dtoverlay=clockworkpi-uconsole-cm5
dtoverlay=vc4-kms-v3d-pi5,cma-384
dtparam=pciex1=off
dtparam=uart0

[all]
ignore_lcd=1
max_framebuffers=2
disable_overscan=1
dtparam=audio=on
dtoverlay=audremap,pins_12_13
dtoverlay=dwc2,dr_mode=host
dtparam=ant2
```
</details>

P33M, 2025-06-23:
Reproduced here. `dtoverlay=dwc2,dr_mode=host` is required on the CM5IO board (or jumper USB_OTG to ground).

I occasionally get lines in dmesg such as `[    3.713238] usb-storage 1-1.3:1.0: Max LUN 101 is not valid, using 0 instead` which makes no sense. DMA addresses wrong?

P33M, 2025-06-23:
UMS sets `URB_NO_TRANSFER_DMA_MAP` for bulk transfers which when used with the mechanism to shunt `urb->transfer_buffer` to a kmalloc'd one, results in dwc2's DMA engine pointed at the old (mapped) transfer buffer, which for IN buffers then gets stomped with zeroes when the alignment buffer is unwound.

A side-effect of using the alignment mechanism in a broader context means that every UMS CDB/CSB gets a kmalloc/dma_map_single/kfree which is unecessarily expensive.

P33M, 2025-06-23:
In about 40 minutes, you should be able to do `sudo rpi-update pulls/6923` to install a kernel with this regression fixed. Usual caveats apply: only test on a system that you have ready access to, and you don't care if the root FS gets corrupted.

ak-rex, 2025-06-23:
PR [#6923](https://github.com/raspberrypi/linux/pull/6923) fixes my issue. Thank you!