raspberrypi/linux #6912: ptp4l broken in 6.12.25

catubc, 2025-06-19:
### Describe the bug

Hello
I've got dozens of rpis on a ptp4l network with a dell workstation master.  Seems the rpis with the 6..6.74 kernel synchronize well (the switch to listening mode) but the 6.12.25 kernel RPis aren't doing so.  It's the same issue with the previous kernel that was supposed to be fixed in 6.12.10 I believe.

I posted the issue here with a printout and a link to the previous kernel bug.

https://forums.raspberrypi.com/viewtopic.php?t=389111

### Steps to reproduce the behaviour

Run ptp4l on a raspbbery pi with 6.12.25 kernel.

### Device (s)

Raspberry Pi CM5

### System

[raspinfo.txt](https://github.com/user-attachments/files/20823217/raspinfo.txt)

### Logs

The logs are: 

cat@raspberrypi:~ $ sudo ethtool -i eth0
driver: macb
version: 6.12.25+rpt-rpi-2712
firmware-version: 
expansion-rom-version: 
bus-info: 1f00100000.ethernet
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

cat@raspberrypi:~ $ sudo ptp4l -i eth0 -f /etc/ptp4l.conf -m -l 7
ptp4l[9949.499]: config item (null).assume_two_step is 0
ptp4l[9949.499]: config item (null).check_fup_sync is 0
ptp4l[9949.500]: config item (null).tx_timestamp_timeout is 1
ptp4l[9949.500]: config item (null).hwts_filter is 0
ptp4l[9949.500]: config item (null).clock_servo is 0
ptp4l[9949.500]: config item (null).clock_type is 32768
ptp4l[9949.500]: config item (null).clock_servo is 0
ptp4l[9949.500]: config item (null).clockClass is 248
ptp4l[9949.500]: config item (null).clockAccuracy is 254
ptp4l[9949.500]: config item (null).offsetScaledLogVariance is 65535
ptp4l[9949.500]: config item (null).productDescription is ';;'
ptp4l[9949.500]: config item (null).revisionData is ';;'
ptp4l[9949.500]: config item (null).userDescription is ''
ptp4l[9949.500]: config item (null).manufacturerIdentity is '00:00:00'
ptp4l[9949.500]: config item (null).domainNumber is 0
ptp4l[9949.501]: config item (null).slaveOnly is 1
ptp4l[9949.501]: config item (null).gmCapable is 1
ptp4l[9949.501]: config item (null).gmCapable is 1
ptp4l[9949.501]: config item (null).G.8275.defaultDS.localPriority is 128
ptp4l[9949.501]: config item (null).maxStepsRemoved is 255
ptp4l[9949.501]: config item (null).time_stamping is 1
ptp4l[9949.501]: config item (null).twoStepFlag is 1
ptp4l[9949.501]: config item (null).twoStepFlag is 1
ptp4l[9949.501]: config item (null).time_stamping is 1
ptp4l[9949.501]: config item (null).priority1 is 248
ptp4l[9949.501]: config item (null).priority2 is 248
ptp4l[9949.501]: interface index 2 is up
ptp4l[9949.501]: config item (null).free_running is 0
ptp4l[9949.501]: selected /dev/ptp0 as PTP clock
ptp4l[9949.501]: config item (null).clockIdentity is '000000.0000.000000'
ptp4l[9949.502]: config item (null).uds_address is '/var/run/ptp4l'
ptp4l[9949.502]: section item /var/run/ptp4l.announceReceiptTimeout now 0
[... 103 lines cut ...]
ptp4l[9949.508]: config item /var/run/ptp4l.G.8275.portDS.localPriority is 128
ptp4l[9949.508]: config item /var/run/ptp4l.logSyncInterval is 4
ptp4l[9949.508]: config item /var/run/ptp4l.operLogSyncInterval is 0
ptp4l[9949.508]: config item /var/run/ptp4l.logMinPdelayReqInterval is 0
ptp4l[9949.508]: config item /var/run/ptp4l.operLogPdelayReqInterval is 0
ptp4l[9949.508]: config item /var/run/ptp4l.neighborPropDelayThresh is 20000000
ptp4l[9949.508]: config item /var/run/ptp4l.min_neighbor_prop_delay is -20000000
ptp4l[9949.508]: config item /var/run/ptp4l.asCapable is 1
ptp4l[9949.508]: config item /var/run/ptp4l.inhibit_delay_req is 0
ptp4l[9949.508]: config item (null).uds_address is '/var/run/ptp4l'
ptp4l[9949.508]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[9949.508]: port 1: received link status notification
ptp4l[9949.508]: interface index 2 is up
ptp4l[9949.609]: port 1: setting asCapable
ptp4l[9952.185]: port 1: new foreign master b49691.fffe.7472d0-1
ptp4l[9969.784]: selected best master clock b49691.fffe.7472d0
ptp4l[9969.785]: port 1: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[9978.584]: PI servo: sync interval 8.000 kp 0.087 ki 0.037500
ptp4l[9978.585]: port 1: have FOLLOW_UP 473, expecting FOLLOW_UP but got SYNC 474, dropping
ptp4l[9982.492]: port 1: delay timeout
....

### Additional context

Rpi with 6.6.74 kernel works fine. Identical everything else.

catubc, 2025-06-20:
Some more logs after running
`
sudo apt full-upgrade -y`

Still same issues:

```
...
ptp4l[149.024]: config item /var/run/ptp4l.inhibit_delay_req is 0
ptp4l[149.024]: config item (null).uds_address is '/var/run/ptp4l'
ptp4l[149.024]: port 0: INITIALIZING to LISTENING on INIT_COMPLETE
ptp4l[149.024]: port 1: received link status notification
ptp4l[149.024]: interface index 2 is up
ptp4l[149.697]: port 1: setting asCapable
ptp4l[149.926]: port 1: new foreign master b49691.fffe.7472d0-1
ptp4l[165.926]: selected best master clock b49691.fffe.7472d0
ptp4l[165.926]: port 1: LISTENING to UNCALIBRATED on RS_SLAVE
ptp4l[169.139]: port 1: delay timeout
ptp4l[173.925]: PI servo: sync interval 8.000 kp 0.087 ki 0.037500
ptp4l[173.925]: port 1: have FOLLOW_UP 10481, expecting FOLLOW_UP but got SYNC 10482, dropping
ptp4l[190.724]: port 1: delay timeout

```

by, 2025-06-24:
Kernel 6.16-rc2 also fails with `ptp4l[3001]: [10.124] Failed to open /dev/ptp0: Device or resource busy`

There wasn't an issue with 6.15

pelwell, 2025-07-01:
P.S./N.B. Be careful if you are relying on the initramfs (e.g. for exotic filesystems or drivers) in order to get to your root filing system - rpi-update won't rebuild that for you, you'd be better of waiting for the next kernel package in apt.

Ve2mrx, 2025-07-03:
Is this issue present in 6.12.34+rpt as well? I'm having PTP issues since I updated this morning...

Ve2mrx, 2025-07-10:
Do we have a rough ETA for the next kernel packages that fix this?

Steve-Tech, 2025-09-26:
6.12.47 looks to be packaged now in bookworm!