raspberrypi/linux #6952: brcmfmac: memcpy: detected field-spanning write (size 27) of single field "eventmask_msg->mask"

gizahNL, 2025-07-10:
### Describe the bug

I've enabled FORTIFY_SOURCE & HARDENED_USERCOPY in the interest of hardening our kernel build.
```
CONFIG_HARDENED_USERCOPY=y
CONFIG_FORTIFY_SOURCE=y
```

The kernel reports a field spanning write in the brcmfmac driver:

```
[    6.901916] ------------[ cut here ]------------
[    6.906580] memcpy: detected field-spanning write (size 27) of single field "eventmask_msg->mask" at /drivers/net/wireless/broadcom/brcm80211/brcmfmac/fweh.c:461 (size 1)
[    6.924713] WARNING: CPU: 1 PID: 51 at /drivers/net/wireless/broadcom/brcm80211/brcmfmac/fweh.c:461 brcmf_fweh_activate_events+0x294/0x2c0 [brcmfmac]
[    6.938228] Modules linked in: ecdh_generic ecc libaes raspberrypi_hwmon videobuf2_common gpu_sched brcmfmac_wcc spi_bcm2835 gpio_keys dwc2 brcmfmac drm_shmem_helper mc brcmutil ms_pwr_mgr(O) rp1(+) sdhci_of_dwcmshc spi_dw_mmio raspberrypi_gpiomem spi_dw rp1_adc bridge rp1_mailbox nvmem_rmem stp llc cfg80211 rfkill uio_pdrv_genirq uio sch_fq_codel qmi_wwan cdc_wdm fuse nfnetlink overlay
[    6.972704] CPU: 1 UID: 0 PID: 51 Comm: kworker/1:5 Tainted: G           O       6.12.25-v8-16k #1
[    6.981705] Tainted: [O]=OOT_MODULE
[    6.985201] Hardware name: Raspberry Pi Compute Module 5 Rev 1.0 (DT)
[    6.991666] Workqueue: events request_firmware_work_func
[    6.997006] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    7.003998] pc : brcmf_fweh_activate_events+0x294/0x2c0 [brcmfmac]
[    7.010243] lr : brcmf_fweh_activate_events+0x294/0x2c0 [brcmfmac]
[    7.016484] sp : ffffc00080903ac0
[    7.019805] x29: ffffc00080903ac0 x28: ffff800002fa4380 x27: ffff80000338c8e0
[    7.026975] x26: ffff800006676800 x25: ffff800006f292c4 x24: 000000000000001f
[    7.034144] x23: ffff800006672800 x22: ffffd06f84981918 x21: ffff800006f292c0
[    7.041311] x20: ffff8000062c7940 x19: 000000000000001b x18: 0000000000000000
[    7.048479] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[    7.055647] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    7.062815] x11: ffff80007fe04000 x10: 0000000000001a40 x9 : ffffd06fce4883a0
[    7.069983] x8 : ffff800002aee0a0 x7 : 0000000000000000 x6 : 000000000000b67e
[    7.077150] x5 : 000000000000b67e x4 : 0000000000001ab0 x3 : 0000000000000a04
[    7.084317] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff800002aec600
[    7.091486] Call trace:
[    7.093935]  brcmf_fweh_activate_events+0x294/0x2c0 [brcmfmac]
[    7.099826]  brcmf_cfg80211_attach+0x9a4/0x1368 [brcmfmac]
[    7.105367]  brcmf_attach+0x1c0/0x510 [brcmfmac]
[    7.110035]  brcmf_sdio_firmware_callback+0x7c0/0x980 [brcmfmac]
[    7.116102]  brcmf_fw_request_done+0x15c/0x1b0 [brcmfmac]
[    7.121554]  request_firmware_work_func+0x5c/0xb8
[    7.126277]  process_one_work+0x174/0x3f8
[    7.130306]  worker_thread+0x2dc/0x400
[    7.134070]  kthread+0x118/0x128
[    7.137308]  ret_from_fork+0x10/0x20
[    7.140897] ---[ end trace 0000000000000000 ]---

```

This is on 6.12.25 (1f6ab68a1d)

Sorry for the noise if this was already fixed (I couldn't find it)

### Steps to reproduce the behaviour

Enable fortify_source and boot a Pi CM5 

### Device (s)

Raspberry Pi CM5

### System

6.12.25-v8-16k

pelwell, 2025-07-10:
Have you tried the rpi-6.16.y branch? That function has changed since 6.12. However, I think it's somewhere between a false positive and a technicality - the field that it is being exceeded looks to be a single entry byte array which is used as a placeholder for a variable length array.

Either way, though, that looks like a bug in the upstream kernel, i.e. not one that we have introduced.

pelwell, 2025-07-10:
> I agree it's a technicality, the write is correct memory wise.

That would be something good to mention in a bug report.

closed 2025-07-10 by pelwell