raspberrypi/linux #6955: Kernel Upgrade to 6.12.34 causes Pi Zero 2 W to hang when running `gpsd`

agrif, 2025-07-13:
### Describe the bug

After upgrading my kernel from 6.1.21 to 6.12.34, my Pi Zero 2 W started hanging randomly after boot. I tracked it down to specifically the `gpsd` program accessing my GPS device attached over USB serial. As soon as `gpsd` opens the USB serial device, or very shortly after, the Pi stops responding to any input short of power cycling it. Just to be sure, I also tried on a fresh install of Raspberry Pi OS, and the problem still happens.

If I downgrade the kernel on the fresh SD card image to 6.1.21, the problem stops happening.

There is no panic or other error message in the serial console. The Pi simply stops responding, both over the network and on the serial console. Sometimes, but not always, the green status LED on the Pi blinks at about 1Hz after this freeze.

### Steps to reproduce the behaviour

1. Install Raspberry Pi OS with arm64 and kernel 6.12.34 (most recent as of today) to a Pi Zero 2 W.
2. `sudo apt-get install gpsd`
3. Plug in a Garmin 18x USB GPS
4. `gpsd -Nn -D2 /dev/ttyACM0`
5. Witness system freeze. Sometimes it works anyway -- shutdown, unplug everything, wait, and try again.

### Device (s)

Raspberry Pi Zero 2 W

### System

## /etc/rpi-issue
````
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage2
````

## vcgencmd version
````
Apr 30 2025 13:35:18
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)
````

## uname -a
````
Linux hyperion 6.12.34+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
````

### Logs

I have yet to see the kernel generate a message about this, anywhere.

### Additional context

When I downgrade to [this kernel (linux-image-6.1.0-rpi1-rpi-v8_6.1.21-1+rpt1)](https://archive.raspberrypi.org/debian/pool/main/l/linux/linux-image-6.1.0-rpi1-rpi-v8_6.1.21-1+rpt1_arm64.deb), my system doesn't freeze any more. I have not tried to determine which kernels work in the versions between, but if this is helpful I can try to do so.

On 6.12.34, the freeze can be avoided by first opening the USB serial device in another program, e.g. `picocom`, but this is not consistent. I can't begin to guess why `gpsd` triggers the freeze but `picocom` does not.

My GPS device is a Garmin 010-00321-31 18x. On `lsusb` it reports as `ID 04d8:00dd Microchip Technology, Inc. MCP2221(a) UART/I2C Bridge`. I'm fairly certain `gpsd` interacting with this device causes the freeze, since I can control when `gpsd` opens it. It runs fine, right up until it starts talking to this device.

I am very sure this is not a power supply issue. I have tried a few different supplies that should all be powerful enough, I've tried many cables, and my USB power meter does not indicate a draw above 400mA at any time before or after.

pelwell, 2025-07-13:
> I have not modified my cmdline.txt or config.txt

In that case have you installed a 64-bit OS, and if so, why? 

We can settle the question of which USB driver is active - what does `dmesg | grep -i -E '(dwc|usb)'` report?

agrif, 2025-07-13:
Adding `dtoverlay=dwc2,dr_mode=host` to my config.txt appears to have fixed this issue for me. Since it looks like dwc2 is going to be the default on 64-bit v8 kernels (now? soon?) this is probably The Fix. I'll close this issue, unless my pi starts freezing again.

I guess the only info to add here is that my pi appears to have booted just fine and enumerated usb devices, and even use usb in a limited capacity even though it was running a 64-bit v8 kernel and dwc_otg with FIQ enabled. The only other issue I've found seems to say booting doesn't even work under that configuration, so maybe that's news? Maybe not.

@pelwell, thank you for your help and getting me on the right trail here.