raspberrypi/linux #6962: 6.12.34 breaks Waveshare RS232 HAT

ronnybremer, 2025-07-17:
### Describe the bug

After upgrading from 6.12.20 to 6.12.34 the Waveshare overlay still gets loaded, however, communication via ttySC1 is no longer possible, the user app just freezes.

dmesg output for 6.12.34
`
[    4.693022] sc16is7xx spi1.0: Native CS is not supported - please configure cs-gpio in device-tree
[    4.697683] spi1.0: ttySC0 at I/O 0x0 (irq = 59, base_baud = 921600) is a SC16IS752
[    4.701597] spi1.0: ttySC1 at I/O 0x1 (irq = 59, base_baud = 921600) is a SC16IS752
`

dmesg output for 6.12.20
`
[    5.635965] spi1.0: ttySC0 at I/O 0x0 (irq = 44, base_baud = 921600) is a SC16IS752
[    5.636656] spi1.0: ttySC1 at I/O 0x1 (irq = 44, base_baud = 921600) is a SC16IS752
`

The config.txt snippet to load the overlay:
`
# Waveshare RS-232 HAT
dtoverlay=sc16is752-spi1,int_pin=24
`

### Steps to reproduce the behaviour

Upgrade kernel to 6.12.30, try to communicate with /dev/ttySC1, notice the app freezes and the activity LEDs on the Waveshare don't blink

Downgrade to kernel 6.12.20, communication is fine and activity LEDs blink as expected

### Device (s)

Raspberry Pi 4 Mod. B

### System

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Firmware:
Apr 30 2025 13:33:39
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)

Kernel (working):
6.12.20+rpt-rpi-v8

Kernel (failing):
6.12.34+rpt-rpi-v8

6by9, 2025-07-17:
#6865 is likely the cause.

6by9, 2025-07-19:
The easiest way to test a kernel PR on Raspberry Pi OS is to use `sudo rpi-update pulls/<PR-num>`, eg `sudo rpi-update pulls/6964` in this case.

I've just approved the CI builds, so they should be complete in about 40mins.

ronnybremer, 2025-07-23:
@6by9 Finally got my hands on a test device, they seem to be in high demand. Installed the latest kernel, didn't work as expected. Tried the `rpi-update` command, but it doesn't seem to work.

```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:2349daafacfb7a7abe2cfecf30a49ae837bdf2c6
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 6.12.34+rpt-rpi-v8
WANT_32BIT:0 WANT_64BIT:1 WANT_64BIT_RT:0 WANT_PI4:1 WANT_PI5:1
Downloading bootloader tools
Downloading bootloader images
 *** Downloading specific artifact revision (this will take a few minutes)
curl  -L https://builds.raspberrypi.com/github/linux/f1c00a6ce0508f721799985572499d0a4b2cdaf1/bcmrpi | zcat | tar xf - -C //root/.rpi-firmware --strip-components=2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0

gzip: stdin: unexpected end of file
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
Invalid artifact specified. Response: 404.
```

Did the pull request change?

ronnybremer, 2025-07-23:
Thank you @nbuchwitz, that explains it. I have tried your workaround by changing `/boot/firmware/config.txt` and can confirm, that serial communication is working again on the `6.12.34+rpt-rpi-v8` kernel. When the PR has rebuilt, I will check with that too.

MSRS-Shop, 2025-07-23:
.25 works for me. .34 killed my overlay. Anyone know when this will be fixed?
`dtoverlay=sc16is752-spi1,int_pin=24`

MSRS-Shop, 2025-07-24:
> Did you read the above (eg `sudo rpi-update pulls/6964` or the updated `config.txt` from above)?

That worked.

silverberry-rpi, 2025-08-03:
This fix my problem also. On both pi3b+ and pi5. Thank you very much.
However maybe this issue is unresolved now:
https://github.com/raspberrypi/linux/issues/6106
On pi5 the irq is using 33-34% cpu, no such problem on pi3b+.
   557 root     -51   0       0      0      0 D  34.2   0.0   0:15.72 irq/186-spi1.0
Maybe PR 6114 is not in this branch or something caused it to re-emerge?

pelwell, 2025-08-04:
> Maybe PR 6114 is not in this branch or something caused it to re-emerge?

That's an interesting theory, and not completely out of the question given the changes that have happened recently.

The overlay `sc16is752-spi1` no longer exists, but the same functionality should be available with no changes on the part of the user. That's because the firmware translates `dtoverlay=sc16is752-spi1,...` into `dtoverlay=sc16is75x-spi,sc16is752,spi1-0`. This results in the application of an overlay that looks like this:
```
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/interrupt-controller/irq.h>
#include <dt-bindings/pinctrl/bcm2835.h>

/ {
	compatible = "brcm,bcm2835";
	fragment@0 {
		target = <&spi1>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";
			sc16is752@0 {
				compatible = "nxp,sc16is752";
				reg = <0>;
				clocks = <&sc16is75x_clk>;
				interrupt-parent = <&gpio>;
				interrupts = <24 IRQ_TYPE_EDGE_FALLING>;
				pinctrl-0 = <&sc16is75x_pins>;
				pinctrl-names = "default";
				gpio-controller;
				#gpio-cells = <2>;
				spi-max-frequency = <4000000>;
			};
		};
	};
	fragment@1 {
		target-path = "/";
		__overlay__ {
			sc16is75x_clk: sc16is75x_clk {
				compatible = "fixed-clock";
				#clock-cells = <0>;
				clock-frequency = <14745600>;
			};
		};
	};
	fragment@2 {
		target = <&gpio>;
		__overlay__ {
			sc16is75x_pins: sc16is75x_spi1_0_pins {
				brcm,pins = <24>;
				brcm,function = <BCM2835_FSEL_GPIO_IN>;
				brcm,pull = <BCM2835_PUD_OFF>;
			};
		};
	};
};
```
Notice that `sc16is752@0` has the property `pinctrl-0 = <&sc16is75x_pins>;`, and that the node with the label `sc16is75x_pins` declares GPIO 24 as an input with no pull.

You can confirm whether this has worked using the following command (at least on Raspberry Pi OS - if not you'll have to install it from https://github.com/raspberrypi/utils/pinctrl):
```
pi@raspberrypi:~$ pinctrl 24
24: ip    pn | lo // GPIO24 = input
```
The failure case that caused #6106 would give the following output:
```
24: no pd | -- // PIN18/GPIO24 = none
```

nbuchwitz, 2025-08-05:
Both look ok (input, no pulls). Does the value stay high on pi5 or are there any changes? Eg. `sudo pinctrl poll 24`

Gemikro1, 2025-08-05:
I also had the issue with a not working Waveshare 2-Ch RS485/232 with a freshly installed SD image and was almost throwing the card to the trash when i found your post by hazard just in time.
I am using an old Pi 2B with kernel 6.12.34+rpt-rpi-v7 and can confirm that the modifications in the config.txt also fixed the issue here.
`pinctrl 24` returns stable
`24: ip    -- | hi // GPIO24 = input`

silverberry-rpi, 2025-08-06:
I did more investigation with a new clean pi5 and latest release:
Linux pi5mobile 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

I did not install the hat. Actually hat has nothing to do with any issue, it operates correctly on both pi5 and pi3b with the two dtoverlay statements:
[all]
dtoverlay=spi1-1cs,cs0_spidev=off
dtoverlay=sc16is75x-spi,sc16is752,spi1-0,int_pin=24

I checked for irq/186 after each stage of using
1. Neither overlay
#dtoverlay=spi1-1cs,cs0_spidev=off
#dtoverlay=sc16is75x-spi,sc16is752,spi1-0,int_pin=24
2. Just this
dtoverlay=spi1-1cs,cs0_spidev=off
3. Both
dtoverlay=spi1-1cs,cs0_spidev=off
dtoverlay=sc16is75x-spi,sc16is752,spi1-0,int_pin=24

Only after step 3 did the irq spin start. On pi3b it never occurs.

But the problem is deeper than that.  Someone else has same problem using different hat and overlays:
https://raspberrypi.stackexchange.com/questions/150169/why-irq-186-spi0-0-takes-25-of-cpu-time-a-rs485-hat

On my pi5 lsmod | grep spi gives:
6is7xx_spi          49152  0
regmap_spi             49152  1 sc16is7xx_spi
sc16is7xx              49152  1 sc16is7xx_spi
spidev                 49152  0
spi_bcm2835            49152  0
spi_dw_mmio            49152  0
spi_dw                 49152  1 spi_dw_mmio

There is another module in the kernel using spi. I tried to use modprobe -r to remove it but could not.

On pi3b the same command yields:
sc16is7xx_spi          12288  0
regmap_spi             12288  1 sc16is7xx_spi
sc16is7xx              24576  1 sc16is7xx_spi
spi_bcm2835aux         12288  0

I cant explain why pin 24 is different on pi5 vs pi3b/pi2 but it doesn't affect the hat function. It works properly on both.

I also can't explain why there is irq spin on pi5 but not pi3b.  But I  don't use rs485 hat on pi5 only pi3b, so I have no open problem.  However I  do hope the cause can be found.

                                                                                                                        

nbuchwitz, 2025-08-06:
top also doesn't show anything suspicious:

```
top - 08:39:09 up 20:31,  1 user,  load average: 0.00, 0.00, 0.00
Tasks:   1 total,   0 running,   1 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  0.0 sy,  0.0 ni,100.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
MiB Mem :   4049.9 total,   3658.9 free,    209.9 used,    246.8 buff/cache     
MiB Swap:    512.0 total,    512.0 free,      0.0 used.   3840.0 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                
    527 root     -51   0       0      0      0 S   0.0   0.0   0:00.00 [irq/171-spi1.0]  
```

silverberry-rpi, 2025-08-06:
> top also doesn't show anything suspicious:
>
> ```
> top - 08:39:09 up 20:31,  1 user,  load average: 0.00, 0.00, 0.00
> Tasks:   1 total,   0 running,   1 sleeping,   0 stopped,   0 zombie
> %Cpu(s):  0.0 [...]

Interesting.  Can you share the results of:
lsmod | grep spi

nbuchwitz, 2025-08-06:
```
$ lsmod | grep spi
sc16is7xx_spi          49152  0
regmap_spi             49152  1 sc16is7xx_spi
sc16is7xx              49152  1 sc16is7xx_spi
spidev                 49152  0
spi_bcm2835            49152  0
spi_dw_mmio            49152  0
spi_dw                 49152  1 spi_dw_mmio
```

pelwell, 2025-08-13:
Thanks for the feedback. I've merged that PR now, so the fix will appear in future releases.

The warning is expected because of how the change has been implemented; if you'd rather it went away, edit config.txt to replace `dtoverlay=sc16is752-spi1` with `dtoverlay=sc16is75x-spi,sc16is752,spi1-1cs,spi1-0`. Or, even better:
```
dtoverlay=spi1-1cs,cs0_spidev=off
dtoverlay=sc16is75x-spi,sc16is752,spi1-0
```

ronnybremer, 2025-08-13:
Thats what I will do. Just posted the warning for completeness.
Thanks for the merge!

closed 2025-08-13 by ronnybremer

papypom, 2025-09-13:
I'm using a WaveShare RS485 board (2-CH RS485 HAT) which uses the same SC16IS752 chip. I've applied the fix mentioned here https://github.com/raspberrypi/linux/issues/6962#issuecomment-3106429802

It works for a few minutes, then the system starts throwing Input/output errors (for any command entered) - seems also to overheat. Note : I cannot dig deeper at the moment, as the system is now unbootable.

System : Raspberry Pi 3B, Raspberry Pi OS Lite 64 Bits, Kernel 6.12.34

papypom, 2025-09-22:
Hi ! Try the comment https://github.com/raspberrypi/linux/issues/6962#issuecomment-3106429802 - it seems to work, but on some rapsberry pi, this could lead to overheating (see my previous comment).

Keep us posted !

balvie, 2025-09-24:
F.y.i.: I've made some progres with my dedicated dts for my situation. 

About the problem I've seen above: overheating the chip, can it be possible that the interupt configuaration is not perfect? If something with "interrupt=" in combination with "brcm,pull = " is not correct, the interrupt may be kept active and the driver continuesly tries to service it without the required effect, i.e. the reset of the interrupt (normally done by reading a data register of a transmit write action).

Also one has to study the hw details. The SC16IS752 has a open drain IRQ-line that need either a hw pull resistor on the pcb or a correct setting of the soc.

Could this be of any help?

nbuchwitz, 2025-09-24:
I will have a look later this week as I don't have the hardware with me. 

btw: This issue is closed so maybe @pelwell can re-open it or you can create a new issue with the current issue description? Not my call