raspberrypi/linux #6970: Raspberry Pi 3 does not see SD-Card under Kernel 6.15.3,  Raspberry Pi 4/5 does

mkreisl, 2025-07-23:
### Describe the bug

A Raspberry Pi 3 cannot boot from an SD card under kernel 6.15.3. It still works fine with kernel 6.12.x.
The strange thing is that a Raspberry Pi 4 and 5 can access the SD card without any problems with the 6.15.3 kernel.

I have not tested kernel 6.13 and 6.14.

Does something need to be changed in the kernel config so that it works again on the Pi 3?
The kernel config is used as usual from the respective previous versions.

Many thanks

### Steps to reproduce the behaviour

Install XBian and then this kernel

### Device (s)

Raspberry Pi 3 Mod. B

### System

XBian / Debian Bookworm

vcgencmd version
Oct 17 2023 15:43:26 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start_x)

popcornmix, 2025-07-25:
> Does something need to be changed in the kernel config so that it works again on the Pi 3?

not that we know about. But kernels newer than 6.12 get limited testing, so there may be an issue.
Info on whether 6.13, 6.14 and 6.16 work would be useful in narrowing things down.

popcornmix, 2025-07-25:
Can you confirm if this is an issue with your own local bcm2709_defconfig file, or the one from this repo?
It looks like bcm2709_defconfig for arm32 from this repo will set:
```
CONFIG_MMC_BCM2835_MMC=y
# CONFIG_MMC_SDHCI_IPROC is not set
```
which I think you are saying is correct.

popcornmix, 2025-07-25:
Okay, so nothing to do here.

closed 2025-07-25 by mkreisl