raspberrypi/linux #6971: pwm-pio OOPS when enabling on 6.12.34

rpdom, 2025-07-23:
### Describe the bug

When testing pwm-pio on a Pi 5 2GB running an up-to-date Raspberry Pi OS 64-bit Bookworm Lite, I encountered errors which killed the current process.

The dmesg report showed WARNINGS and an OOPS.

### Steps to reproduce the behaviour

Install Bookworm Lite using imager.

Run `sudo apt update && sudo apt full-upgrade`.

Enable pwm-pio in config.txt

`dtoverlay=pwm-pio`

Reboot.

Then the following command line sequence caused the problem
```
rpdom@raspberrypi:~ $ cd /sys/class/pwm/pwmchip1/
rpdom@raspberrypi:/sys/class/pwm/pwmchip1 $ cat npwm 
1
rpdom@raspberrypi:/sys/class/pwm/pwmchip1 $ echo "0" >export 
rpdom@raspberrypi:/sys/class/pwm/pwmchip1 $ cd pwm0
rpdom@raspberrypi:/sys/class/pwm/pwmchip1/pwm0 $ echo 50 >duty_cycle 
rpdom@raspberrypi:/sys/class/pwm/pwmchip1/pwm0 $ echo 1000 > period 
rpdom@raspberrypi:/sys/class/pwm/pwmchip1/pwm0 $ echo 1 > enable 
```
At this point the shell exited and dmesg showed the trace.

### Device (s)

Raspberry Pi 5

### System

https://pastebin.com/c3WcrYjV

### Logs

rpdom@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.34+rpt-rpi-2712 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-12 (Debian 12.2.0-14+deb12u1) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000003a000000..0x000000003dffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd16657 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] Faking node 0 at [mem 0x0000000000000000-0x000000001fffffff] (512MB)
[    0.000000] Faking node 1 at [mem 0x0000000020000000-0x000000003fffffff] (512MB)
[    0.000000] Faking node 2 at [mem 0x0000000040000000-0x000000005fffffff] (512MB)
[    0.000000] Faking node 3 at [mem 0x0000000060000000-0x000000007fffffff] (512MB)
[    0.000000] NUMA: Initialized distance table, cnt=4
[    0.000000] NODE_DATA(0) allocated [mem 0x1fffd200-0x1fffffff]
[    0.000000] NODE_DATA(1) allocated [mem 0x3fbfd200-0x3fbfffff]
[    0.000000] NODE_DATA(2) allocated [mem 0x5fffd200-0x5fffffff]
[    0.000000] NODE_DATA(3) allocated [mem 0x7fe9c080-0x7fe9ee7f]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[... 683 lines cut ...]
[  211.339234] x8 : 000000000000000a x7 : 00000000ffffffc9 x6 : 000000000000002a
[  211.346398] x5 : ffff800041061d02 x4 : 00000000ffffffd3 x3 : ffffd06fa2d5c538
[  211.353563] x2 : 0000000000000032 x1 : 00000000ffffffea x0 : 00000000000003e8
[  211.360728] Call trace:
[  211.363175]  rp1_pio_set_error+0x8/0x18 [rp1_pio]
[  211.367894]  __pwm_apply+0x70/0x1b8
[  211.371393]  enable_store+0xd8/0x108
[  211.374977]  dev_attr_store+0x20/0x40
[  211.378649]  sysfs_kf_write+0x4c/0x68
[  211.382320]  kernfs_fop_write_iter+0x128/0x200
[  211.386775]  vfs_write+0x248/0x388
[  211.390184]  ksys_write+0x7c/0x120
[  211.393592]  __arm64_sys_write+0x24/0x38
[  211.397524]  invoke_syscall+0x50/0x120
[  211.401283]  el0_svc_common.constprop.0+0x48/0xf0
[  211.406001]  do_el0_svc+0x24/0x38
[  211.409323]  el0_svc+0x30/0x100
[  211.412471]  el0t_64_sync_handler+0x100/0x130
[  211.416839]  el0t_64_sync+0x190/0x198
[  211.420510] Code: cf2192a8 ffffd06f aa1e03e9 d503201f (b9001401) 
[  211.426624] ---[ end trace 0000000000000000 ]---

### Additional context

Originally reported on the forums https://forums.raspberrypi.com/viewtopic.php?t=390233

nbuchwitz, 2025-07-23:
Seems to be around https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/pwm/pwm-pio-rp1.c#L118 (`ppwm->pio` is NULL). Candidate for a fix: https://github.com/raspberrypi/linux/pull/6972

rpdom, 2025-07-24:
> [@rpdom](https://github.com/rpdom) if you're willing to test the patch, you can do so with `sudo rpi-update pulls/6972`.
> 
> Make sure to have a recovery strategy / backup for the (unlikely) case that the kernel wont boot. I've tested it on a CM5 and everything seems to look ok.

That seems to have fixed the problem, thank you. :-)

nbuchwitz, 2025-07-28:
The issue is unrelated to Bluetooth (invalid argument is a generic error message). Also as far as I know Ubuntu builds their own Kernel, so the bug report is probably better placed with them. If the same error happens with the latest Raspberry Pi OS and kernel I would suggest to open a new issue here.