raspberrypi/linux #6976: OOM Killer & Kodi Crash While Playing A Video

dinobagio, 2025-07-28:
### Describe the bug

Playing a content from a local samba share on my Kodi (over arch arm, PI 4B),
After a few minutes, like half an hour on first occurrence, playing stops, Kodi restarts.

### Steps to reproduce the behaviour

Just the ordinary, play a content that lasts more than half an hour. Series or movie will do.

### Device (s)

Raspberry Pi 4 Mod. B

### System

````
Arch Linux ARM / 6.12.40-1-rpi arch: armv7l bits: 32
Linux kodi 6.12.40-1-rpi #1 SMP Fri Jul 25 05:18:01 MDT 2025 armv7l GNU/Linux

# kodi --version
Kodi Media Center 21.2 (21.2.0) Git:20250706-nogitfound
Copyright (C) 2005-2025 Team Kodi - http://kodi.tv

latest packages at time of last occurrence (i.e. logs below)

### Logs

````
Jul 27 21:35:36 kodi kernel: MHD-connection invoked oom-killer: gfp_mask=0xcc0(GFP_KERNEL), order=0, oom_score_adj=0
Jul 27 21:35:36 kodi kernel: CPU: 3 UID: 972 PID: 938 Comm: MHD-connection Tainted: G         C         6.12.40-1-rpi #1
Jul 27 21:35:36 kodi kernel: Tainted: [C]=CRAP
Jul 27 21:35:36 kodi kernel: Hardware name: BCM2711
Jul 27 21:35:36 kodi kernel: Call trace:
Jul 27 21:35:36 kodi kernel:  unwind_backtrace from show_stack+0x18/0x1c
Jul 27 21:35:36 kodi kernel:  show_stack from dump_stack_lvl+0x60/0x84
Jul 27 21:35:36 kodi kernel:  dump_stack_lvl from dump_header+0x50/0x180
Jul 27 21:35:36 kodi kernel:  dump_header from oom_kill_process+0x234/0x2a4
Jul 27 21:35:36 kodi kernel:  oom_kill_process from out_of_memory+0x100/0x350
Jul 27 21:35:36 kodi kernel:  out_of_memory from __alloc_pages_noprof+0xb30/0xf88
Jul 27 21:35:36 kodi kernel:  __alloc_pages_noprof from skb_page_frag_refill+0xdc/0x104
Jul 27 21:35:36 kodi kernel:  skb_page_frag_refill from sk_page_frag_refill+0x1c/0x94
Jul 27 21:35:36 kodi kernel:  sk_page_frag_refill from tcp_sendmsg_locked+0x214/0xccc
Jul 27 21:35:36 kodi kernel:  tcp_sendmsg_locked from tcp_sendmsg+0x30/0x48
Jul 27 21:35:36 kodi kernel:  tcp_sendmsg from __sock_sendmsg+0x44/0x78
Jul 27 21:35:36 kodi kernel:  __sock_sendmsg from ____sys_sendmsg+0x20c/0x234
Jul 27 21:35:36 kodi kernel:  ____sys_sendmsg from ___sys_sendmsg+0x9c/0xd0
Jul 27 21:35:36 kodi kernel:  ___sys_sendmsg from sys_sendmsg+0x78/0xbc
Jul 27 21:35:36 kodi kernel:  sys_sendmsg from ret_fast_syscall+0x0/0x5c
Jul 27 21:35:36 kodi kernel: Exception stack(0xf2351fa8 to 0xf2351ff0)
Jul 27 21:35:36 kodi kernel: 1fa0:                   00000000 00000000 00000045 7de825b0 00004000 00000000
Jul 27 21:35:36 kodi kernel: 1fc0: 00000000 00000000 a2b3f9f8 00000128 00000001 a19194f8 00000000 00000000
Jul 27 21:35:36 kodi kernel: 1fe0: 7de82560 7de8254c b4f36580 b4f44348
Jul 27 21:35:36 kodi kernel: Mem-Info:
Jul 27 21:35:36 kodi kernel: active_anon:7953 inactive_anon:36401 isolated_anon:0
                              active_file:213720 inactive_file:97563 isolated_file:0
                              unevictable:2643 dirty:82 writeback:0
                              slab_reclaimable:2025 slab_unreclaimable:7589
                              mapped:39085 shmem:2867 pagetables:622
                              sec_pagetables:0 bounce:0
                              kernel_misc_reclaimable:0
                              free:561989 free_pcp:257 free_cma:105350
Jul 27 21:35:36 kodi kernel: Node 0 active_anon:10912kB inactive_anon:166504kB active_file:165656kB inactive_file:1079480kB unevictable:10572kB isolated(anon):0kB isolated(file):0kB mapped:156340kB dirty:328kB writeback:0kB shmem:11468kB writeback_tmp:0kB kernel_stack:1928kB pagetables:2488kB sec_pagetables:0kB all_unreclaimable? no
Jul 27 21:35:36 kodi kernel: DMA free:427104kB boost:0kB min:3224kB low:4028kB high:4832kB reserved_highatomic:0KB active_anon:0kB inactive_anon:0kB active_file:2256kB inactive_file:0kB unevictable:0kB writepending:8kB present:786432kB managed:660340kB mlocked:0kB bounce:0kB free_pcp:920kB local_pcp:0kB free_cma:421400kB
Jul 27 21:35:36 kodi kernel: lowmem_reserve[]: 0 0 3188 3188
Jul 27 21:35:37 kodi kernel: DMA: 21*4kB (UEC) 94*8kB (UC) 308*16kB (UE) 8*32kB (C) 4*64kB (C) 3*128kB (C) 7*256kB (C) 0*512kB 1*1024kB (C) 0*2048kB 102*4096kB (C) = 427268kB
Jul 27 21:35:37 kodi kernel: 314165 total pagecache pages
Jul 27 21:35:37 kodi kernel: 0 pages in swap cache
Jul 27 21:35:37 kodi kernel: Free swap  = 0kB
Jul 27 21:35:37 kodi kernel: Total swap = 0kB
Jul 27 21:35:37 kodi kernel: 1012736 pages RAM
Jul 27 21:35:37 kodi kernel: 816128 pages HighMem/MovableOnly
Jul 27 21:35:37 kodi kernel: 31523 pages reserved
Jul 27 21:35:37 kodi kernel: 131072 pages cma reserved
Jul 27 21:35:37 kodi kernel: Tasks state (memory values in pages):
Jul 27 21:35:37 kodi kernel: [  pid  ]   uid  tgid total_vm      rss rss_anon rss_file rss_shmem pgtables_bytes swapents oom_score_adj name
Jul 27 21:35:37 kodi kernel: [    264]     0   264    13610     4980      211     4767         2   110592        0          -250 systemd-journal
Jul 27 21:35:37 kodi kernel: [    306]   977   306     4578     2866      416     2450         0    61440        0             0 systemd-resolve
Jul 27 21:35:37 kodi kernel: [    307]   976   307     5630     1634      128     1506         0    61440        0             0 systemd-timesyn
Jul 27 21:35:37 kodi kernel: [    315]     0   315     7297     1977      320     1657         0    65536        0         -1000 systemd-udevd
Jul 27 21:35:37 kodi kernel: [    321]   980   321     3579     2076      192     1884         0    53248        0             0 systemd-network
Jul 27 21:35:37 kodi kernel: [    446]    81   446     1761      864      128      736         0    40960        0          -900 dbus-broker-lau
Jul 27 21:35:37 kodi kernel: [    447]    81   447      826      511       91      420         0    32768        0          -900 dbus-broker
Jul 27 21:35:37 kodi kernel: [    449]     0   449     2947     1858      128     1730         0    53248        0             0 hostapd
Jul 27 21:35:37 kodi kernel: [    450]   102   450    12818     2005      192     1813         0    73728        0             0 polkitd
Jul 27 21:35:37 kodi kernel: [    451]     0   451     1682     1394      160     1234         0    40960        0         -1000 sshd
Jul 27 21:35:37 kodi kernel: [    452]     0   452     3369     1793      195     1598         0    49152        0             0 systemd-logind
Jul 27 21:35:37 kodi kernel: [    454]     0   454    12951     1737      224     1513         0    81920        0             0 upowerd
Jul 27 21:35:37 kodi kernel: [    486]   972   486      957      715       32      683         0    28672        0             0 kodi-standalone
Jul 27 21:35:37 kodi kernel: [    487]   968   487   151750     5517     1984     3533         0   122880        0             0 prometheus-node
Jul 27 21:35:37 kodi kernel: [    521]   972   521     4974      797      301      496         0    57344        0             0 (sd-pam)
Jul 27 21:35:37 kodi kernel: [  24488]     0 24488     3573     2469      288     2181         0    53248        0             0 sshd-session
Jul 27 21:35:37 kodi kernel: [  24491]  1000 24491     3613     1694      380     1314         0    53248        0             0 sshd-session
Jul 27 21:35:37 kodi kernel: [  24492]  1000 24492     1150      941      224      717         0    28672        0             0 bash
Jul 27 21:35:37 kodi kernel: [  24509]     0 24509     3573     2463      288     2175         0    53248        0             0 sshd-session
Jul 27 21:35:37 kodi kernel: [  24513]  1000 24513     4473     2585     1275     1310         0    65536        0             0 sshd-session
Jul 27 21:35:37 kodi kernel: [  24514]  1000 24514     2007      898      128      770         0    40960        0             0 sftp-server
Jul 27 21:35:37 kodi kernel: [  31394]   972 31394      957      668       32      636         0    28672        0             0 kodi
Jul 27 21:35:37 kodi kernel: [  31399]   972 31399   252419    64191    36138    28053         0  1224704        0             0 kodi.bin
Jul 27 21:35:37 kodi kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=kodi.bin,pid=31399,uid=972
Jul 27 21:35:37 kodi kernel: Out of memory: Killed process 31399 (kodi.bin) total-vm:1009676kB, anon-rss:144552kB, file-rss:112212kB, shmem-rss:0kB, UID:972 pgtables:1196kB oom_score_adj:0
Jul 27 21:35:37 kodi kodi-standalone[31394]: /usr/bin/kodi: line 215: 31399 Killed                     ${KODI_BINARY} ${ENV_ARGS} $SAVED_ARGS
Jul 27 21:35:38 kodi kodi-standalone[947]: [E] pw.loop [loop.c:70 pw_loop_new()] 0x62f9e28: can't make support.system handle: No such file or directory

dinobagio, 2025-07-28:
Some comments:
 - This started just some days ago. And I've been using this Kodi for at least a year.
 - Once it happens, then it repeats more quickly.
 ex: 
```
Jul 22 19:16:26 kodi kernel: Out of memory: Killed process  532 (kodi.bin) total-vm:1561568kB, anon-rss:397140kB, file-rss:112348kB, shmem-rss:0kB, UID:972 pgtables:1724kB oom_score_adj:0
Jul 22 20:00:53 kodi kernel: Out of memory: Killed process 1543 (kodi.bin) total-vm:1538540kB, anon-rss:415464kB, file-rss:111648kB, shmem-rss:0kB, UID:972 pgtables:1760kB oom_score_adj:0
Jul 22 20:15:55 kodi kernel: Out of memory: Killed process 2969 (kodi.bin) total-vm:1502444kB, anon-rss:427716kB, file-rss:111588kB, shmem-rss:0kB, UID:972 pgtables:1740kB oom_score_adj:0
```

dinobagio, 2025-07-28:
It didn't happen before July for sure,
Here is all package movements since first of July:

```
[2025-07-01T08:27:42+0000] [ALPM] upgraded raspberrypi-bootloader (20250625-1 -> 20250630-1)
[2025-07-01T08:27:48+0000] [ALPM] upgraded linux-rpi (6.12.34-1 -> 6.12.35-1)
[2025-07-01T08:27:52+0000] [ALPM] warning: /etc/sudoers installed as /etc/sudoers.pacnew
[2025-07-01T08:27:53+0000] [ALPM] upgraded sudo (1.9.16.p2-2.1 -> 1.9.17.p1-1)
[2025-07-03T17:32:04+0000] [ALPM] upgraded openssl (3.5.0-1 -> 3.5.1-1)
[2025-07-03T17:32:04+0000] [ALPM] upgraded ca-certificates-mozilla (3.113-1 -> 3.113.1-1)
[2025-07-03T17:32:04+0000] [ALPM] upgraded libpng (1.6.49-1 -> 1.6.50-1)
[2025-07-03T17:32:05+0000] [ALPM] upgraded fontconfig (2:2.17.0-2 -> 2:2.17.1-1)
[2025-07-03T17:32:08+0000] [ALPM] upgraded jq (1.8.0-1 -> 1.8.1-1)
[2025-07-03T17:32:08+0000] [ALPM] upgraded leancrypto (1.4.0-1 -> 1.5.0-1)
[2025-07-03T17:32:09+0000] [ALPM] upgraded libheif (1.19.8-3 -> 1.20.1-1)
[2025-07-03T17:32:24+0000] [ALPM] upgraded llvm-libs (20.1.6-3 -> 20.1.7-1)
[2025-07-03T17:32:25+0000] [ALPM] upgraded nss (3.113-1 -> 3.113.1-1)
[2025-07-05T13:50:27+0000] [ALPM] upgraded findutils (4.10.0-2 -> 4.10.0-3)
[2025-07-05T13:50:28+0000] [ALPM] upgraded gsfonts (20200910-4 -> 20200910-5)
[2025-07-05T13:50:28+0000] [ALPM] upgraded hwdata (0.396-1 -> 0.397-1)
[2025-07-05T13:50:28+0000] [ALPM] upgraded libjxl (0.11.1-3 -> 0.11.1-4)
[2025-07-07T18:48:17+0000] [ALPM] upgraded python-pillow (11.2.1-1 -> 11.3.0-1)
[2025-07-07T18:48:17+0000] [ALPM] upgraded raspberrypi-bootloader (20250630-1 -> 20250707-1)
[2025-07-07T18:48:22+0000] [ALPM] upgraded linux-rpi (6.12.35-1 -> 6.12.36-1)
[2025-07-07T18:48:27+0000] [ALPM] upgraded kodi-rpi (21.2-14 -> 21.2-15)
[2025-07-07T18:48:30+0000] [ALPM] upgraded sdl2-compat (2.32.56-1 -> 2.32.56-2)
[2025-07-09T06:21:54+0000] [ALPM] upgraded archlinux-keyring (20250430.1-1 -> 20250707-1)
[2025-07-09T06:22:24+0000] [ALPM] upgraded ldb (2:4.22.2-1 -> 2:4.22.3-1)
[2025-07-09T06:22:24+0000] [ALPM] upgraded libwbclient (2:4.22.2-1 -> 2:4.22.3-1)
[2025-07-09T06:22:26+0000] [ALPM] upgraded mesa (1:25.1.4-1 -> 1:25.1.5-1)
[2025-07-09T06:22:27+0000] [ALPM] upgraded smbclient (2:4.22.2-1 -> 2:4.22.3-1)
[2025-07-09T09:18:08+0000] [ALPM] upgraded gettext (0.25-1 -> 0.25.1-1)
[2025-07-09T09:18:10+0000] [ALPM] upgraded git (2.50.0-1 -> 2.50.1-1)
[2025-07-11T21:53:32+0000] [ALPM] upgraded e2fsprogs (1.47.2-2 -> 1.47.3-1)
[2025-07-11T21:53:32+0000] [ALPM] upgraded debugedit (5.1-1 -> 5.2-1)
[2025-07-11T21:53:32+0000] [ALPM] upgraded graphite (1:1.3.14-4 -> 1:1.3.14-5)
[2025-07-11T21:53:32+0000] [ALPM] upgraded linux-firmware-whence (20250627-1 -> 20250708-1)
[2025-07-11T21:53:34+0000] [ALPM] upgraded linux-firmware-amdgpu (20250627-1 -> 20250708-1)
[2025-07-11T21:53:49+0000] [ALPM] upgraded linux-firmware-atheros (20250627-1 -> 20250708-1)
[2025-07-11T21:54:00+0000] [ALPM] upgraded linux-firmware-broadcom (20250627-1 -> 20250708-1)
[2025-07-11T21:54:01+0000] [ALPM] upgraded linux-firmware-cirrus (20250627-1 -> 20250708-1)
[2025-07-11T21:54:37+0000] [ALPM] upgraded linux-firmware-intel (20250627-1 -> 20250708-1)
[2025-07-11T21:54:41+0000] [ALPM] upgraded linux-firmware-mediatek (20250627-1 -> 20250708-1)
[2025-07-11T21:54:59+0000] [ALPM] upgraded linux-firmware-nvidia (20250627-1 -> 20250708-1)
[2025-07-11T21:55:07+0000] [ALPM] upgraded linux-firmware-other (20250627-1 -> 20250708-1)
[2025-07-11T21:55:08+0000] [ALPM] upgraded linux-firmware-radeon (20250627-1 -> 20250708-1)
[2025-07-11T21:55:10+0000] [ALPM] upgraded linux-firmware-realtek (20250627-1 -> 20250708-1)
[2025-07-11T21:55:10+0000] [ALPM] upgraded linux-firmware (20250627-1 -> 20250708-1)
[2025-07-11T21:55:10+0000] [ALPM] upgraded raspberrypi-bootloader (20250707-1 -> 20250711-1)
[2025-07-11T21:55:19+0000] [ALPM] upgraded linux-rpi (6.12.36-1 -> 6.12.37-1)
[2025-07-13T21:10:30+0000] [ALPM] upgraded readline (8.2.013-2 -> 8.3.0-3)
[2025-07-13T21:10:30+0000] [ALPM] upgraded bash (5.2.037-5 -> 5.3.0-1)
[2025-07-13T21:10:31+0000] [ALPM] upgraded gnutls (3.8.9-1 -> 3.8.10-1)
[2025-07-13T21:10:31+0000] [ALPM] upgraded libwebp (1.5.0-1 -> 1.6.0-1)
[2025-07-14T22:00:20+0000] [ALPM] upgraded readline (8.3.0-3 -> 8.3.001-1)
[2025-07-14T22:00:20+0000] [ALPM] upgraded automake (1.18-1 -> 1.18.1-1)
[2025-07-14T22:00:21+0000] [ALPM] upgraded git (2.50.1-1 -> 2.50.1-3)
[2025-07-14T22:00:58+0000] [ALPM] upgraded leancrypto (1.5.0-1 -> 1.5.1-1)
[2025-07-14T22:00:59+0000] [ALPM] upgraded graphviz (13.0.1-1 -> 13.1.0-1)
[2025-07-14T22:01:00+0000] [ALPM] upgraded wireless-regdb (2025.02.20-1 -> 2025.07.10-1)
[2025-07-18T19:04:57+0000] [ALPM] upgraded libass (0.17.3-1 -> 0.17.4-1)
[2025-07-18T19:04:58+0000] [ALPM] upgraded libogg (1.3.5-2 -> 1.3.6-1)
[2025-07-18T19:04:58+0000] [ALPM] upgraded sdl3 (3.2.16-1 -> 3.2.18-1)
[2025-07-18T19:15:28+0000] [ALPM] installed hostapd (2.11-4)
[2025-07-18T19:21:18+0000] [ALPM] installed bridge-utils (1.7.1-2)
[2025-07-23T12:52:35+0000] [ALPM] installed prometheus-node-exporter (1.9.1-2)
[2025-07-23T13:09:08+0000] [ALPM] installed gnu-netcat (0.7.1-10)
[2025-07-23T19:43:24+0000] [ALPM] upgraded libldap (2.6.10-1 -> 2.6.10-2)
[2025-07-23T19:43:25+0000] [ALPM] upgraded sqlite (3.50.2-1 -> 3.50.3-1)
[2025-07-23T19:43:25+0000] [ALPM] upgraded ca-certificates-mozilla (3.113.1-1 -> 3.114-1)
[... 11 lines cut ...]
[2025-07-23T19:43:56+0000] [ALPM] upgraded x264 (3:0.164.r3108.31e19f9-2 -> 3:0.164.r3108.31e19f9-3)
[... cut ...]
[2025-07-23T19:45:24+0000] [ALPM] upgraded perl-timedate (2.33-7 -> 2.33-8)
[2025-07-23T19:45:24+0000] [ALPM] upgraded perl-mailtools (2.22-1 -> 2.22-2)
[2025-07-23T19:45:24+0000] [ALPM] upgraded tevent (1:0.16.2-1 -> 1:0.17.0-1)
[2025-07-25T20:15:15+0000] [ALPM] upgraded bash (5.3.0-1 -> 5.3.3-1)
[2025-07-25T20:15:18+0000] [ALPM] upgraded libcec (7.0.0-2 -> 7.1.1-1)
[2025-07-25T20:15:20+0000] [ALPM] upgraded raspberrypi-bootloader (20250717-1 -> 20250725-1)
[2025-07-25T20:15:34+0000] [ALPM] upgraded linux-rpi (6.12.38-1 -> 6.12.40-1)
```

dinobagio, 2025-07-29:
Alright @pelwell I guess I'll have to wait for a commit on https://github.com/archlinuxarm/PKGBUILDs (linux-rpi), usually made by @graysky.

graysky2, 2025-07-29:
> Alright ***@***.***(https://github.com/pelwell) I guess I'll have to wait for a commit on https://github.com/archlinuxarm/PKGBUILDs (linux-rpi), usually made by ***@***.***(https://github.com/graysky2).

Will update later today

popcornmix, 2025-07-29:
Yes, the `buff/cache` value from `free` is expected to increase to consume all free memory with time - that is the file cache, and it will be dropped when memory is needed elsewhere. 

I wouldn't expect hostapd to be used by kodi, but it's possible if hostapd (or any other process) was leaking memory, that kodi could be killed by OOM killer just because it is naturally a significant user of memory.

dinobagio, 2025-07-29:
> I wouldn't expect hostapd to be used by kodi, but it's possible if hostapd (or any other process) was leaking memory, that kodi could be killed by OOM killer just because it is naturally a significant user of memory.

Indeed. No use of hostapd for Kodi. I installed it for another need by that time. This need is over. I removed the package. And now that I think of it problem first came out after I installed hostapd. Maybe it was the 'bad guy'. Lets see how it goes from now.

graysky2, 2025-07-29:
https://github.com/archlinuxarm/PKGBUILDs/commit/0ac815fa73b1e31400694be7d2bb85f12734c4c1