raspberrypi/linux #6982: kernel-side crash for a bad CSI device: v4l2-subdev.c:457 call_s_stream+0xe8/0x110 [videodev]

xrdelectro, 2025-07-31:
### Describe the bug

Posting in case someone might be interested in having a look at the crash dump and making kernel-side code more robust against faulty (or badly soldered) units.

some kernel module crash seemed to be provoked when a badly soldered imx290 CSI camera unit was attached and `libcamera-hello` was run.

As far as I understood/remember, a weird edge case was provoked, where same subdev release/deallocation occured more times than necessary when after a failure to initialize HW... or something along those lines.

I have fixed resoldered/fixed my camera unit, so probably won't be reproducing this issue for some time now.

What I noticed as well - an issue stopped to occur when I recompiled at that time latest rpicam-apps and libcamera and installed that on top of the latest updated 6.12.y system (between 7th and 11th of July), I think that from the logs you should see all the versions needed.

Cheers,
Zigmārs

### Steps to reproduce the behaviour

hardly possible - must have bad/misconfigured hardware (XMASTER for IMX291 was set incorrectly - as slave), might have had some MIPI lane connections missing.

### Device (s)

Raspberry Pi 5

### System

This is an old issue I have fixed for myself, so only way is to look at the versions dumped in logs.

### Logs

```sh
$ LIBCAMERA_LOG_LEVELS=*:DEBUG libcamera-hello
[0:02:40.540027976] [1140] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[0:02:40.541292897] [1140] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[0:02:40.613992816] [1140] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[0:02:40.614047151] [1140] DEBUG IPAManager ipa_manager.cpp:229 Loaded IPA module '/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[0:02:40.614069875] [1140]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
[0:02:40.614190194] [1145] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:02:40.620483858] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media2
[0:02:40.620586677] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 10 missing dependencies
[0:02:40.621649072] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:02:40.621658387] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rp1-cfe
[0:02:40.621726186] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpi-hevc-dec" created from /dev/media3
[0:02:40.621736649] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:02:40.621848969] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:02:40.621857024] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpi-hevc-dec
[0:02:40.621935157] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[0:02:40.621945176] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 8 missing dependencies
[0:02:40.622027068] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:02:40.622036791] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 8 missing dependencies
[0:02:40.622838861] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:02:40.622847454] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[0:02:40.623637912] [1145] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:02:40.623645949] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:02:40.623930701] [1145] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[0:02:40.625878725] [1145] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:02:40.625899337] [1145] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:02:40.625914504] [1145] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[0:02:40.625931135] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[0:02:40.625962302] [1145] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
[0:02:40.626053677] [1145] DEBUG ClockRec clock_recovery.cpp:69 configure 100 2000 10 50000
[0:02:40.626071622] [1145] DEBUG ClockRec clock_recovery.cpp:83 reset
[0:02:40.626222369] [1145]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[0:02:40.637526587] [1145] DEBUG CameraSensor camera_sensor_raw.cpp:210 imx290 10-001a: unsupported number of sinks (0) or sources (1)
[0:02:40.637579164] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Exposure (0x00980911)
[0:02:40.637606405] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Horizontal Flip (0x00980914)
[0:02:40.637624777] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Vertical Flip (0x00980915)
[0:02:40.637643944] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Camera Orientation (0x009a0922)
[0:02:40.637666112] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Camera Sensor Rotation (0x009a0923)
[0:02:40.637684649] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Vertical Blanking (0x009e0901)
[0:02:40.637702020] [1145] DEBUG V4L2 v4l2_device.cpp:721 'imx290 10-001a': Control: Horizontal Blanking (0x009e0902)
[... 432 lines cut ...]
[16437.806220]  vb2_core_queue_release+0x2c/0x88 [videobuf2_common]
[16437.812510]  _vb2_fop_release+0x7c/0x90 [videobuf2_v4l2]
[16437.818102]  vb2_fop_release+0x34/0x68 [videobuf2_v4l2]
[16437.823605]  v4l2_release+0xec/0x100 [videodev]
[16437.828421]  __fput+0xd0/0x2e0
[16437.831743]  ____fput+0x1c/0x30
[16437.835150]  task_work_run+0x78/0xe0
[16437.838995]  do_exit+0x2e8/0x9c0
[16437.842492]  do_group_exit+0x3c/0xa0
[16437.846336]  get_signal+0x9e0/0x9e8
[16437.850094]  do_signal+0x94/0x10d0
[16437.853763]  do_notify_resume+0xd0/0x150
[16437.857958]  el0_svc+0xd8/0x100
[16437.861366]  el0t_64_sync_handler+0x13c/0x158
[16437.865995]  el0t_64_sync+0x190/0x198
[16437.869927] ---[ end trace 0000000000000000 ]---
[19487.345160] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19488.561441] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19489.745152] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19490.929203] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19492.117125] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19493.297106] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19494.477119] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
[19495.661109] rp1-cfe 1f00110000.csi: Using a link rate of 594 Mbps
```

6by9, 2025-08-04:
It's a catch for CSI2 receiver drivers that have called `s_stream(true)` to start streaming when already streaming, or `s_stream(false)` to stop streaming when already stopped. You've obviously found a path in the CFE driver that can achieve that.

Without some more guidance on how to reproduce this, there is limited chance to find that path. It would appear to be in the error recovery path, but a number of things can trigger that.

I'm also conscious that this is on our downstream version of the CFE driver. The CFE driver has been upstreamed to mainline Linux now, but currently has a few limitations so we haven't adopted it. Tracking down issues in the downstream driver that will hopefully get thrown away within a year is of limited gain.

6by9, 2025-08-07:
Amusingly I've just hit the same WARNING. It appears to be that the "no frame received" timeout in libcamera manages to trigger it fairly reliably with imx219.

It doesn't appear to affect ov5647, so I was expecting that it was only sensors using embedded metadata that triggered it, but that wouldn't apply to the imx290 (at least the version in mainline and our kernel). And now double checking I see that the imx219 kernel driver has dropped embedded metadata in 6.12. Count me confused!