raspberrypi/linux #6983: Pi 5 + IMX219: Sensor detected in dmesg, but not registered in media graph (regression in 6.12.34 / libcamera 0.5.1)

bialy41, 2025-07-31:
### Describe the bug

### Summary

After a kernel update on July 29, 2025, my Raspberry Pi 5 stopped detecting an IMX219 camera module in the media pipeline. The sensor is visible on I²C and correctly identified in `dmesg`, and the kernel module loads successfully. However, no media graph nodes are created, and `rpicam-hello` reports "no cameras available".

This setup worked correctly before the kernel update. Reverting to either auto-detect or legacy overlay configuration does not restore functionality.

---

### Hardware

- Raspberry Pi 5 (8GB)
- RP1 I/O chip
- Camera: 3rd-party 8MP IMX219 wide-angle module (FOV 120°)
- Connection: standard 200mm MIPI CSI ribbon
- No additional HATs or overlays involved

---

### Software

- OS: Raspberry Pi OS Bookworm 64-bit
- Kernel: `6.12.34+rpt-rpi-2712`
- Bootloader: `Thu 8 May 14:13:17 UTC 2025` (confirmed latest via `rpi-eeprom-update`)
- libcamera: `0.5.1+rpt20250722-1`
- rpicam-apps: rebuilt from source against current system

---

### Timestamps / Evidence of Regression

$ ls -l /boot/firmware/kernel8.img 
-rwxr-xr-x 1 root root 9958826 Jul 29 16:18 /boot/firmware/kernel8.img

- Camera worked **before** this timestamp
- Broke **immediately after**

### What works
- IMX219 is correctly detected in `dmesg`:
rp1-cfe: Using sensor imx219 11-0010 for capture
- `imx219` kernel module loads
- Sensor appears on I²C at address 0x10
- `/dev/video*` and `/dev/media*` show only `pispbe`, `rpi-hevc-dec` nodes

### What fails
- No sensor or CSI/ISP nodes in media graph (checked via `media-ctl`)
- `rpicam-hello` and `rpicam-hello --list-cameras` return:
ERROR: *** no cameras available ***

Changing boot/config - Neither `camera_auto_detect=1` (default) nor legacy as below change anything:
camera_auto_detect=0
dtoverlay=imx219

Additional Verification:
/sys/class/video4linux/video*/name only lists decoder and pispbe outputs
Rebuilt rpicam-apps from source against libcamera 0.5.1 — no change
Confirmed /boot/firmware/config.txt is clean; no conflicting overlays

This appears to be a regression introduced between kernel 6.12.XX (pre-July 29) and 6.12.34+rpt-rpi-2712. It breaks RP1-based IMX219 sensor registration into the media graph. Sensor is correctly detected on I²C and initialized, but libcamera cannot discover it due to missing graph nodes.

Please advise on patch status, workaround, or if a specific DT binding or alias is now required.

### Steps to reproduce the behaviour

IMX219 Device on CSI => rpicam-hello

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-11-19
version 69471177 
Linux pi01 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

6by9, 2025-08-07:
You haven't provided the output from `dmesg` to see what parts of the media graph have succeeded or failed, so it's difficult to diagnose what part has failed.
Also of use would be `cat /sys/kernel/debug/v4l2-async/pending_async_subdevices` and `cat /sys/kernel/debug/devices_deferred`

I've just tested a Pi5 with `6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26)` and an imx219 manually configured with `dtoverlay=imx219`, and it works fine.

bialy41, 2025-08-07:
Thanks, especially for testing the physical setup.
I captured dmesg with: 
dmesg | grep -i -e imx -e rp1 -e csi -e cam

Key lines:
[   12.220292] rp1-cfe 1f00128000.csi: found subdevice /axi/pcie@1000120000/rp1/i2c@80000/imx219@10
[   12.516505] rp1-cfe 1f00128000.csi: Using sensor imx219 11-0010 for capture
[   12.516598] rp1-cfe 1f00128000.csi: Registered [...] as /dev/video0 ... /dev/video7
[dmesg.log](https://github.com/user-attachments/files/21670006/dmesg.log)

Even though the camera is detected and registered, rpicam-hello still reports: ERROR: *** no cameras available ***

From /sys/kernel/debug/v4l2-async/pending_async_subdevices:

imx219 11-0010:
rp1-cfe 1f00128000.csi:

So it looks like the async binding between the CSI and the sensor never completes, despite the subdevices being present.

cat /sys/kernel/debug/devices_deferred returns nothing.

boot/firmware/config.txt is currently at:
camera_auto_detect=0
dtoverlay=imx219

Let me know if you'd like a full media-ctl dump or to test a patch. I'm holding off on a kernel upgrade to keep this state reproductable.

bialy41, 2025-08-07:
Thanks. Here’s the full output of media-ctl -p -d /dev/media0 
[media-ctl.txt](https://github.com/user-attachments/files/21670479/media-ctl.txt)

the sensor (IMX219) is correctly detected, and the link to CSI2 is enabled:
imx219 11-0010 -> csi2:0 [ENABLED,IMMUTABLE]

Downstream links (e.g., csi2 → rp1-cfe-csi2_ch0 → /dev/video0) are also present and valid.

At this point I'm not sure how to proceed, as everything looks connected but the camera still isn't usable.

6by9, 2025-08-07:
That looks correct. Which partly points at a libcamera issue rather than the kernel.

@naushir What's the easiest way of getting logging for `--list-cameras`? Setting `LIBCAMERA_LOG_LEVELS=*:DEBUG` gives nothing.

bialy41, 2025-08-07:
Thanks to both of you!
Here’s the output of LIBCAMERA_LOG_LEVELS=*:DEBUG rpicam-hello:
[rpicam-hello.txt](https://github.com/user-attachments/files/21671573/rpicam-hello.txt)

Key part i notice here is:
[167:50:10.628802504] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/pisp'
[167:50:10.628897671] DEBUG RPI pisp.cpp:890 Unable to acquire a CFE instance

It looks like the rpi/pisp pipeline handler is detected, but fails to acquire a CFE instance even though media-ctl shows the media graph is fully built and /dev/video* nodes are registered. So indeed this might be a libcamera issue rather than kernel-side.

Please let me know if there’s a way to trace the failure in rpi/pisp.cpp more precisely or test any possible fix.

closed 2025-08-08 by bialy41