raspberrypi/linux #6998: Unmet build dependencies: libssl-dev

frabervo, 2025-08-13:
### Describe the bug

Hello everyone, 
I 'm cross building the kernel 6.14 for the raspberry PI4 (aarch64). For this purpose i used the .config from the Ubuntu server 25.04. The goal is to have a custom kernel with Rust support enabled and compile the samples Rust kernel modules such rust-minimal.rs. 

I have an issue, when i try to build the source and binary deb kernel packages to install the custom kernel on the raspberry PI4. 

Host:  
- Architecture: amd64
- OS: Ubuntu 24.04.2 LTS
- System information: `Linux vm138 6.8.0-71-generic #71-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64 GNU/Linux`

Source of the .config file: `Linux ubuntu 6.14.0-1005-raspi #5-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr  5 14:33:30 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux`

Command, that generate the error: ` sudo make LLVM=1 -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- deb-pkg`

**Error**:
```shell
  SYNC    include/config/auto.conf.cmd
  GEN     debian
dpkg-architecture: warning: specified GNU system type aarch64-linux-gnu does not match CC system type x86_64-pc-linux-gnu, try setting a correct CC environment variable
dpkg-buildpackage --build=source,binary --no-pre-clean --unsigned-changes --unsigned-source --compression=gzip -R'make -f debian/rules' -j1 -a$(cat debian/arch)
dpkg-buildpackage: info: source package linux-upstream
dpkg-buildpackage: info: source version 6.14.11-g5c1d35bb54c5-18
dpkg-buildpackage: info: source distribution noble
dpkg-buildpackage: info: source changed by root <root@vm138>
dpkg-architecture: warning: specified GNU system type aarch64-linux-gnu does not match CC system type x86_64-pc-linux-gnu, try setting a correct CC environment variable
dpkg-buildpackage: info: host architecture arm64
 dpkg-source --compression=gzip --before-build .
dpkg-source: info: using options from linux/debian/source/local-options: --diff-ignore --extend-diff-ignore=.*
dpkg-checkbuilddeps: error: Unmet build dependencies: libssl-dev
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
make[2]: *** [scripts/Makefile.package:126: deb-pkg] Error 3
make[1]: *** [/home/fbat/Documents/linux-kernel/linux/Makefile:1625: deb-pkg] Error 2
make: *** [Makefile:251: __sub-make] Error 2
````

### Steps to reproduce the behaviour

- Cross compile the linux kernel for the raspberry PI 4 based on the following repository: 
`git clone --depth=1 --branch rpi-6.10.y https://github.com/raspberrypi/linux`
- build the source and binary deb kernel packages to install the custom kernel on the raspberry PI4
```shell 
make LLVM=1 -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- deb-pkg
``` 

### Device (s)

Other

### System

Linux vm138 6.8.0-71-generic #71-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 22 16:52:38 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

pelwell, 2025-08-13:
What makes you think that the dependencies should have been met automatically? Which step do you think should be doing so?

frabervo, 2025-08-13:
- Depedencies 
```shell
# Dependencies
sudo apt install bc bison flex libssl-dev make libc6-dev libncurses5-dev debhelper pkg-config libelf-dev llvm clang lld
# Toolchain for 64 bit arm
sudo apt install crossbuild-essential-arm64
```
- Source: `git clone --depth=1 --branch rpi-6.14.y https://github.com/raspberrypi/linux`
- .config file see attached file
- Set the old configuration
```shell
KERNEL=kernel8
make -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- olddefconfig
```
- set `LOCALVERSION` to `-v1-raspi`
- Disable Signature
```shell 
scripts/config --disable SYSTEM_TRUSTED_KEYS 
scripts/config --disable SYSTEM_REVOCATION_KEYS

#Following configuration will afterwards looks like this: 
# CONFIG_SYSTEM_TRUSTED_KEYS is not set
# CONFIG_SYSTEM_REVOCATION_KEYS is not set
``` 
- check that there is no + sign in the kernel release name: `make -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- kernelrelease`
- if there is a + sign in the kernel release name, comment out the part responsible for setting scm version string in `scripts/setlocalversion`
```shell 
# scm version string if not at the kernel version tag or at the file_localversion
if grep -q "^CONFIG_LOCALVERSION_AUTO=y$" include/config/auto.conf; then            # comment this line out
        # full scm version string
    scm_version="$(scm_version)"                                                    # comment this line out
elif [ "${LOCALVERSION+set}" != "set" ]; then                                       # comment this line out
        # If the variable LOCALVERSION is not set, append a plus
        # sign if the repository is not in a clean annotated or
        # signed tagged state (as git describe only looks at signed
        # or annotated tags - git tag -a/-s).
        #
        # If the variable LOCALVERSION is set (including being set
        # to an empty string), we don't want to append a plus sign.
    scm_version="$(scm_version --short)"                                            # comment this line out
fi                                                                                  # comment this line out
``` 
- Activate the Rust Support
```shell 
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
# In the linux kernel directory 
rustup override set $(scripts/min-tool-version.sh rustc)
cargo install --locked --version $(scripts/min-tool-version.sh bindgen) bindgen-cli
rustup component add rust-src
make LLVM=1 -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-  rust-analyzer
```
- Disable `MODVERSIONS` 
-  activate the Rust kernel modules samples:  
```shell 
CONFIG_SAMPLES_RUST=y
CONFIG_SAMPLE_RUST_MINIMAL=m
CONFIG_SAMPLE_RUST_MISC_DEVICE=m
CONFIG_SAMPLE_RUST_PRINT=m
CONFIG_SAMPLE_RUST_DRIVER_PLATFORM=m
```
- Build
```shell
make LLVM=1 -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
``` 
- Build both source and binary deb kernel packages
```shell 
make LLVM=1 -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- deb-pkg
```