raspberrypi/linux #7000: System hangs on some GUI windows

Luminisc, 2025-08-13:
### Describe the bug

I have fresh new Raspberry Pi 5, with freshly installed OS and all updates (firmware, OS, packages).
I installed Pi-Apps, and when I opening 'Games' category and scroll list of items for some time, whole system hangs, only hard reboot helps.

My issue is similar to https://github.com/raspberrypi/linux/issues/6872, but with exceptions:
1. I have Pi-Apps
2. System hangs not immediately, but only in specific scenario and application
3. I had v3d errors only twice in around 10 tries
4. Usually I have zero messages in `dmesg -wk` and `journalctl -f` when problem happens
5. Even if I do `dmesg | tee ~/dmesg.log` (and same for journalctl) - no messages are written during problem.

I tried:
- freshly installed OS - same problem
- installed on 2 SD cards and 1 NVMe SSD - same

Original issue in Pi-Apps repo (there are some other details too): https://github.com/Botspot/pi-apps/issues/2820

P.S. Same behavior when I managed to install Steam from Pi-Apps CLI (`~/pi-apps/manage install Steam`), but when I start Steam and it start to download update - system hangs like with Pi-Apps.

### Steps to reproduce the behaviour

1. Install and open Pi-Apps
2. Open 'Games' category
3. Scroll a bit (~ 5 seconds)
4. System fully hangs, become totally unresponsive.

### Device (s)

Raspberry Pi 5

### System

>$ cat /etc/rpi-issue 

Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage4

>$ vcgencmd version

2025/05/08 15:13:17 
Copyright (c) 2012 Broadcom
version 69471177 (release) (embedded)

>$ uname -a

Linux rpi 6.12.34+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

>$ rpi-eeprom-update 

BOOTLOADER: up to date
   CURRENT: Thu  8 May 14:13:17 UTC 2025 (1746713597)
    LATEST: Thu  8 May 14:13:17 UTC 2025 (1746713597)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.

Luminisc, 2025-08-13:
> How much ram do you have on the Pi5? And how much is in use when it starts to hang?
> 
> Without further evidence (e.g. errors in dmesg) , this doesn't sound like a kernel issue. It is possible to bring linux to a grinding halt if too much memory is used.

8GB, and htop showing no more than 500MB used at any time (on OS start, after app run, when app hangs), but tested only with Pi-Apps, not tested with Steam.

Is there any way to get some information from system? Some more verbose logging? Because nor dmesg not journalctl showing any messages. Not even in files if direct output to file. I thought maybe it is problem with rendering only and then dmesg/jctl at least should have write something in file, but still nothing there.  

Or serial console via UART is only option? As I am pretty new to RPi stuff, I don't know how can I debug such behaviors, so I need assistance.

Luminisc, 2025-08-13:
> uart would be the ideal way to get debug info. Can you ssh in before the issue? Is ssh still operational after the hang?

I can ssh before, but ssh stops to work when system hangs. Same with Raspberry Connect - it gets disconnected after few seconds after hanging

popcornmix, 2025-08-13:
Are you overclocking?
Is power supply good enough?

From ssh run `while : ; do vcgencmd get_throttled; done` and report if it is non-zero before the hang.

Luminisc, 2025-08-13:
> Do you ever see this hang without using Pi-Apps? e.g. when using browser (perhaps on a 3d heavy site like https://webglsamples.org/aquarium/aquarium.html)
(For some reason your link doesn't work, chromium getting timeouts or something about 'QUIC protocol'), I was searching for results.

First time stuck immediately, second time - it tried to render models, loaded some of the textures, but then black screen instead of page and again - system full hanging and ssh getting timeout. 
But finally some information - V3D_ERR_STAT, like in attached issue.

Here is latest logs from: dmesg, journalctl, get_throttled, and htop
[image]

Luminisc, 2025-08-14:
Does anybody have ideas how to debug this? I don't want to buy and wait for another RPi :(

Luminisc, 2025-08-16:
`over_voltage_delta=50000` allowed to open chromium, open Aquarium page, sit there for a few seconds after loading all textures, but then again - full system hang. And zero information this time, not in journalctl, not in dmesg.

[image]