raspberrypi/linux #7012: No atomic I2C transfer handler when using ssd1306

fyc09, 2025-08-24:
### Describe the bug

I use ssd1306 driver provided by kernel using "dtoverlay=ssd1306". After shutting down and message "Reaching target Power-Off", it repeatedly warns "No atomic I2C transfer handler for i2c-0" and power does not really shutdown.

### Steps to reproduce the behaviour

* Connect a SSD1306 OLED with i2c interface
* Add "dtoverlay=ssd1306" at the end of /boots/firmware/config.txt
* Add fbcon=map:1 into cmdline.txt to enable fbcon on the oled
* Shutdown

### Device (s)

Raspberry Pi Zero

### System

OS:
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f548f25325e28bea61904f5038a9b9fd0c92b07b, stage2

Firmware:
Apr 30 2025 13:35:18
Copyright (c) 2012 Broadcom
version 5560078dcc8591a00f57b9068d13e5544aeef3aa (clean) (release) (start)

Kernel:
Linux raspberrypi 6.12.25+rpt-rpi-v6 #1 Raspbian 1:6.12.25-1+rpt1 (2025-04-30) armv6l GNU/Linux

### Logs

I don't know how to copy the shutdown log, but i found [a similar problem on banana pi](https://forum.armbian.com/topic/20312-banana-pi-dont-shutdown-kernel-panic/)

fyc09, 2025-08-24:
I'm not sure why the [raspberrypi version](https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/gpu/drm/solomon/ssd130x.c) is slightly different from the [Torvalds version](https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/solomon/ssd130x.c). Is this possible to be reason?

6by9, 2025-08-26:
As long as the device tree bindings are equivalent, then drivers/video/fbdev/ssd1307fb.c ought to get disabled. 
At least staging/fbtft is using `solomon,ssd1306` as the compatible to make it different.

pelwell, 2025-08-26:
> As long as the device tree bindings are equivalent, then drivers/video/fbdev/ssd1307fb.c ought to get disabled.

Are you sure? `drivers/gpu/drm/solomon/ssd130x-i2c.c` says:
```
		/* Deprecated but kept for backward compatibility */
		.compatible = "solomon,ssd1306fb-i2c",
```

6by9, 2025-08-27:
The one niggle with using the DRM driver is that there is no fixed binding between the card and /dev/fbN node. In my testing I did get about a 50:50 split as to which display driver loaded first and claimed /dev/fb0.

There is `dtparam=drm_fb0_vc4` which should force vc4 (HDMI outputs etc) to be /dev/fb0, and then ssd1306 would always get /dev/fb1.

fyc09, 2025-08-27:
On my Raspberry Pi Zero, the ssd1306 ALWAYS get /dev/fb1 while /dev/fb0 is not found, though I didn't connect HDMI outputs.

closed 2025-09-03 by pelwell