raspberrypi/linux #7013: Camera Module v2 (imx219) missing 206FPS mode

mbway, 2025-08-24:
### Describe the bug

[The documentation](https://www.raspberrypi.com/documentation/accessories/camera.html) for the camera modules lists 640×480p 206fps as a supported mode for the camera module v2 but when I use `rpicam-vid --list-cameras` the 206FPS mode is not available, replaced with 103FPS instead:

```shell
pi@pi3:~/camera $ uname -a
Linux pi3 6.12.41-v8+ #1897 SMP PREEMPT Fri Aug  1 12:20:46 BST 2025 aarch64 GNU/Linux
pi@pi3:~/camera $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
```

I found this bug report https://github.com/raspberrypi/linux/issues/5438 that was resolved around the time of the 6.1 kernel. So I built that version and booted into it. With that kernel version the 206FPS mode was listed:

```shell
pi@pi3:~/camera $ uname -a
Linux pi3 6.1.93-v8+ #1 SMP PREEMPT Mon Aug 11 03:31:21 BST 2025 aarch64 GNU/Linux
pi@pi3:~/camera $ rpicam-vid --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
```

So it looks like there has been a regression since that issue was originally fixed.

### Steps to reproduce the behaviour

boot into the latest Raspberry Pi OS with kernel 6.12, plug in a Camera v2 module and run `rpicam-vid --list-cameras`. Observe that the 206FPS mode is not listed.

### Device (s)

Raspberry Pi 3 Mod. B

### System

```
pi@pi3:~/camera $ cat /etc/rpi-issue
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage4

pi@pi3:~/camera $ vcgencmd version
Jul 17 2025 17:32:49 
Copyright (c) 2012 Broadcom
version a668b6e6edce3274de221324b93cb8741e4a7f7c (clean) (release) (start)

pi@pi3:~/camera $ uname -a
Linux pi3 6.1.93-v8+ #1 SMP PREEMPT Mon Aug 11 03:31:21 BST 2025 aarch64 GNU/Linux
```

NWalker4483, 2025-08-24:
Looking for a solution to this also

mbway, 2025-09-04:
Some further investigation:

Support was first added to version 5.15 in https://github.com/raspberrypi/linux/pull/4880 and based on https://github.com/raspberrypi/linux/pull/5439 I think there was a regression between 6.0 and 6.1 that was fixed before the end of 6.1, then 6.7 is the first branch where the fix can no longer be found. So versions 5.15 to 6.6 all have the fix.

I'm not sure how much of the driver is developed in upstream Linux vs here. The `rpi-6.12.y` version of `imx219.c` looks
fairly different to the upstream 6.12 version. The upstream 6.12 version also does not contain the fix. This commit https://github.com/torvalds/linux/commit/f513997119f481a3a3bb9a0a8fc55b6cd50a0940 implements the fix into the upstream driver which is present for for versions 6.15 onwards.

It looks like this fix could be back-ported to the raspberry pi 6.12 version of the driver.