raspberrypi/linux #7014: CM5 Lite excessive reboot/shutdown delay if SD card is not present

mairas, 2025-08-25:
### Describe the bug

Discussed on this forum post: https://forums.raspberrypi.com/viewtopic.php?t=383051

A device using a CM5 Lite (in my case, [HALPI2](https://shop.hatlabs.fi/products/halpi2-computer)) and an NVMe will have a delay of up to 80 seconds at shutdown or reboot if an SD card is not inserted. With an SD card inserted (doesn't need to be formatted, just present) shutdowns are immediate.

### Steps to reproduce the behaviour

With a CM5 Lite inserted on a HALPI2 or (as reported in the forum thread) the standard CM5 IO Board:

- Flash on OS on an NVMe drive
- Ramp up the kernel logging: `systemd.log_level=debug systemd.log_target=kmsg log_buf_len=1M enforcing=0 ignore_loglevel`
- Issue command: `sudo reboot`
- The system will rapidly perform other reboot steps and then remain stuck in "Powering off" for a long time.

### Device (s)

Raspberry Pi CM5 Lite

### System

Raspinfo output attached.

[raspinfo.log](https://github.com/user-attachments/files/21967241/raspinfo.log)

### Logs

Didn't have UART logging set up, so there's just a bad photo. The system is stuck here for 10-80 seconds or even more.

[image]

mairas, 2025-08-26:
> Related to / same as [#6647](https://github.com/raspberrypi/linux/issues/6647) ?

Definitely related! In this case, the device just hangs waiting on mmc0 instead of mmc1. However, the required fix would be different. The SD card interface has valid uses even if it's not the boot disk. @pelwell's suggestion of reducing the kernel retry duration would be one possible approach, although I have no idea what side effects it might have.

Having said that, I have tried using `dtparam=sd=off` and that is a valid workaround for this issue. I will be enabling that for our pre-configured images - customers that want to use the SD card will have to manually enable the interface.

GamingTux, 2026-10-24:
### Hardware Root Cause & Fix Found (Preserves 100% SD Card Hotplug & Boot!)

We investigated this on a **CM5 Lite Rev 1.0** (Argon ONE UP laptop booting from NVMe with an empty microSD slot, kernel `6.12.62+rpt-rpi-2712`) by sampling `/sys/kernel/debug/mmc0/ios` and `/sys/kernel/debug/mmc0/err_stats` at 500 ms intervals during `mmc_rescan` and `reboot`.

---

### 1. Why it hangs for 20–80s (and occasionally reboots instantly)
Because CM5 Lite uses `broken-cd` on `mmc0` (`1000fff000.mmc`, `brcm,bcm2712-sdhci`), `mmc_rescan` polls the empty slot every second across `freqs[] = { 400000, 300000, 200000, 100000 }` in `drivers/mmc/core/core.c`.

By watching `/sys/kernel/debug/mmc0/ios` alongside `/sys/kernel/debug/mmc0/err_stats` in real time, we found a dramatic hardware difference between **`400 kHz`** and **`< 400 kHz`** (`300 kHz`, `200 kHz`, `100 kHz`) on `1000fff000.mmc`:

* **At `400 kHz`:**
  * Every probe command to the empty slot triggers the hardware `SDHCI_INT_TIMEOUT` interrupt in **~0.8 ms**.
  * `# Command Timeout Occurred` increments cleanly, and **`# Request Timedout` stays `0`**.
  * The entire `400 kHz` probe completes in **~35 ms**.
* **At `300 kHz`, `200 kHz`, and `100 kHz`:**
  * Once `sdhci_set_clock()` steps down below `392 kHz` (where the 10-bit V3 clock divider touches the upper bits in `SDHCI_DIV_HI_MASK` at `0x2C`, followed by 8-bit `SDHCI_SOFTWARE_RESET` writes at `0x2F` in the same 32-bit word), the BCM2712 Arasan SDHCI controller **stops asserting `SDHCI_INT_TIMEOUT`** on unconnected CMD lines.
  * Instead of a `0.8 ms` hardware timeout interrupt, **every single command hangs until the 10.0-second software watchdog timer (`sdhci_timeout_timer`) expires**:
    `mmc0: Timeout waiting for hardware cmd interrupt` (`# Request Timedout` increments by 4 per frequency = **40 seconds per frequency**!).
  * If `reboot` or `shutdown` is triggered while the `kworker` is between rescans or in the `400 kHz` phase, the system reboots in **2 seconds**. If `reboot` catches `mmc_rescan` inside the `300 / 200 / 100 kHz` sweep, shutdown blocks uninterruptibly for **20–80+ seconds** waiting for the remaining 10-second `sdhci_timeout_timer` expirations!

---

### 2. Live Telemetry Comparison (`/sys/kernel/debug/mmc0/err_stats`)

**Unpatched kernel (`400 / 300 / 200 / 100 kHz`) after 2 minutes with empty SD slot:**
* `# Command Timeout Occurred`: `84`
* `# Request Timedout`: **`130`** (130 × 10.0s software timeouts!)
* Reboot delay with empty SD slot: **60–85 seconds**

**With `mmc_rescan` clamped to `400 kHz` (`f_min = 400000`) + `no-sdio; no-mmc;` on `&sdio1`:**
* `# Command Timeout Occurred`: `820` (clean 0.8 ms HW interrupts)
* `# Request Timedout`: **`0`**
* Reboot delay with empty SD slot: **0 seconds** (~2s shutdown, full reboot cycle from `sudo reboot` -> EEPROM -> kernel -> Wi-Fi -> SSH login takes **21.0 seconds total**), while **preserving 100% SD card hotplug and SD boot** without `dtparam=sd=off`!

---

### 3. Suggested Upstream Fix for `raspberrypi/linux`

1. **Prevent `1000fff000.mmc` (`brcm,bcm2712-sdhci`) from dropping below `400 kHz` during `mmc_rescan`:**
   `mmc_rescan_try_freq()` already clamps the scan frequency via `max(freqs[i], host->f_min)`. Setting `host->f_min = 400000;` for `brcm,bcm2712-sdhci` (or fixing the 10-bit divider / `SDHCI_SOFTWARE_RESET` interaction in `sdhci-iproc.c`) completely eliminates the 10-second `sdhci_timeout_timer` stalls while remaining well within the SD specification (`100–400 kHz` initialization range).
2. **Add `no-sdio;` and `no-mmc;` to `&sdio1` on CM5 / CM5 Lite:**
   Since `&sdio1` (`1000fff000.mmc`) on CM5 Lite is wired to the external microSD card slot (not an SDIO Wi-Fi chip or eMMC), adding `no-sdio; no-mmc;` skips useless `CMD5` (SDIO) and `CMD1` (eMMC) probes on every 1-second poll, cutting empty-slot commands from 13 down to 4 (`CMD8` + `ACMD41` only).

---

### 4. Drop-in Workaround for Current Kernels (No Kernel Recompile Needed)

For anyone affected right now who wants fast reboots **without** losing the SD card slot (`dtparam=sd=off`):

**Step A: Device Tree Overlay (`cm5-sdcard-fast.dts`)**
```dts
/dts-v1/;
/plugin/;
/ {
    compatible = "brcm,bcm2712";
    fragment@0 {
        target = <&sdio1>;
        __overlay__ {
            no-sdio;
            no-mmc;
        };
    };
};
```
Compile and enable in `/boot/firmware/config.txt`:
```bash
sudo dtc -@ -I dts -O dtb -o /boot/firmware/overlays/cm5-sdcard-fast.dtbo cm5-sdcard-fast.dts
echo "dtoverlay=cm5-sdcard-fast" | sudo tee -a /boot/firmware/config.txt
```

**Step B: Patch `freqs[]` in `kernel_2712.img` / `kernel8.img` to `400 kHz` (`/etc/kernel/postinst.d/z99-cm5-mmc-fix`)**
```python
#!/usr/bin/python3
import gzip, glob, os, shutil, struct, tempfile

OLD_FREQS = struct.pack("<IIII", 400000, 300000, 200000, 100000)
NEW_FREQS = struct.pack("<IIII", 400000, 400000, 400000, 400000)

paths = sorted(set(glob.glob("/boot/vmlinuz-*") + glob.glob("/boot/firmware/kernel*.img")))
for p in paths:
    if not os.path.isfile(p) or os.path.islink(p):
        continue
    try:
        with gzip.open(p, "rb") as f:
            raw = f.read()
    except Exception:
        continue
    if OLD_FREQS not in raw:
        continue
    patched = raw.replace(OLD_FREQS, NEW_FREQS, 1)
    fd, tmp = tempfile.mkstemp(dir=os.path.dirname(p))
    os.close(fd)
    with gzip.open(tmp, "wb", compresslevel=6) as f:
        f.write(patched)
    shutil.copymode(p, tmp)
    os.replace(tmp, p)
    print(f"z99-cm5-mmc-fix: Patched mmc_rescan freqs[] in {p}")
```
Save as `/etc/kernel/postinst.d/z99-cm5-mmc-fix`, make executable (`sudo chmod 755 /etc/kernel/postinst.d/z99-cm5-mmc-fix`), and run it once. Every subsequent `apt upgrade` will automatically patch new kernels.

timg236, 2026-10-24:
This commit might resolve the issue
https://github.com/PineappleBeech/linux/commit/8e578c34bc9b213e89aff80309231ee6deaed7c3