raspberrypi/linux #7018: 7" touchscreen not working when drivers are compiled in into the kernel (Rpi5)

nooploop, 2025-08-26:
### Describe the bug

I'm trying to build a kernel with compiled in drivers for 7" official touchscreen (V1) for Rpi5. But there seems to be some i2c initialization issue and the touchscreen fails to initialize. When compiling touchscreen related drivers as modules and loading them manually after boot everything works.

### Steps to reproduce the behaviour

Build kernel without loadable modules and support for the 7" touchscreen: [config_kern.txt](https://github.com/user-attachments/files/21992984/config_kern.txt)
---> touchscreen does not work:  [dmesg_kern.txt](https://github.com/user-attachments/files/21993000/dmesg_kern.txt)

Build kernel with loadable modules: [config_mod.txt](https://github.com/user-attachments/files/21993025/config_mod.txt)
and loading the modules manually afterwards: [lsmod_mod.txt](https://github.com/user-attachments/files/21993028/lsmod_mod.txt)
---> touchscreen works: [dmesg_mod.txt](https://github.com/user-attachments/files/21993033/dmesg_mod.txt)

### Device (s)

Raspberry Pi 5

### System

kernel 6.12.42 (made with buildroot)

6by9, 2025-08-27:
Building the kernel with your config results in the system not fully booting. A diff between that and our normal defconfig shows many more changes than just switching from modules to built-in.

Trying the simplistic edit of .config to search for `=m` and replace with `=y` throws a wobbler in the build and restarts the config.

Taking our defconfig and switching the relevant modules to being built in is still giving me a working display and touch screen.
[config_working.txt](https://github.com/user-attachments/files/22005817/config_working.txt)
[kernel_log.txt](https://github.com/user-attachments/files/22005818/kernel_log.txt)
Based on lsmod I don't believe I've missed any modules out.

6by9, 2025-08-27:
> Yes, the config deviates from the defconfig in the kernel. I stripped it down by a lot, because i don't need a lot of stuff. 

Having done so you've significantly increased the pain in trying to reproduce your issue, particularly as initially you hadn't said so.
No network stack or NVMe make life very awkward.

Having reenabled NVME, CGROUPS, and AUTOFS for a standard system, I can boot your config.

> But the only change between the two configs i supplied is enabling modules and making touchscreen related drivers modules:
> 
> ```
> CONFIG_TOUCHSCREEN_EDT_FT5X06=m
> CONFIG_I2C_BRCMSTB=m
> CONFIG_I2C_DESIGNWARE_CORE=m
> CONFIG_I2C_DESIGNWARE_PLATFORM=m
> CONFIG_REGULATOR_RASPBERRYPI_TOUCHSCREEN_ATTINY=m
> CONFIG_DRM=m
> CONFIG_DRM_KMS_HELPER=m
> CONFIG_DRM_DISPLAY_HELPER=m
> CONFIG_DRM_TTM=m
> CONFIG_DRM_VRAM_HELPER=m
> CONFIG_DRM_TTM_HELPER=m
> CONFIG_DRM_GEM_DMA_HELPER=m
> CONFIG_DRM_GEM_SHMEM_HELPER=m
> CONFIG_DRM_SCHED=m
> CONFIG_DRM_PANEL_SIMPLE=m
> CONFIG_DRM_TOSHIBA_TC358762=m
> CONFIG_DRM_V3D=m
> CONFIG_DRM_VC4=m
> CONFIG_DRM_RP1_DSI=m
> CONFIG_DRM_PANEL_ORIENTATION_QUIRKS=m
> CONFIG_SND_SOC_HDMI_CODEC=m
> ```
> 
> And then loading them manually after boot.

That's a significant number of options that you've changed all at once. Changing one at a time would have been more useful.

However I have now reproduced this. It looks like pinctrl for GPIOs 38&39 or 40&41 (depending on which CAM/DISP connector you're using) haven't been set to the relevant alt functions for the I2C controller. They're still inputs.
Loading as a module has just changed the probe order slightly, so the pinctrl driver has loaded before the I2C controller.

@pelwell I would have expected to get to get -EPROBE_DEFER if the pinctrl driver wasn't ready at the point that the I2C controller probed. Is that wrong?

6by9, 2025-08-27:
I took the config_kern.txt from the 1st post as `.config`, and then used `make menuconfig` to enable CONFIG_NVME_CORE=y, CONFIG_CGROUPS=y, and CONFIG_AUTOFS=y in order to get it to boot on a Raspberry Pi OS image on an NVMe drive.
Nothing was specified as a module.

NB That config still doesn't have a network stack, so I was using a `[gpio2=0]` block in config.txt to switch back to an alternate kernel.img in order to do each update.

nooploop, 2025-08-27:
I did some more testing. Starting from my config_kern.txt and just enabling modules `CONFIG_MODULES=y`, but still keeping everything as compiled in, i get the screen properly initialized. I guess this option also changes the probing order somehow.