raspberrypi/linux #7020: enable_uart=1 does not cause corresponding GPIOs to be busy

webmeister, 2025-08-27:
### Describe the bug

I set `enable_uart=1` in `/boot/config.txt` to enable access to a serial console on GPIO14+15. This works perfectly fine. However, the kernel sees those GPIOs still as "unused", even though they are actively used by the UART:

```
$ lsgpio -n gpiochip0
GPIO chip: gpiochip0, "pinctrl-bcm2711", 58 GPIO lines
[...]
        line 14: "GPIO14" unused [input]
        line 15: "GPIO15" unused [input]
```

This means that I can execute something like `gpioset GPIO15=low` and break UART functionality. I would rather expect gpioset to complain `gpioset: unable to request lines on chip '/dev/gpiochip0': Device or resource busy` as it does for other GPIOs when they are already in use (e.g. when using `dtoverlay=gpio-shutdown,gpio_pin=26` and trying to execute `gpioset GPIO26=low`).

### Steps to reproduce the behaviour

1. Set `enable_uart=1` in `/boot/config.txt` and reboot
2. Run `gpioset GPIO15=low`
3. Observe that the command succeeds unexpectedly and UART functionality is now broken

### Device (s)

Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B

### System

Arch Linux ARM

```
$ vcgencmd version
May 14 2025 12:23:36
Copyright (c) 2012 Broadcom
version 17084b403fb60475b8ee2641c26049a7d54bf153 (clean) (release) (start)
```

```
$ uname -a
Linux rpi 6.12.43-1-rpi #1 SMP PREEMPT Tue Aug 26 13:22:07 MDT 2025 aarch64 GNU/Linux
```

pelwell, 2025-08-27:
The GPIO and pinctrl spaces are currently independent. It should be possible to enable "strict" mode, but numerous things would be likely to break and require attention - there might even be some use cases relying on the flexibility - so the motivation to fix this is very low.

webmeister, 2025-08-27:
> I could write some application that just opens the GPIO without doing anything with it

Or maybe I can't? As soon as I open one of the UART GPIOs from some application, the UART functionality is broken and can only be restored by rebooting (not by closing the GPIO or the application). I'd guess this is due to the GPIO hardware getting reconfigured (since when opening a GPIO, I have to specify whether to open it as input or output) and I don't know how to restore the original configuration used by the UART.

Any other idea how I can block all applications from opening specific GPIOs?