raspberrypi/linux #7025: MMC problems on Pi3 after change to upstream driver

spockfish, 2025-08-30:
### Describe the bug

I've got at least 2 customers who reported a crash with the latest kernels. It happens only on the Pi3. One of my customers can reproduce it by pulling the power. Within a few attempts the system runs into filesystem problems that start with the message:

```
sdhost-bcm2835: ... unexpected command 25 error
sdhost-bcm2835: ... previous command (25) not complete
sdhost-bcm2835: ... previous command (25) not complete
...
```

Followed by a bunch of filesystem related errors about journal problems and superblock updates. 

Now, I wrote a ticket before (#6981), which I also closed myself as I thought it was a filesystem bug. And as I was not able to reproduce it myself (I even bought a 3A just to be sure I got the same hardware setup as my customer), so I closed the ticket and moved on.

Recently another customer reported the same issue and it bothered me enough to start searching again. Because one of my customers is able to reproduce it 100% I started bisecting stuff. Everything was ok until 6.12.30, and after that things went wrong.

Some more digging pointed me to this post: https://forums.raspberrypi.com/viewtopic.php?t=389169

Which describes exactly the problem as I'm seeing it, except that I don't have this weird upstream/downstream mix: my kernel is downstream only. 

But then I noticed the remark about the change to the upstream MMC driver, specifically this commit: https://github.com/raspberrypi/linux/commit/cb6b027dcd8c3bc50009e6c653fd84cc76f49369

Long story short: reverting this commit fixes the issue. 

### Steps to reproduce the behaviour

Pull the plug, and within a few attempts (normally between 1 and 5) the issue appears.

### Device (s)

Raspberry Pi 3 Mod. A+

### System

6.12.42

custom built OS with buildroot, based on downstream defconfig with very little changes.

spockfish, 2025-09-01:
* happening during boot. it happens on a random partition (both EXT4), and there's doesn't seem to be a particular pattern. It can be triggered by pulling the power cord, and after that booting up. Within 5 attempts it fails.

This is done on a system with (virtually) no write action(s) to the SD card. 

Transcend Ultimate (600x) 8GB and Kensington Industrial Grade 8GB MicroSD both result in failure.

pelwell, 2025-09-02:
It would help me to understand the symptoms a bit more.

Simply pulling the plug on a running system is asking for trouble, but its not clear whether you are seeing corruption or a transient fault. Are you saying that there is some chance (20% say) that at any boot the card might unusable in some way, but it might work after trying again? Or that after 5-6 attempts the card is effectively broken/corrupted?

BTW, I don't see the linked forum thread as being the same issue at all - that was about a mismatched set of software.

spockfish, 2025-09-02:
We're talking about RoPieee, a media streamer solution for the Pi. It runs in RAM, and only writes stuff to the SD card when changing a config setting (which in this case is not what the customer is doing). The OS is built with buildroot, boots with u-boot and then starts a service (like Roon Bridge for example). The kernel is custom, based on your kernel (latest stable, 6.12.x in this case) with a very minimal set of changes on top of the standard defconfig. Think of setting a specific governor, preemptive, scheduler and disabling a few things that are enabled in the defconfig and simply not needed. So fairly standard, and certainly no fancy stuff.

> Simply pulling the plug on a running system is asking for trouble, but its not clear whether you are seeing corruption or a transient fault.

Agreed. And the logs show that there's a FS (Ext4) corruption (which can be fixed with a FSCK), *after* the MMC unexpected errors. That's why I originally closed the first issue again, because I thought that it was actually a filesystem issue (I didn't look close enough to the MMC errors). But the fact that this is simply reproducible (100%) and fixable (also 100%) is what makes me at least think this is not a FS issue, but something related to the MCC part.

> BTW, I don't see the linked forum thread as being the same issue at all - that was about a mismatched set of software.

I should have been more precise in my language: the reported error (MMC unexpected command error 25) is exactly the same. That's what triggered me looking into the change to the upstream MMC driver, which then resulted in a simple "I really don't understand why, but let's try this with this simple revert".

pelwell, 2025-09-03:
Thanks for all of the additional details.

>> BTW, I don't see the linked forum thread as being the same issue at all - that was about a mismatched set of software.
...
> the reported error (MMC unexpected command error 25) is exactly the same

I suppose that if the result of that mismatched software was running the upstream driver, then it could be another manifestation of the same underlying bug.

> I've have the suspicion that the handover from U-Boot to Linux doesn't work properly.

Having two differences - U-boot and the power cycling - does make the issue more complicated.

First of all, I'm not aware of any clear differences between the two drivers that could account for this. It's a shame it's not the downstream driver which is problematic because that has extensive logging facilities.

A few more questions:
* Have users who experienced the problem had more luck with other cards? (I'm not trying to blame the cards, just narrow down the necessary preconditions)
* Briefly power-cycling a device could leave it in a different state to switching it off for 10 seconds or more. Software _should_ cope with both (unless it breaks the power-on reset hardware), but in your testing how long were the power outages?

spockfish, 2025-09-03:
> Have users who experienced the problem had more luck with other cards? (I'm not trying to blame the cards, just narrow down the necessary preconditions)

We tried with 3 cards, that were also known to work 'in the past' (as in: the previous version). 

Briefly power-cycling a device could leave it in a different state to switching it off for 10 seconds or more. Software should cope with both (unless it breaks the power-on reset hardware), but in your testing how long were the power outages?

For the user to be able to see the issue he needs to wait before RoPieee has fully started, which is somewhere between half a minute or so. I'll ask him in how much time he takes between power off and power on again.

What is also difficult that it is rare: RoPieee has a fairly modest user base (a few thousand users), and I know only 2 users that actually reported it. Now, the software is capable of recovering from it (reboot, which then forces an fsck), but still.

Here's a screenshot from the second user:

[image]

TeSoDNI, 2026-09-17:
Still reproducible on current rpi-6.18.y, on a Pi Zero 2 W, with a deterministic trigger that doesn't involve power loss.

**Environment**
- Raspberry Pi Zero 2 W Rev 1.0, Raspberry Pi OS Trixie arm64 (pi-gen build)
- `6.18.50+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.18.50-1+rpt1 (2026-09-11)`, firmware 2026-09-07
- SanDisk Extreme 128 GB (`mmc0:d555 SM128 119 GiB`); also seen on a 16 GB card

**Reproducer:** fresh flash, first boot (`systemd-growfs-root` grows the root fs from ~3.2 GB to ~119 GB) with `dtoverlay=sc16is752-spi1,int_pin=24` in config.txt. Nothing opens ttySC0/1, and it fails the same with or without the SC16IS752 hardware fitted.

Controlled A/B — same image freshly flashed, same board and PSU, first boot, only config.txt differs:

| config.txt | result |
|---|---|
| `spi=on`, `i2c_arm=on`, `sc16is752-spi1,int_pin=24` | card lost 0.9 s into the resize |
| none of these (2 runs, 2 cards) | resize completes, boots to login |
| `spi=on`, `i2c_arm=on` only | resize completes, boots to login |

With the overlay the same failure also happened in six earlier runs on two cards, including runs with `systemd-growfs-root` masked and with cloud-init disabled — the resize is just a convenient load, not a requirement. Stock Raspberry Pi OS on the same card and board is clean.

```
[   19.330174] EXT4-fs (mmcblk0p2): resizing filesystem from 839680 to 31116800 blocks
[   20.268164] sdhost-bcm2835 3f202000.mmc: unexpected command 25 error
[   20.269804] sdhost-bcm2835 3f202000.mmc: previous command (25) not complete (EDM 00010907)
[   20.276324] sdhost-bcm2835 3f202000.mmc: previous command (25) not complete (EDM 00010803)
                (6 in total)
[   20.295774] I/O error, dev mmcblk0, sector 1948216 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2
[   20.295847] mmc0: card d555 removed
[   20.311632] Aborting journal on device mmcblk0p2-8.
[   20.324662] EXT4-fs warning (device mmcblk0p2): ext4_resize_fs:2188: error (-5) occurred during file system resize
[   20.351222] EXT4-fs (mmcblk0p2): shut down requested (2)
[   21.430236] mmc0: new high speed SDXC card at address d555
```

**`sd_force_pio` has no effect with the upstream driver**, so PIO can't be used to test around this. With `dtparam=sd_force_pio=on` the host still reports `loaded - DMA enabled (>1)`. In `bcm2835_probe()` the property is read into `allow_dma`, which is never used afterwards — DMA depends only on `dma_request_chan(dev, "rx-tx")` succeeding:

```c
allow_dma = !of_property_read_bool(node, "brcm,force-pio");
if (host->pio_limit == 0x7fffffff)
	allow_dma = false;
...
host->dma_chan_rxtx = dma_request_chan(dev, "rx-tx");
```

The same applies to `sd_pio_limit=0x7fffffff`. (Note "mmc: bcm2835: DMA mapping improvements" v2, applied to mmc next in July, replaces `use_dma` with `dma_dev`.)

`dtparam=sd_overclock=25` delayed the failure (card lost 48.5 s into the resize instead of <1 s) but didn't prevent it.

**Related crash.** On one run, error recovery ended in:
```
sdhost-bcm2835 3f202000.mmc: unexpected command 25 error
Unable to handle kernel NULL pointer dereference at virtual address 0000000000000014
pc : bcm2835_finish_data+0x2c/0x80
lr : bcm2835_dma_complete_work+0x100/0x138
Workqueue: events bcm2835_dma_complete_work
```
which matches the race addressed by "mmc: bcm2835: guard against host->data being NULL in bcm2835_finish_data()" (linux-mmc, Aug 2026). This issue looks like one real-world way to reach it.

Happy to test patches or share full serial logs — the reproducer is deterministic here.

spockfish, 2026-10-02:
@TeSoDNI at least you found a way to reproduce this... 

I've 'solved' it by backporting the old driver.