raspberrypi/linux #7033: Kernel crash (NULL pointer dereference) in brcmfmac driver when hostapd is running

HsienChou, 2025-09-03:
### Describe the bug

The `hostapd` service running on a Raspberry Pi 4 `segfault` due to a kernel panic (a `NULL pointer dereference` in the `brcmfmac` Wi-Fi driver).
The crash occurs shortly after a Wi-Fi client (in this case, an `iPhone`) sends a specific `action frame` to the AP.

This issue can be triggered by a simple, non-malicious user action, highlighting a potential stability vulnerability in the driver.

### Steps to reproduce the behaviour

1. Set up a Wi-Fi Access Point (AP) on a Raspberry Pi 4 Model B running Raspberry Pi OS Bookworm using the `hostapd` service.
2. Use an `iPhone` with `iOS 18.6.1 or later` to search for the Wi-Fi AP.
3. Navigate to Settings > Wi-Fi list on the iPhone and tap the `(i)` icon next to the SSID provided by the Raspberry Pi. This action frequently triggers a kernel panic on the Raspberry Pi.

(Note: This behavior occurs regardless of whether the iPhone has previously paired with or is currently connected to the AP. The crash is triggered by the action of tapping the `(i)` icon from the iPhone.)

[image]

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
Raspberry Pi 2024-05-09
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f1c166a2833950a7c44fe19b01780723635a7aa3, stage2
```
```
Apr 17 2024 17:27:09
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)
```

This is the original kernel version`(6.6.28)` that could already trigger this crash.
The logs attached is the latest crash logs after `apt full-upgrade`, so the logs show a newer kernel version`(6.12.34)`.
```
Linux raspi 6.6.28+rpt-rpi-v7l #1 SMP Raspbian 1:6.6.28-1+rpt1 (2024-04-22) armv7l GNU/Linux
```
```
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

### Logs

The output logs was from `/var/syslog` when `hostapd` runs with `-dd` args in `systemd service`

```
Message from syslogd@raspi at Sep  2 17:05:29 ...
 kernel:[  495.011671] Internal error: Oops: 0000000096000007 [#1] PREEMPT SMP

Message from syslogd@raspi at Sep  2 17:05:29 ...
 kernel:[  495.013524] Code: f9401e80 aa1603e2 f9403be1 5280e483 (f9400000)
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: Event message available
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan0
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: MLME event 59 (NL80211_CMD_FRAME) on wlan0(dc:a6:32:7b:12:1c) A1=dc:a6:32:7b:12:1c A2=a6:ff:9a:d9:db:69
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: MLME event frame - hexdump(len=60): d0 00 00 00 dc a6 32 7b 12 1c a6 ff 9a d9 db 69 dc a6 32 7b 12 1c 00 00 04 0a 94 6c 02 00 00 1b 00 00 01 08 00 0c 01 05 01 08 01 07 01 dd dd 07 00 50 6f 9a 11 01 00 03 00 00 00 00
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: Frame event
Sep 02 17:05:29 raspi hostapd[1071]: nl80211: RX frame da=dc:a6:32:7b:12:1c sa=a6:ff:9a:d9:db:69 bssid=dc:a6:32:7b:12:1c freq=2462 ssi_signal=0 fc=0xd0 seq_ctrl=0x0 stype=13 (WLAN_FC_STYPE_ACTION) len=60
Sep 02 17:05:29 raspi hostapd[1071]: wlan0: Event RX_MGMT (18) received
Sep 02 17:05:29 raspi hostapd[1071]: mgmt::action
Sep 02 17:05:29 raspi hostapd[1071]: RX_ACTION category 4 action 10 sa a6:ff:9a:d9:db:69 da dc:a6:32:7b:12:1c len 60 freq 2462
Sep 02 17:05:29 raspi hostapd[1071]: wlan0: GAS: GAS Initial Request from a6:ff:9a:d9:db:69 (dialog token 148)
Sep 02 17:05:29 raspi hostapd[1071]: ANQP: 4 Info IDs requested in Query list
Sep 02 17:05:29 raspi hostapd[1071]: ANQP: Domain Name not available
Sep 02 17:05:29 raspi hostapd[1071]: ANQP: Roaming Consortium not available
Sep 02 17:05:29 raspi hostapd[1071]: ANQP: 3GPP Cellular Network not available
[... 149 lines cut ...]
COMMIT
# Completed on Wed Apr 13 10:28:51 2016
```
5. customized hostapd systemd service (to show more detail logs from hostapd in syslog)
```
$ cat /etc/systemd/system/raspi_hostapd.service
```
```
[Unit]
Description=Enable Access Point
After=network.target dnsmasq.service

[Service]
Type=exec
WorkingDirectory=/usr/raspi/configs/
ExecStartPre=-/usr/bin/killall hostapd
ExecStartPre=/usr/sbin/sysctl -w net.ipv4.ip_forward=1
ExecStartPre=/bin/bash -c '/usr/sbin/iptables-restore < iptables/iptables.rules'
ExecStartPre=/usr/sbin/service dnsmasq restart
ExecStart=/usr/sbin/hostapd -dd hostapd/hostapd.conf
ExecReload=/bin/kill -HUP $MAINPID

[Install]
WantedBy=multi-user.target
```

nolimitdev, 2025-09-16:
Exactly the same issue here. Thanks for knowledge that issues is triggered by iPhones. We did not know it so far. Now we need to find out how to avoid this issue.

aspriel, 2025-09-25:
@HsienChou Can you provide the kernel .config file so I can reproduce the driver build?

Thanks, Arend

amitrohatgi, 2025-09-25:
Oh wow. This has been driving me crazy for the past 2 days on my raspberrypi running bookworm. Hostapd is at 2.10-12 and I'm on kernel 6.1.21-v7l+. I didn't realize that my iphone may be the one triggering it. Watching this thread very closely ...

HsienChou, 2025-09-26:
> [@HsienChou](https://github.com/HsienChou) Can you provide the kernel .config file so I can reproduce the driver build?
> 
> Thanks, Arend

@aspriel Hi, I just sent you the kernel .config file via email.

richard-govivid, 2025-10-02:
I found the bug in kernel 6.1.21 and wrote a patch for it.

(I haven't looked into the newer 6.6. or 6.12 kernels yet, as our device is currently locked on the older kernel version for other reasons.)

richard-govivid, 2025-10-02:
https://lore.kernel.org/linux-wireless/20251002232002.124609-1-richard@govivid.ai/