raspberrypi/linux #7035: Chromium crashes RPiOS when 6.12.44 PREEMPT_RT kernel used

reraikes, 2025-09-03:
### Describe the bug

Somewhere between kernels 6.12.34 and 6.12.44, a problem was introduced in PREEMPT_RT builds.

Both Bookworm and Trixie exhibit the symptom:

6.12.44: Chromium, after visiting several web sites, totally crashes up-to-date RPiOS running on a Raspberry Pi 5.

6.12.34: No problems encountered.

Commits:

6.12.44: eb87a2027c0766c9f670d7c8fa68b86170a0d035

6.12.34: 8f77e03530f65209a377d25023e912b288e039cd

### Steps to reproduce the behaviour

Replace 6.12.34 PREEMPT_RT kernel with 6.12.44 PREEMPT_RT kernel.

Visit several web sites with Chromium.

### Device (s)

Raspberry Pi 5

### System

2025-05-13-raspios-bookworm-arm64.img (updated)

root@raspberrypi:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

root@raspberrypi:~# vcgencmd version
2025/08/20 16:21:58
Copyright (c) 2012 Broadcom
version cd866525 (release) (embedded)

root@raspberrypi:~# uname -a
Linux raspberrypi 6.12.44+rpt-rpi-2712 #1 SMP PREEMPT_RT Tue Sep  2 23:01:23 CDT 2025 aarch64 GNU/Linux

reraikes, 2025-09-05:
Only RT kernels exhibit the issue.

Commit 5a72e3ae00ecdd02244e867c2880a3ac0653ee25 (drm/v3d: Address race-condition between per-fd GPU stats and fd release) on 7/25/25 is the point where this issue is introduced.

nbuchwitz, 2025-09-05:
My bet is on the spinlocks. Try making them raw spinlocks and check if it works then. But this would be nothing more than a workaround.

popcornmix, 2025-09-06:
@mairacanal - any suggestions?

melissawen, 2025-09-10:
Hi, I'm investigating this issue.

Can you check if this patch addresses the issue and didn't cause any regression?

[0001-drm-v3d-create-a-dedicated-lock-to-collect-stats.patch](https://github.com/user-attachments/files/22263536/0001-drm-v3d-create-a-dedicated-lock-to-collect-stats.patch)