raspberrypi/linux #7069: WARNING: Packet RAM has to be on to store the packet, vc4_hdmi.c

andrum993, 2025-09-29:
### Describe the bug

When running a pygame app on Raspberry Pi OS Trixie RC1 image (from https://forums.raspberrypi.com/viewtopic.php?t=392192) I get a kernel warning after switching off the monitor, then switching it back on after a few minutes. This same warning has also been discovered in the past but a reproducer was not found - I now have one.

Backtrace:

```
[  702.937301] ------------[ cut here ]------------
[  702.937308] Packet RAM has to be on to store the packet.
[  702.937346] WARNING: CPU: 3 PID: 1049 at drivers/gpu/drm/vc4/vc4_hdmi.c:726 vc4_hdmi_write_infoframe+0x3e4/0x478 [vc4]
[  702.937375] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm algif_hash algif_skcipher af_alg bnep brcmfmac_wcc binfmt_misc hci_uart btbcm brcmfmac bluetooth aes_ce_blk aes_ce_cipher brcmutil ghash_ce gf128mul sha2_ce cfg80211 sha256_arm64 rpi_hevc_dec sha1_ce pisp_be ecdh_generic ecc v4l2_mem2mem sha1_generic videobuf2_dma_contig libaes videobuf2_memops rfkill videobuf2_v4l2 joydev raspberrypi_hwmon videodev videobuf2_common mc pwm_fan rp1_adc raspberrypi_gpiomem uio_pdrv_genirq uio sch_fq_codel i2c_dev zram lz4_compress fuse nfnetlink ip_tables x_tables ipv6 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper snd_soc_core spidev snd_compress snd_pcm_dmaengine snd_pcm v3d gpu_sched drm_shmem_helper drm_kms_helper snd_timer drm snd i2c_brcmstb spi_bcm2835 drm_panel_orientation_quirks backlight gpio_keys rp1_pio rp1_mailbox rp1_fw nvmem_rmem
[  702.937441] CPU: 3 UID: 1000 PID: 1049 Comm: pipewire Not tainted 6.12.47+rpt-rpi-2712 #1  Debian 1:6.12.47-1+rpt1
[  702.937445] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  702.937447] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  702.937449] pc : vc4_hdmi_write_infoframe+0x3e4/0x478 [vc4]
[  702.937464] lr : vc4_hdmi_write_infoframe+0x3e4/0x478 [vc4]
[  702.937477] sp : ffffc000854cb880
[  702.937478] x29: ffffc000854cb8c0 x28: 000000000000000e x27: ffffc00082af7800
[  702.937482] x26: ffff8000c1843000 x25: ffffc000854cb929 x24: 0000000000000000
[  702.937486] x23: ffff8000c1843000 x22: 00000000fffffff4 x21: ffff800180d42080
[  702.937489] x20: ffff800180d425c0 x19: 0000000000000084 x18: 0000000000000000
[  702.937493] x17: 0000000000000000 x16: ffffd06fce49b418 x15: 0000000000000000
[  702.937496] x14: fffffffffffcb6af x13: 2e74656b63617020 x12: 6568742065726f74
[  702.937499] x11: ffffd06fcf9884f8 x10: 0000000000000269 x9 : ffffd06fce51e634
[  702.937503] x8 : 0000000000017fe8 x7 : 00000000fffff268 x6 : ffffd06fcf9e04f8
[  702.937506] x5 : ffff80003f7b7408 x4 : 40000000fffff268 x3 : ffffaf906ff34000
[  702.937509] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff800140576b00
[  702.937512] Call trace:
[  702.937514]  vc4_hdmi_write_infoframe+0x3e4/0x478 [vc4]
[  702.937527]  write_device_infoframe+0xb0/0x170 [drm_display_helper]
[  702.937532]  drm_atomic_helper_connector_hdmi_update_audio_infoframe+0x8c/0xb8 [drm_display_helper]
[  702.937535]  vc4_hdmi_audio_prepare+0x704/0xf88 [vc4]
[  702.937547]  hdmi_codec_prepare+0xf4/0x170 [snd_soc_hdmi_codec]
[  702.937550]  snd_soc_pcm_dai_prepare+0x7c/0x118 [snd_soc_core]
[  702.937578]  __soc_pcm_prepare+0x4c/0x130 [snd_soc_core]
[  702.937598]  soc_pcm_prepare+0x38/0x68 [snd_soc_core]
[  702.937617]  snd_pcm_do_prepare+0x38/0x60 [snd_pcm]
[  702.937631]  snd_pcm_action_single+0x50/0xc0 [snd_pcm]
[  702.937641]  snd_pcm_action_nonatomic+0x9c/0xc0 [snd_pcm]
[  702.937650]  snd_pcm_prepare+0x84/0xe8 [snd_pcm]
[  702.937659]  snd_pcm_common_ioctl+0xd74/0x1730 [snd_pcm]
[  702.937669]  snd_pcm_ioctl+0x3c/0x60 [snd_pcm]
[  702.937678]  __arm64_sys_ioctl+0xb4/0x100
[  702.937683]  invoke_syscall+0x50/0x120
[  702.937688]  el0_svc_common.constprop.0+0xc8/0xf0
[  702.937691]  do_el0_svc+0x24/0x38
[  702.937694]  el0_svc+0x30/0xf8
[  702.937697]  el0t_64_sync_handler+0x120/0x130
[  702.937699]  el0t_64_sync+0x190/0x198
[  702.937702] ---[ end trace 0000000000000000 ]---
```

System is a Raspberry Pi 5 8GB with no special hardware attached. The hardware I do have attached is the official RTC battery, Active Cooler, official 27W PSU, USB keyboard and mouse, Ethernet connection. Tested on two Raspberry Pi 5 8GB, and with Pi 4 PSU

### Steps to reproduce the behaviour

0. Raspberry Pi OS Trixie RC1 64-bit with graphical desktop running, on a Pi 5 with 8GB RAM. (Other Pis may also work - I have not tried)
1. Save the following python code to a file:
```
import time
import pygame

pygame.init()
x = pygame.display.set_mode((320, 240))
x.fill((0,0,0))
pygame.display.update()

while True:
        print("sleeping...")
        time.sleep(40)
[... 1070 lines cut ...]
     Active: inactive (dead)
TriggeredBy: ○ pulseaudio.socket

○ pulseaudio.socket - Sound System
     Loaded: loaded (/usr/lib/systemd/user/pulseaudio.socket; enabled; preset: enabled)
     Active: inactive (dead)
   Triggers: ● pulseaudio.service
     Listen: /run/user/1000/pulse/native (Stream)

Server String: /run/user/1000/pulse/native
Library Protocol Version: 35
Server Protocol Version: 35
Is Local: yes
Client Index: 103
Tile Size: 65472
User Name: andrew
Host Name: test2
Server Name: PulseAudio (on PipeWire 1.4.2)
Server Version: 15.0.0
Default Sample Specification: float32le 2ch 48000Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.platform-107c701400.hdmi.hdmi-stereo
Default Source: alsa_output.platform-107c701400.hdmi.hdmi-stereo.monitor
Cookie: c988:b58b
```

andrum993, 2025-09-29:
Rest of raspinfo output to get round github 64K text input limit:

```
config.txt
----------

arm_64bit=1
arm_boost=1
arm_freq=2400
arm_freq_min=1500
arm_peri_high=1
audio_pwm_mode=2
auto_initramfs=1
avs_temp=43325
camera_auto_detect=1
core_freq=910
core_freq_min=500
disable_commandline_tags=2
disable_fw_kms_setup=1
disable_l2cache=1
disable_overscan=1
display_auto_detect=1
display_default_lcd=-1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=4
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_depth=16
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq_min=500
hdmi_enable_4kp60=1
hevc_freq=910
hevc_freq_min=500
ignore_lcd=-1
init_uart_clock=0x2dc6c00
initial_turbo=60
isp_freq=910
isp_freq_min=500
mask_gpu_interrupt1=16418
max_framebuffers=2
over_voltage_avs=0x27f10
pause_burst_frames=1
pciex4_reset=1
pmic_turbo_threshold=600
program_serial_random=1
total_mem=8192
v3d_freq=960
v3d_freq_min=500
vpred=9236
vpred_max=9236
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
device_tree=-
overlay_prefix=overlays/
hdmi_cvt:0=
hdmi_cvt:1=
hdmi_edid_filename:0=
hdmi_edid_filename:1=
hdmi_timings:0=
hdmi_timings:1=

cmdline.txt
-----------
reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave nvme.max_host_mem_size_mb=0  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=D8:3A:DD:7D:7D:61 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=7d873838-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=GB

pin configuration
-----------------
[... 809 lines cut ...]
[  702.937598]  soc_pcm_prepare+0x38/0x68 [snd_soc_core]
[  702.937617]  snd_pcm_do_prepare+0x38/0x60 [snd_pcm]
[  702.937631]  snd_pcm_action_single+0x50/0xc0 [snd_pcm]
[  702.937641]  snd_pcm_action_nonatomic+0x9c/0xc0 [snd_pcm]
[  702.937650]  snd_pcm_prepare+0x84/0xe8 [snd_pcm]
[  702.937659]  snd_pcm_common_ioctl+0xd74/0x1730 [snd_pcm]
[  702.937669]  snd_pcm_ioctl+0x3c/0x60 [snd_pcm]
[  702.937678]  __arm64_sys_ioctl+0xb4/0x100
[  702.937683]  invoke_syscall+0x50/0x120
[  702.937688]  el0_svc_common.constprop.0+0xc8/0xf0
[  702.937691]  do_el0_svc+0x24/0x38
[  702.937694]  el0_svc+0x30/0xf8
[  702.937697]  el0t_64_sync_handler+0x120/0x130
[  702.937699]  el0t_64_sync+0x190/0x198
[  702.937702] ---[ end trace 0000000000000000 ]---

EEPROM
------

BOOTLOADER: up to date
   CURRENT: Wed 27 Aug 19:01:47 UTC 2025 (1756321307)
    LATEST: Wed 27 Aug 19:01:47 UTC 2025 (1756321307)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

popcornmix, 2025-09-29:
Thanks - I can reproduce. On my Dell monitor, just switching monitor to standby didn't generate the WARN.
I needed to physically unplug it. I saw the WARN on a replug.

Note: this likely means your monitor deasserts hotplug on standby, but mine doesn't. The HDMI specs are unclear on what the correct behaviour is, but in use, not deasserting is the preferred behaviour.

andrum993, 2025-09-29:
I tested with all 3 of my screens, all of which de-assert HPD at some point during the process of switching off then switching on. I was using the monitor/TV's power or standby switch in all cases.

6by9, 2026-01-09:
This is still on my list of things to look at. I've recreated the issue by having VLC play an audio file with looping and disconnecting the display. pygame is obviously doing the same in having the audio channel open at all times.

ALSA gets notified that the audio device is connected as part of the hotplug event, and if something had the audio output device open already then Pipewire (or similar) can start trying to play audio before Labwc has woken up the video side.
I did look yesterday at swapping enabling the packet RAM to the pm_runtime handlers, but it didn't have the expected result as suspend was apparently called even though video was active. It needs a bit more digging to work out that one.

CutterXYZ, 2026-04-06:
I have the same issue with Kodi running inside [cage](https://github.com/cage-kiosk/cage) on Trixie. Turning the TV off for a few seconds then back on makes Kodi lose sound and Kodi's CPU usage varies between 20% and 100% and the framerate drops. `dmesg` shows the same warning and trace as the OP. Setting `vc4.force_hotplug=3` creates more problems as Kodi then cannot detect the correct resolution.